SensorsIot/IOTstack #160: [Feature Request] add Traefik

greggameplayer, 2020-10-27:
It would be good to add the so called Traefik

robertcsakany, 2020-10-31:
+1 From next week I will have time. I've already planned several weeks ago to add traefik support to be able to proxy ports other than HTTP.

robertcsakany, 2020-12-14:
I have some working prototype in a docker compose file which can do the following things:
 - All web containers proxied over traefik - means subdomans / URL's can be used
 - Automatic Let's encrypt management - means the proxy over HTTPS
 - SSO Single SignOn with Google - means the management pages can be accessed from outside, but protected with login.
 - Samba image added - so configurations can be edited from my desktop machine without any magic outside IOTStack :) - I make some PR today for that.

Please guys check it. Have to discuss how can be integrated to current templates - @Slyke ?

```
version: '3.3'

networks:
  inbound:
    driver: bridge
  internal:
    driver: bridge

services:

# #############################################################################
# #############################################################################

  portainer:
    container_name: portainer
    image: portainer/portainer
    restart: unless-stopped
    ports:
      - target: 9000
        published: 9000
        mode: host
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - ./volumes/portainer/data:/data
    networks:
      - inbound
    labels:
      - "traefik.enable=true"
      - "traefik.docker.network=inbound"

      - "traefik.http.routers.portainer.entrypoints=http"
      - "traefik.http.routers.portainer.rule=Host(`portainer.<!!! YOURDOMAIN !!!>`)"
      - "traefik.http.routers.portainer.middlewares=secured"

      - "traefik.http.routers.portainer-secure.entrypoints=https"
      - "traefik.http.routers.portainer-secure.rule=Host(`portainer.<!!! YOURDOMAIN !!!>`)"
      - "traefik.http.routers.portainer-secure.tls.certresolver=tls"
      - "traefik.http.routers.portainer-secure.middlewares=secured"

      - "traefik.http.routers.portainer-secure.service=portainer-secure"
      - "traefik.http.services.portainer-secure.loadbalancer.server.port=9000"

# #############################################################################
# #############################################################################

  nodered:
    container_name: nodered
    build: ./services/nodered/.
    restart: unless-stopped
    user: "0"
    privileged: true
    env_file: ./services/nodered/nodered.env
    ports:
      - target: 1880
        published: 1880
        mode: host
    volumes:
      - ./volumes/nodered/data:/data
    depends_on:
      - mosquitto
    networks:
[... 378 lines cut ...]
        published: 137
        mode: host
        protocol: udp
      - target: 138
        published: 138
        mode: host
        protocol: udp
      - target: 139
        published: 139
        mode: host
        protocol: tcp
      - target: 445
        published: 445
        mode: host
        protocol: tcp
    read_only: true
    tmpfs:
      - /tmp
    stdin_open: true
    tty: true
    volumes:
      - ./:/mnt:z
    command: '-s "IOT Stack;/mnt;yes;no;no;iot" -u "iot;iot" -p'

```

Paraphraser, 2020-12-15:
I had no idea what Traefik was until I Googled it and landed on this [docs page](https://doc.traefik.io/traefik/). I worked in data-comms at one point in my career so I think I have a reasonable handle on the problem Traefik is solving. My initial reaction was "wow" and "cool", along with amazement at the time and effort you must have put in to getting it to work.

I have no wish to detract from your achievement or to come across as the resident "Mr Negative" but, on reflection, I'm can't say that I'm convinced that the problem being solved is one that the average IOTstack user is likely to have.

My between-the-lines reading of the Discord discussions and GitHub issues is that the average IOTstack user has a small home network to which they are steadily adding IoT devices and for which a single RPi serves as their data concentrator. To the extent that the average IOTstack user looks outwards beyond their home network, it seems to be for services like Cayenne or LoRaWAN. To the extent that the average IOTstack user looks inwards from beyond their home network, it seems to be for remote access solutions like WireGuard.

BUT, I accept that I may be:

1. only seeing a very limited sample of IOTstack users (the ones with problems), and therefore making misguided generalisations about the IOTstack user population;
2. utterly wrong about the interests, skills and needs of the average IOTstack user; and
3. totally missing the point you are trying to make.

On number 3 in that list, if you thought that the compelling IOTstack use case would be immediately apparent upon seeing your `docker-compose.yml`, I'm really sorry but that didn't happen for me. No sudden flash of insight. No star on the horizon. I really need help to understand the use case(s) you see for the average IOTstack user.

When I look at your exemplar `docker-compose.yml` the overall impression I get is that the file looks like it drove over some kind of electronic IED.

> If you'll permit a small amount of editorialising, the Traefik doco claims their main goal is, "make it simple to use". If the `docker-compose.yml` changes are "simple" then I'd really hate to see their definition of "complicated".

What worries me about increasing the complexity of `docker-compose.yml` is the likely impact it will have on comprehension for the average IOTstack user. There are plenty of users who readily acknowledge Portainer as their first instinct, who report discomfort during their occasional visits to the command line, and who will never admit to more than a minimal understanding of the **current** level of compose file nitty gritty. Imagine such a person's likely reaction to all the Traefik scaffolding suddenly appearing the next time they run the menu.

There's also the question of the ability of most IOTstackers to **test** changes they want to propose in PRs for compatibility with Traefik. I look at all that extra detail and, honestly, it gives me the heebies. How should the average IOTstacker approach the problem of developing a new container or making a substantial change to an existing container?

* Ignore the Traefik problem entirely?
* Guess at the appropriate Traefik options and hope for the best?
* Get the impression that low-end users are implicitly being denied the opportunity to contribute because they lack the time, knowledge and wherewithal to test at the high-end?

Absent a compelling use case and/or evidence of widespread demand, my two cents is that I think Traefik support is a probably a niche requirement rather than mainstream. Instead of trying to bolt it into SensorsIot/IOTstack, it might be more appropriate to fork a separate repo for those who want Traefik support. A separate community of interest would have the capability and motivation to test changes made to the upstream SensorsIot/IOTstack, augment those changes where necessary to make them "Traefik compatible", and decide when to put changes into production.

However, nobody put me in charge of anything so this is all just one person's opinion. I'll be very interested to see how other people react, and whether they have any ideas on how to manage the increased complexity if Traefik support comes into the current repo.

robertcsakany, 2020-12-16:
I tend to always generilaze problems, and use to be not care how long the generated codes are. The reason in my work I'm using meta models and transformations, so for me programming is a tool to create and convert models. Thats the reason I like this project - easy to manage abstract logics - and if you see there are repeating patterns what have to be set in traefik. Maybe it can be generalized more, but I spent 2-3 hours to achive that state, I have no reason to make more abstract, because for me enouh to make templates. 

As a matter of fact not a big deal if this compexity is not suitable for this project, I'm managing my own version, my goal is a tool which helps me, and its a big plus if anyone can learn or get some inspirations from my solutions. My IOTStack is a sandbox where I can test things without big consequences. For me more important how easy to access services than the length or complexity of configuration. As I see most of the user don't care about details, only wanna use services which required for their projects. With a good generator they haven't understand the underlying logic. With some good wizard some variations can be configured automatically. Is there DYNS? Allow access from remote - As I see there is PiHole which I think more complex and can cause more damage for local-net (multiple DHCP / routing problem for example) I think this solution seems complex but the result is very simple. I think most of the users does not have clue how the generator / menu itself works, so the complxity there I think does not matter - maybe I'm wrong.

Maybe the problem itself does not exists for most of the users, but there was a topic inside of my PR  (this solution is more general and has more feature - https://github.com/SensorsIot/IOTstack/pull/21 . 
Maybe most of the users does not know the fact that features can be done this easly. I think very little effort to extend the existed merger to add another layers of attributes - its very same as override. I think the complexity can be hide easly for users who doesn't wanna care routing. I think the flexibility can be increased without big increase of complexity. 

Originally my intention was not about to solve the external access of my services, but to resolve port hell. I'm unable to. memorize 10 ports for 10 service - and I have more than 100 devices at home, mqtt, bluetooth, zigbee and wifi, so I need that the DNS names of devices and services be easly managed. I need that I have a central unit which have be completely replaced in 5 minutes - so if something goes wrong, with this solution I can move most of services to a cloud provider (except zigbee) - my wife kills if something goes wrong :) SSO was another thing, I don't wanna make user management per service, and I had lot of experience SAML / SSO authentication in recent times.

So if there are some user interesting, I can make the documentation / template / generator mods. 

And about the templates. I have an idea, these whole bunch of container management can be done without compose, with a docker image which can have the configurator and can modelling the images as a box and the links between the services as lines over a web frontend - like how portainer manages container  (over mounted unix socket) - but it is not a fork, its a new system and its about service orchestration.

Schnuecks, 2020-12-16:
Hi, i think this project could benefit from that. I use nginx as reverse proxy to have it all available under my local domain. I think traefik is a great enhancement. The complexity for users is already given with the port hassle, i don't think normal users will take a deeper look at the docker-compose.yml or create a compose-override.yml but if they do they're not normal users and could take responsibility of the yaml files. normal users run menu.sh create the stack and run it and if it works great. jm2c

Kladderadatasch, 2021-09-19:
Has there been any progress on that matter ? 
There's currently a pull request for nginx in the pipeline, but it looks to me like Treafik would be a much better reverse proxy and I would very much enjoy having one in the main project.

marcelstoer, 2021-09-24:
I was pointed here from #410. That full monty Traeffik configuration looks nice but is much more (both feature wise and complexity wise) than I for one would need.
- Integrating Samba is certainly optional but I use it as well
- The benefit of adding DynDNS is LetsEncrypt support. Personally I would be ok with self-signed certs as my IOTstack is not available outside my private network.

ukkopahis, 2022-06-22:
I think it's time to support HTTPS with proper certificates. There are multiple issues and Discord mentions about scenarios needing this. Both linuxserver/swag and Treafik are able to provide it.

I would even consider providing it by default instead of something users have to explicitly add?

@robertcsakany 
> * All web containers proxied over traefik - means subdomans / URL's can be used
> * Automatic Let's encrypt management - means the proxy over HTTPS

Traefik adds a bit of docker-compose.yml verbosity by way of the labels, which you may just ignore if not using Traefik.

Your example seems pretty comprehensive, but I'd add the DuckDNS acme [provider](https://doc.traefik.io/traefik/https/acme/#providers).

> * SSO Single SignOn with Google - means the management pages can be accessed from outside, but protected with login.

I think SSO integrations should be left as a documented user-options or as a completely separate service, not included by default. Keep it as simple as possible. All services should be able to present their own password prompts or the network is secured and no passwords are even wanted. Note: Nextcloud also has web service integration endpoint URLs that must be open and without SSO interference.

Maybe we could add two mutually exclusive Treafik-services; one with SSO and one without it.

> I think very little effort [would be needed] to extend the existing merger to add another layers of attributes - its very same as override.

It would be easy, true. Just add support for service specific overrides. e.g. `.tempates/traefik/overrides.yml` that work like the already supported `compose-override.yml`.

But is it worth the additional complexity? I would say it isn't. I feel it would be better to add the labels directly to all service.yml files as needed or use a static configuration.

---

@Paraphraser 
> What worries me about increasing the complexity of docker-compose.yml is the likely impact it will have on comprehension for the average IOTstack user.

Yup, it's a valid concern. But maybe a not a too large sacrifice for having all relevant configuration for a service in one place? Maybe even a benefit.

> There's also the question of the ability of most IOTstackers to test changes they want to propose in PRs for compatibility with Traefik.

The Traefik stuff being there will remind them that it too needs to be tested. I feel this is a benefit.

Instead of labels, Traefik also support file based configurations. This would be a static configuration listing all possible services. For unreachable/uninstalled services Traefik will just reply "Bad gateway".

mattmeye, 2023-01-14:
Hi,

**tl;dr** my proposal for a traefik container
- keep it simple (kiss) 
- provide traefik almost naked
- use yaml as configuration
- provide only (some) containers from iotstack as traefik services  (may use, but are not obliged to use)
- provide traefik dashboard with a default port and password ( https://sensorsiot.github.io/IOTstack/Basic_setup/Default-Configs )
- provide the two "default" entrypoints of traefik (web 8080, websecure 8443)
- only document interesting use cases

-----------------------------------------------
 
I also missed traefik in this stack. I came from a rasperry pi kubernetes cluster (microk8s) and used traefik a lot and can contribute a little / can create a pull request. For me, the distributed data level (longhorn) was not stable enough and the performance was rather bad, so I'm keeping my hands off Kubernets on RasperryPis for the time being. I spent quite a lot of money on the Raspberry Pis, so I don't want to replace the hardware with ordinary arm/x64 servers at the moment.

In kubernetes the nicest configuration option of traefik is via custom resource definitions - traefik monitors these resources in every (configured) namespaces and automatically restarts - so every seperate project can just create their needs.

But in this project it doesn't work. In my opinion, the best approach is the infrastructure-as-code approach via YAML configuration. It can be easily backed up and restored.

In my case I used Traefik for the following features:
- automatic TLS certificates via Lets encrypt dns challenge for internal and external routes (aws route53 provider) (e.g. "nextcloudalsofromwww.mydomain.de", "onlyinternalserviceandnotreachablefromwww.mydomain.de"
- proxy for every nicenames (internet to home, home to home); only traefik is configured in my fritzbox for incoming requests
- forwarding http to https (redirect scheme)
- adding basic auth to some routes (services who does not have basic auth, or basic auth from internet accesses)
- no traefik feature, but important use case: i assign all defined routes to pihole dns, so all devices in my network did not leave the network for internal services like https://nextcloudalsofromwww.mydomain.de

Due to the many configuration options, I would specify as little as necessary and provide Traefik almost naked. (no sso and so on). I prefer to see interesting use cases and examples documented.

However, I am thinking about already providing the following things:
- other iotstack containers as traefik services (e.g. for nextcloud, influxdb), so they _can_ be used
- traefik dashboard with password protection on a seperate port (like other containers)
- the two "default" entrypoints of traefik: "web", "websecure"
- use yaml for configuration
- outline some interesting use cases in the documentation (letsencrypt, basic auth, scheme redirect, load balancing, sticky sessions, ... ) - do not copy the traefik documenation

Are you interested in a pull request?

Kind regards, 
Matthias