SensorsIot/IOTstack #161: how to add a new custom docker container

benpoe, 2020-10-31:
i remmebered there was a section on how to add a custom container in the docs, but it is now gone.

anyone got a tip on how to add a custom container manually in my iotstack?
so let's say i need elk https://hub.docker.com/_/elasticsearch

what are the steps i need to follow?
just adding a new entry in docker-compose does install that image, but then i'll just get a command exec error.

Paraphraser, 2020-10-31:
Perhaps read [this gist](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-understand-how-iotstack-works-). Basically, if you browse through the existing .templates you'll probably find something that is pretty similar to what you want to do. Most of the time if you just setup a ./services sub folder to hold any configuration data, and then add a block to define the service in docker-compose.yml it will just work. The gist is really aimed at going the next step of preparing a pull request to make features available to everyone. For that, you'd need to add the necessary hooks to the menu and supply a documentation page. @Slyke has been working on a new menu so you might want to checkout the experimental branch and see how you would hook into that too.

Slyke, 2020-10-31:
Hey @benpoe , as @Paraphraser said, both the current `master` and `experimental` branch support custom containers and configurations.

See: https://sensorsiot.github.io/IOTstack/Custom/ if you're on master.

If you switch to the `experimental` branch, there's a little bit more to explain since the `experimental` branch no longer uses `*.env` files by default (but you can still add them in if you want): https://github.com/SensorsIot/IOTstack/blob/experimental/docs/Custom.md#using-env-files-instead-of-docker-compose-variables

benpoe, 2020-10-31:
thanks, that helped me adding a custom python container for my python service. now to try one of those dockerhubg stacks.

marceldarvas, 2021-07-24:
Thank you once again @Paraphraser for the excellent explanation!

Could you please try fixing the docs link about adding custom containers?

https://sensorsiot.github.io/IOTstack/Custom/

Docs menu => https://sensorsiot.github.io/Custom.html

Both seem to be broken.

In what situations is it better to have independent Docker Stacks on RPI?
I have several containers I would like to add to my small server...

ukkopahis, 2022-06-24:
When services not available in IOTstack don't have to interact with services in the stack, it's simpler and easier to have them separate. https://sensorsiot.github.io/IOTstack/Basic_setup/Custom/ updated to explain this. The best override is one you don't have to do.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke