SensorsIot/IOTstack #162: blynk_server fails to build

liamhogan, 2020-11-01:
Hi,

I've just started trialling the use of IOTstack but ran into an issue when trying to bring up the stack when I include Blynk Server. It fails with an error:
```
Building blynk_server
Step 1/11 : FROM adoptopenjdk/openjdk11
latest: Pulling from adoptopenjdk/openjdk11
ERROR: Service 'blynk_server' failed to build: no matching manifest for linux/arm/v7 in the manifest list entries
```

If I remove Blynk server from the build then bringing up the stack works as normal. I'm brand new to Docker so not really sure how to troubleshoot this error or what is causing it. 

Any help would be greatly appreciated, thanks!

Paraphraser, 2020-11-01:
Perhaps see [this gist](https://gist.github.com/877dev/cfa52044336e1e44d74156ce8b1458aa). My guess is @877dev might be able to help if you have further questions. You might find other hints by searching for "blynk" on the SensorsIot/IOTstack Discord channel.

liamhogan, 2020-11-01:
I followed the instructions on that gist and have tried removing all traces of blynk_server and building again, but the config files that it says will be generated are not, and I am still getting the same error nonetheless...

Paraphraser, 2020-11-01:
Sorry but I don't actually use blynk so I have no practical experience to offer. Let's hope @877dev chimes in. You might try writing to @877 on the Discord channel. My understanding is that Discord discussion is what prompted the creation of the gist. Others may have started with the gist, hit the same problem you are having, and have found a solution.

Paraphraser, 2020-11-01:
Perhaps also see [PR131](https://github.com/SensorsIot/IOTstack/pull/131)

Zephael86, 2020-11-02:
Seems not Blynk, but Docker ralted problem for me.

Can be that around the 20th of Oct the 'adoptopenjdk/openjdk11' repo dropped armv7 support?

I still have an older "latest" image on my Pi:

```
docker run --rm -it adoptopenjdk/openjdk11:latest java -version

openjdk version "11.0.8" 2020-07-14
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.8+10)
OpenJDK Server VM AdoptOpenJDK (build 11.0.8+10, mixed mode)
```

However this command also returns an error for me:
```
docker pull adoptopenjdk/openjdk11

Using default tag: latest
latest: Pulling from adoptopenjdk/openjdk11
no matching manifest for linux/arm/v7 in the manifest list entries
```

As I have only light Docker knowledge, I assume, we have to switch to a different Image. e.g.: the official one

```
docker run --rm -it adoptopenjdk:latest java -version

Status: Downloaded newer image for adoptopenjdk:latest
OpenJDK Server VM warning: No monotonic clock was available - timed services may be adversely affected if the time-of-day clock changes
openjdk version "15.0.1" 2020-10-20
OpenJDK Runtime Environment AdoptOpenJDK (build 15.0.1+9)
OpenJDK Server VM AdoptOpenJDK (build 15.0.1+9, mixed mode)
```

File size differs a bit:

```
REPOSITORY                                 TAG                 IMAGE ID            CREATED             SIZE
adoptopenjdk                               latest              b3caf22a36a4        4 days ago          402MB
adoptopenjdk/openjdk11                     latest              d86e48b9fbf1        3 months ago        376MB
```

'adoptopenjdk' has OpenJDK 15.0.1, while 'adoptopenjdk/openjdk11' has 11.0.8.
The only difference between the images metainfo if that 'adoptopenjdk/openjdk11' has an additional ContainerConfig\ArgsEscaped set to true.

@liamhogan : can you please edit your local Docker file under ~/IOTstack/.templates/blynk_server/Dockerfile
from 'FROM adoptopenjdk/openjdk11' to FROM 'adoptopenjdk' and retry the build?

liamhogan, 2020-11-03:
@Zephael86 looks like you're on to something with the 'adoptopenjdk' version difference.

> @liamhogan : can you please edit your local Docker file under ~/IOTstack/.templates/blynk_server/Dockerfile
> from 'FROM adoptopenjdk/openjdk11' to FROM 'adoptopenjdk' and retry the build?

Did this edit, and tried the build using the IOTstack 'docker-compose' command but for some reason it tries to use the 'adoptopenjdk/openjdk11' still. 
So I tried the build manually... I got past the adoptopenjdk issue 👍 
But now another issue 👎 
The build fails with this error:
```
curl: (28) Timeout was reached
The command '/bin/sh -c curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar' returned a non-zero code: 28
```
Another versioning error, perhaps? 
I'm going to try implementing the changes in #131 now that I know how to do that.

mats-nk, 2020-11-03:
In the .template for blynk the environment variable BLYNK_SERVER_VERSION is version 0.41.10

```
ENV BLYNK_SERVER_VERSION 0.41.10
RUN mkdir /blynk
RUN curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar

``` 

And the version is there, https://github.com/blynkkk/blynk-server/releases/download/v0.41.11/server-0.41.11.jar

So did you incorporate the ENV variable when you did the manual step?

liamhogan, 2020-11-03:
> So did you incorporate the ENV variable when you did the manual step?

I did update the variable, yes. 
As I am now using the changes from #131 the variable is set to use the latest version (0.41.14) but I get the same timeout error...

liamhogan, 2020-11-03:
Okay... this is strange. Trying to figure out the curl timeout. I committed the container in question (the one docker is using to download blynk_server) to a temporary image so I can open a shell to it and try things out manually. Curl is giving a timeout error with **any** request.

Zephael86, 2020-11-04:
I can confirm the curl timeout. Try to figure it out.

What I did until now:

Save images:
```
docker save -o=/home/pi/IOTstack/adoptopenjdk_openjdk11.tar adoptopenjdk/openjdk11
docker save -o=/home/pi/IOTstack/iotstack_blynk_server.tar iotstack_blynk_server
```

Export existing container:
`docker export -o=/home/pi/IOTstack/iotstack_blynk_server_container.tar blynk_server`

Stop container: `docker stop blynk_server`

Delete container: `docker rm blynk_server`

Delete images:
```
docker image rm adoptopenjdk/openjdk11
docker image rm adoptopenjdk
docker image rm iotstack_blynk_server
docker image prune
```

Edit Docker file: `sudo nano ~/IOTstack/.templates/blynk_server/Dockerfile`

Content:
```
#FROM adoptopenjdk/openjdk11
FROM adoptopenjdk
MAINTAINER 877dev <877dev@gmail.com>

#RUN apt-get update
#RUN apt-get install -y apt-utils
#RUN apt-get install -y default-jdk curl

ENV BLYNK_SERVER_VERSION 0.41.14
RUN mkdir /blynk
RUN curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar

# Create data folder. To persist data, map a volume to /data
RUN mkdir /data

# Create configuration folder. To persist data, map a file to /config/server.properties
RUN mkdir /config && touch /config/server.properties
VOLUME ["/config", "/data/backup"]

RUN ln -s /config/mail.properties /blynk/mail.properties && ln -s /config/server.properties /blynk/server.properties

# IP port listing:
# 8080: Hardware without ssl/tls support
# 9443: Blynk app, https, web sockets, admin port
EXPOSE 8080 9443

WORKDIR /data
#ENTRYPOINT ["java", "-jar", "/blynk/server.jar", "-dataFolder", "/data", "-serverConfig", "/config/server.properties", "-mailConfig", "/config/mail.properties"]
ENTRYPOINT ["java", "-jar", "/blynk/server.jar", "-dataFolder", "/data"]
```

Start menu.sh, and 'build stack'.
For blynk 'preserve environment and config files' for the rest 'Do not owerwrite'.

`docker-compose up -d`

Result:
```
curl: (28) Timeout was reached
ERROR: Service 'blynk_server' failed to build: The command '/bin/sh -c curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar' returned a non-zero code: 28
```

Zephael86, 2020-11-04:
After starting a terminal inside the base images, the curl inside the 'adoptopenjdk' is throwing a timeout. While the one in 'adoptopenjdk/openjdk11' is fine.

I am checking it why... and if other base image shall be used?

mats-nk, 2020-11-04:
The only lead I have found is : https://stackoverflow.com/questions/53689943/docker-returned-a-non-zero-code-28-when-install-composer

And the recommendation is to minimize no of layers and combine them like `RUN apt update && apt install xxx`

I'm not sure if this applies to you as the no of layers isn't that many.

Ref. Best practices:

https://docs.docker.com/develop/develop-images/dockerfile_best-practices/#minimize-the-number-of-layers

Hope this can be a clue to your detective work.

Zephael86, 2020-11-04:
Ok, 'adoptopenjdk/openjdk14' image has arm version and working curl.
use `FROM adoptopenjdk/openjdk14`in the Dockerfile. Rebuild the stack, and execute the docker-compose command.

Please note, I have a bit different Dockerfile compared to main and #131 coming from @877dev.
I handle config files different, therefore the ENTRYPOINT is changed. 

@liamhogan can you please confirm that with the above image the blynk local server can be built into a container?

@all: what is needed to merge #131 and #162 into the main?

Log output:
```
pi@raspberrypi:~/IOTstack $ docker-compose up -d
Building blynk_server
Step 1/12 : FROM adoptopenjdk/openjdk14
latest: Pulling from adoptopenjdk/openjdk14
de691405df56: Pull complete
e5a766b93ee2: Pull complete
8c97e104d736: Pull complete
d7a97b3bf2bf: Pull complete
402a80ee4033: Pull complete
204281636005: Pull complete
Digest: sha256:af292f6b2471db2de7bed6665975d24305524a816004f176bb1183c5a62ce887
Status: Downloaded newer image for adoptopenjdk/openjdk14:latest
 ---> 4b3c72387798
Step 2/12 : MAINTAINER 877dev <877dev@gmail.com>
 ---> Running in 7baf242bbe11
Removing intermediate container 7baf242bbe11
 ---> f442e1a8aacc
Step 3/12 : ENV BLYNK_SERVER_VERSION 0.41.14
 ---> Running in b3aba658b069
Removing intermediate container b3aba658b069
 ---> 3005eb23694a
Step 4/12 : RUN mkdir /blynk
 ---> Running in b8e8fb1ee3ba
Removing intermediate container b8e8fb1ee3ba
 ---> 7a5c52e4a915
Step 5/12 : RUN curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar
 ---> Running in 3885a34b2370
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   642  100   642    0     0   1667      0 --:--:-- --:--:-- --:--:--  1671
100 18.8M  100 18.8M    0     0  4394k      0  0:00:04  0:00:04 --:--:-- 4973k
Removing intermediate container 3885a34b2370
 ---> fc81d98e586e
Step 6/12 : RUN mkdir /data
 ---> Running in 788ae760024a
Removing intermediate container 788ae760024a
 ---> 9ced3b4f575a
Step 7/12 : RUN mkdir /config && touch /config/server.properties
 ---> Running in c0e4ef87e2ab
Removing intermediate container c0e4ef87e2ab
 ---> d4cf18fd13ee
Step 8/12 : VOLUME ["/config", "/data/backup"]
 ---> Running in b67609bc357a
Removing intermediate container b67609bc357a
 ---> 995f54111edd
Step 9/12 : RUN ln -s /config/mail.properties /blynk/mail.properties && ln -s /config/server.properties /blynk/server.properties
 ---> Running in 09f072ba4cf5
Removing intermediate container 09f072ba4cf5
 ---> 205e1805b291
Step 10/12 : EXPOSE 8080 9443
 ---> Running in fb754c6fccc2
Removing intermediate container fb754c6fccc2
 ---> 7b75b5bcf059
Step 11/12 : WORKDIR /data
 ---> Running in 7a73eb27d280
Removing intermediate container 7a73eb27d280
 ---> 6b36b020963d
Step 12/12 : ENTRYPOINT ["java", "-jar", "/blynk/server.jar", "-dataFolder", "/data"]
 ---> Running in 9169fb0bf918
Removing intermediate container 9169fb0bf918
 ---> dd73fa809396

Successfully built dd73fa809396
Successfully tagged iotstack_blynk_server:latest
WARNING: Image for service blynk_server was built because it did not already exist. To rebuild this image you must use `docker-compose build` or `docker-compose up --build`.
Creating blynk_server ...
Creating blynk_server ... done
```

liamhogan, 2020-11-04:
Success! Great work @Zephael86 
I can confirm that with the change to openjdk14, the blynk_server can successfully be built into a container.
Like you say, we just need to find out how we can push these changes into the main branch...

Paraphraser, 2020-11-04:
See suggested approach on [Discord channel](https://discord.com/channels/638610460567928832/638610461109256194/773688088123670539)

Paraphraser, 2020-11-05:
> push these changes

To make it easier for @877dev to update the existing PR, perhaps you can list the critical files clear of build-time clutter. It looks to me like it's only the Dockerfile that would need to be listed but I have not been following sufficiently closely as to know for certain whether a successful build also needs changes to the blynk block in docker-compose.yml etc.

Over on Discord, 877Dev is aiming for this weekend so, the sooner it becomes crystal clear what "these changes" encompass, the sooner you might "get lucky", as it were.

See also [Testing Pending Pull Requests](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-) which might come in handy once 877Dev has updated the PR but before it gets merged into the master branch.

liamhogan, 2020-11-05:
I'm fairly sure that (thanks to research help from yourself and @Zephael86) the only change needed to fix this issue is switching the adoptopenjdk version in the Dockerfile. Both of the issues that prevented a build were related to this. 
I made a suggestion for this change on the Dockerfile at #131 - with the same idea of trying to make it as simple as possible to implement. Just hoping that @877dev sees it in time.

877dev, 2020-11-05:
Hi,

Apologies for being awol - life!
I'll make the version changes as soon as I can and update the PR, I'm no GitHub expert I'm afraid.

belenhed, 2020-11-17:
Hi I did all the changes rebuild the file but when I try to run it I got into this error:

Building blynk_server
Step 1/13 : FROM adoptopenjdk/openjdk14
 ---> 4b3c72387798

Step 2/13 : FROM adoptopenjdk
 ---> b3caf22a36a4
Step 3/13 : MAINTAINER 877dev <877dev@gmail.com>
 ---> Using cache
 ---> 47cfc161376d
Step 4/13 : ENV BLYNK_SERVER_VERSION 0.41.14
 ---> Using cache
 ---> b1fbb3a823fb
Step 5/13 : RUN mkdir /blynk
 ---> Using cache
 ---> c568937d21e3
Step 6/13 : RUN curl -L https://github.com/blynkkk/blynk-server/releases/download/v${BLYNK_SERVER_VERSION}/server-${BLYNK_SERVER_VERSION}.jar > /blynk/server.jar
 ---> Running in c8c25eb8c90e
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   642    0   642    0     0      0      0 --:--:--   29628d --:--:--     0
100 18.8M    0 18.8M    0     0      0      0 --:--:--    6536d --:--:-- 3109k
Removing intermediate container c8c25eb8c90e
 ---> e4a6f3473d26
Step 7/13 : RUN mkdir /data
 ---> Running in dc05f8170a4b
Removing intermediate container dc05f8170a4b
 ---> 39745e94b904
Step 8/13 : RUN mkdir /config && touch /config/server.properties
 ---> Running in ab42ca639e88
touch: setting times of '/config/server.properties': Operation not permitted
ERROR: Service 'blynk_server' failed to build: The command '/bin/sh -c mkdir /config && touch /config/server.properties' returned a non-zero code: 1

liamhogan, 2020-11-17:
Hi @belenhed 
Looks like you are having a different issue.
The changes mentioned above have now been committed into the master branch. So you should not need to edit the files manually anymore. Can you confirm you are running the latest build of IOTstack? 

Can you also confirm where you have installed the main IOTstack directory? I've only seen 'setting times' errors relating to working on files over a network share.

belenhed, 2020-11-17:
> Hi @belenhed
> Looks like you are having a different issue.
> The changes mentioned above have now been committed into the master branch. So you should not need to edit the files manually anymore. Can [...]

I was running one from 2 to 3 weeks ago.
I will rebuild the whole thing from scratch again and let you know thank you very much for the answer :)

liamhogan, 2020-11-17:
@belenhed no worries. It's a file permissions issue so a fresh start will likely sort out whatever you have going on. 
Like I suggested, also make sure you are installing directly to ~/IOTstack (if you weren't already). 

I'm going to close this issue as it has been solved but if after you rebuild you still have a problem, just open a new issue regarding it.

liamhogan, 2020-11-17:
Fixed in #131 and committed to master branch

closed 2020-11-17 by liamhogan