SensorsIot/IOTstack #164: [Feature Request] please add unifi-controller and frontail

Schnuecks, 2020-11-01:
This would be a nice enhancement for several users.
Found here: https://github.com/linuxserver/docker-unifi-controller

another suggestion would be nice for openhab users, they often use frontail as a webbased logviewer
Found here: https://github.com/mthenw/frontail

mats-nk, 2020-11-02:
You can get started by testing it your self. `compose-override.yml` is a file you can create in the root and add your content into it.

I have been tinkering around with a NGINX and then I just added to `compose-override.yml`:

`services:
  web:
    image: nginx
    volumes:
      - ./volumes/nginx:/usr/share/nginx/html
    ports:
      - "80:80"
    environment:
      - NGINX_HOST=foobar.com
      - NGINX_PORT=80
    networks:
      - iotstack_nw`

Then just run `./menu.sh` and rebuild the IOTstack again.

Schnuecks, 2020-11-02:
Ok, hi thanks for the reply. This is clarifying why my "docker-compose.override.yml" isn't working as expected. The normal docker-way is docker-compose.override.yml beside the docker-compose.yml. I will try it with the compose.override.yml. Actually i edited the docker-compose.yml and integrated it by hand. I think it would be gone if i run menu.sh to rebuild. Ok, found it in the documentation.

mats-nk, 2020-11-03:
Iv `docker-compose.override.yml ` is the standard, please consider posting a change request so IOTstack is following standards/common practice.

lichtteil, 2020-12-22:
I managed to add the Unifi controller.
In your IOTstack folder create a `compose-override.yml` and add the content from here: https://gist.github.com/lichtteil/83b9f4a733e4ca17e2b25789b3dda3f0

Use `menu.sh` to build your stack again (to include the overwrites to your docker-compose file) and start the container `docker-compose up -d`.

ukkopahis, 2022-06-24:
Duplicate of #98 Add Ubiquiti UniFi controller support.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke