SensorsIot/IOTstack #166: Add compatibility with VS Code

CharlesGodwin, 2020-11-03:
Many of us use Visual Studio Code (aka VSCode) for a lot of editing work. If I was a better git user I'd create a pull request but I'm hoping someone will help me with this.

Can we add these two lines to `.gitignore`? This will allow hassle free updates.
`.vscode`
`*.code-workspace`

Thanks

Paraphraser, 2020-11-04:
Pull Request #168 prepared but if you want to learn how to prepare Pull Requests yourself, I've written [a gist on the topic](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d) that you might find helpful.

Incidentally, as a matter of curiosity, while I see the sense in changing IOTstack's .gitignores file to solve this problem for everyone, if I were a VSCode user, I would have added the lines to my ever-evolving global ignores file:

```
$ tail -3 ~/.gitignore_global 
# VSCode
.vscode
*.code-workspace
```

Is there some reason why that would not work for IOTstack? Is there something about VSCode which means Git's global ignores file is, itself, ignored?

CharlesGodwin, 2020-11-04:
Thanks. In my case I have several private repos that actually include these files in my repo

Charles Godwin

CharlesGodwin, 2020-12-15:
This seems to have been implemented

closed 2020-12-15 by CharlesGodwin