SensorsIot/IOTstack #167: add port for zigbee2mqtt frontend

mpuff, 2020-11-03:
Hello,

the latest zigbee2mqtt has a frontend now, please add this port to the config.

Paraphraser, 2020-11-03:
Can you provide details? A link to the information would help.

mats-nk, 2020-11-04:
I think this is the one mpuff is meaning: https://www.zigbee2mqtt.io/information/frontend.html

`frontend:`
`  port: 8080`
`experimental:`
`  new_api: true`

Paraphraser, 2020-11-04:
Thanks @mats-nk - that makes sense.

I don't have CC2531 adapter so I can't test this directly, but I think this is what needs to be done. I'd appreciate it if someone else could test this and see if it works.

## Step 0

Everything here assumes you are working in `~/IOTstack` so your first command should be:

```
$ cd ~/IOTstack
```

## Step 1

If you have not already installed zigbee2mqtt, run the menu:

```
$ ./menu.sh
```

* Select "Build Stack:
* Enable "zigbee2mqtt" (put the cursor on the line and hit SPACEBAR)
* Press TAB to select \<Ok\> and press RETURN
* Follow the process to the end choosing "Do not overwrite" for all other containers.

Do **NOT** run `docker-compose up -d`.

## Step 2

Attach your CC2531 adapter to your Raspberry Pi.

> Some reports say this should be done with a USB extension cable because connecting the adapter directly to the RPi port creates too much interference.

Make sure the adapter appears:

```
$ ls -l /dev/ttyACM0
```
 
If the device does not show up, you will need to work out why and fix it.

## Step 3

Use your favourite text editor to open `docker-compose.yml`.

Find the `zigbee2mqtt` block. It looks like this:

```
  zigbee2mqtt:
    container_name: zigbee2mqtt
    image: koenkk/zigbee2mqtt
    volumes:
      - ./volumes/zigbee2mqtt/data:/app/data
    devices:
      - /dev/ttyAMA0:/dev/ttyACM0
      #- /dev/ttyACM0:/dev/ttyACM0
    restart: unless-stopped
    network_mode: host
```

Edit the `devices` area to look like this:

```
    devices:
      #- /dev/ttyAMA0:/dev/ttyACM0
      - /dev/ttyACM0:/dev/ttyACM0
```

> Remember: don't use TABs in YAML files, only spaces.

[... 139 lines cut ...]

### Host mode

The template defines:

```
    network_mode: host
```

The sole reason for this seems to be the chicken-and-egg created by:

1. zigbee2mqtt placing its configuration in the `./volumes/zigbee2mqtt/data` directory rather than the more conventional (for IOTstack) `./services/zigbee2mqtt`; and
2. the initial configuration containing:

	```
	  server: 'mqtt://localhost'
	```
	
Host mode is never a great idea. It causes confusion for users (who wonder why they can't see the port in `docker ps`) and makes it harder to work around port conflicts. Granted, the zigbee2mqtt configuration would let you change the port from 8080 to something else but it's still TWO more rules to remember (the first being that the configuration file is the place to change the port, the second being **where** zigbee2mqtt sticks its configuration file).

All things considered, it would be better if zigbee2mqtt was massaged into some semblance of consistency with other IOTstack containers.

The changes here are an attempt to make it consistent but, before we can consider propagating any of this into the template, we need to know if it works, or if there is another reason why the container needs to run in host mode.

Over to someone else...

mpuff, 2020-11-23:
Hello,

the only reason why it is setup with host network i think is that the mosquitto container is not linked to the zigbee2mqtt container, so the mqtt://mosquitto does not work, i does not get an connection.
But when i do host mode with mqtt://localhost, everything is working.

mpuff, 2020-11-23:
Everything is working whe i set it up with your description 👍

mpuff, 2020-11-23:
But when we install openhab, port 8080 is used, maybe we can map this port to another one that is free if we install also other containers ;)

Paraphraser, 2020-11-23:
Assuming that it is set up as above, just edit `docker-compose.yml` to change the external port. For example, if you wanted to use port 8181, it would look like this

```
ports:
      - "8181:8080/tcp"
```

To apply the change, you just `docker-compose up -d`.

If you make the same changes in `./services/zigbee2mqtt/service.yml` (ie so it is identical to the zigbee2mqtt chunk in `docker-compose.yml` then your changes will persist across menu runs until you do a "pull full service from template".

closed 2020-11-24 by mpuff