SensorsIot/IOTstack #169: Portainer menu.sh

Schnuecks, 2020-11-04:
Hi, if i select portainer-ce and build the stack, on the next run of menu.sh portainer and portainer-ce are selected.
On every run i must deselect portainer. selection.txt only contains portainer-ce

CharlesGodwin, 2020-11-06:
This is a bug. In `menu.sh` at line 457 there is this test
`[ $(grep "$index" ./services/selection.txt) ] && entry_options+=("ON") || entry_options+=("OFF")`

It finds portainer twice, once for portainer-ce and once for portainer

The solution  is to change it to search for a match of the entire line  in selections.txt
`[ $(grep --line-regexp "$index" ./services/selection.txt) ] && entry_options+=("ON") || entry_options+=("OFF")`

I doubt if I'm skilled enough to build a pull request for this but I may give it a try in a few days, once I finish another project.

Schnuecks, 2020-11-06:
Great, thanks for looking at the code.

DevSaumyadip, 2020-11-30:
on stack building update , add portainer 2.0 with docker pull portainer/portainer-ce:latest

PR #567 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3
Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
commit 3239e9b4 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3

Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
closed 2022-06-12 by Slyke