SensorsIot/IOTstack #170: Python Install Fails

eleff210, 2020-11-04:
I saw this posted on the prior fork but not seeing the same here.  I'm having an issue where when running docker-compose up -d I get an error attributed to the installation/update of Python.  I've traced it back to no "requirements.txt" file being found in the dynamically created tmp path.  Any ideas on how to fix or what might be within the requirements.txt file so that I can create it?

[image]

StohanzlMart, 2020-12-25:
You need to specify your imports here!

https://www.docker.com/blog/containerized-python-development-part-1/

Example requirements.txt (from a Python project I'm trying to containerise rn): https://github.com/ThomDietrich/miflora-mqtt-daemon/blob/master/requirements.txt

StohanzlMart, 2020-12-25:
Apparantly you can "pip freeze > requirements.txt"

source: https://www.youtube.com/watch?v=gHaFqYRJ_aw

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke