SensorsIot/IOTstack #176: [experimental branch] Nextcloud Database Password Options causes bug in docker-compose.yml

JoeyDodson, 2020-11-08:
When I was testing the experimental branch, I discovered a bug with the Database Password Options for Nextcloud. When choosing the option 'Randomise passwords every build', the `docker-compose.yml` file gets built without the `nextcloud_db` service included. 

To confirm the bug, I removed the `services/` & `volumes/` directories as well as the `docker-compose.yml` file. Then I ran `./menu.sh` again, this time with the default option of 'Use default passwords for this build', which created a functioning `docker-compose.yml` including the `nextcloud_db` service. I then repeated the process again with the 'Randomise..' option and confirmed that the bug happened again. 

I suspect the issue is with either `.templates/nextcloud/build.py` or `.templates/nextcloud/passwords.py`, but I am not able to track it down at the moment. I may take another look at it tomorrow. 

I did confirm that the 'Use default...' option successfully, automatically linked 'nextcloud' and 'nextcloud_db', which is a great feature!

JoeyDodson, 2020-11-09:
I thought it might have to do with an improper environment. I realized I may have had two versions of pyyaml (3.1 & 3.1.1), for example and I wasn't sure which one the menu was using. However, I created a fresh conda environment to test this version of IOTstack and I'm seeing the same behavior (except the version warning doesn't show up in the conda environment anymore). I think it may be safe to rule that possibility out.

Slyke, 2020-12-20:
This seems like a bug. I suspect it's filtering out `nextcloud_db` because it doesn't have its work directory in the services folder and there are a bunch of checks for it that had to have exceptions added to them for `nextcloud` and `nextcloud_db`.

Slyke, 2020-12-20:
Confirmed this bug. Seems to happen when saving the options menu for nextcloud. The bug doesn't happen when pressing ESC to go back, which means that it's from the `build_settings.yml` logic.

Slyke, 2020-12-21:
Hey @JoeyDodson I just pushed a fix for this to `master`. Please reopen if you continue to have this issue.

closed 2020-12-21 by Slyke

JoeyDodson, 2020-12-23:
@Slyke I just pulled the repo and ran a quick test and it seems to be fixed now. Cheers!