SensorsIot/IOTstack #183: How to exploit docker-compose default files

CharlesGodwin, 2020-11-25:
docker-compose opens sever

<pre>
</pre>

mats-nk, 2020-11-25:
But all this is already there `.gitignore` and it contains `compose-override.yml`.

So rename your `docker-compose.override.yml` to `compose-override.yml` and everything will work out of the box.

CharlesGodwin, 2020-11-25:
`compose-override.yml` is a file specifically defined by IOTstack and is only used, in conjunction with menu.sh, to build docker-compose.yml. If you make changes to that file you must rerun menu.sh to build a new, merged, docker-compose.yml. `docker-compose.override.yml` is a file defined by docker-compose program as a default **run time** override file and no preprocessing of the file is required. The biggest differences are the run time override aspect of the file and the file is conformant with docker-compose run-time documentation.

`.env` is not in the current .gitignore.

ukkopahis, 2022-06-24:
Problem files: docker-compose.override.yml and .env
#570 will include these into backups
#553 will add these to .gitignore

PR #689 (2023-06-25): Developer setup improvements and custom file handling
This PR:
- adds the .idea folder to .gitignore to enable other IDEs
- adds the standard docker-compose override mechanism both to .gitignore and the backup

fixes #183
closed 2023-06-25 by Slyke