SensorsIot/IOTstack #186: Revert miscellaneous commands (swap & log2ram)

ComeCaramelos, 2020-11-30:
Hello,

how can I revert the **Miscellaneous Commands** options?

Thanks

ComeCaramelos, 2020-11-30:
I think I've got it. As root:

### Re-install swapfile ([source](https://nebl.io/neblio-university/enabling-increasing-raspberry-pi-swap/)) 
Temporarily Stop Swap
`dphys-swapfile swapoff`
Modify the size of the swap
`sudo nano /etc/dphys-swapfile` CONF_SWAPSIZE=100
Initialize Swap File
`dphys-swapfile setup`
Start Swap
`dphys-swapfile swapon`

### Swappiness to default ([source](https://www.howtoforge.com/tutorial/linux-swappiness/))
`sysctl vm.swappiness=60`

### Uninstall log2ram ([source](https://github.com/azlux/log2ram/blob/master/uninstall.sh))
`chmod +x /usr/local/bin/uninstall-log2ram.sh`
`sh /usr/local/bin/uninstall-log2ram.sh`

Hope this help

Slyke, 2020-12-20:
Closing this issue since it seems to be resolved!

closed 2020-12-20 by Slyke