SensorsIot/IOTstack #188: Adding custom PHP container with custom dependencies

ronnyandre, 2020-12-02:
I want to install a Laravel project on my IOTstack machine. I found several guides on how to do it, and the closest one that works requires changes to the Dockerfile to add "RUN" command to install custom PHP dependencies.

I am not sure where to add this script, so I was wondering if that was something that is possible to do through the compose-override.yml file?

Boman, 2020-12-06:
I'm by far not an expert when building Docker Images. But I would go the following way.
Since the docker-compose.yml (and therefore also the compose-override.yml) file cannot contain build commands like "RUN", we have to create a Dockerfile.
First you have to donwload the Dockerfile which you want to use for the build. I would save the Dockerfile in `~/IOTstack/volumes/laravel-app/build/Dockerfile` so that the file can be backed up by the IOTstack backup.
An example of such a Dockerfile can be seen in this tutorial: https://buddy.works/guides/laravel-in-docker#write-a-docker-file
Then you add your specific "RUN" commands to the Dockerfile.
And finally we have to specify the service definition with the corresponding Dockerfile for building the laravel image in the compose-override.yml
```
  laravel-app:
    build: ./volumes/laravel-app/build
    image: laravel-app
    container_name: laravel-app
    restart: unless-stopped
    user: "0"
    ports:
      - "8181:8181"
```
When you build the stack with the menu and start with `docker-compose up -d` the "laravel-app" image is build and stored in your docker cache. After thar the "laravel-app" service is started with the "laravel-app" image.
Whenever you change the Dockerfile you have also have to execute `docker-compose build` so that the image is rebuild and stored in your cache. Alternatively in the menu you can choose `Docker commands -> pull` which will run the script ./scripts/update.sh to rebuild the image and restart the stack with the new image.

ukkopahis, 2022-06-24:
Question seems to be resolved.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke