SensorsIot/IOTstack #189: Problems/questions about iotstack containers

syswombat, 2020-12-03:
Hello
new to Portainer.io and the iotstack and some question. But love your Videos!
I followed the Videos
#352 Raspberry Pi4 Home Automation Server (incl. Docker, OpenHAB, HASSIO, NextCloud)
#295 Raspberry Pi Server based on Docker, with VPN, Dropbox backup, Influx, Grafana, etc: IOTstack

i just need
Portainer, Node-Red, influxDB, Grafana,  Mosquitto, Telegraf and Home Assistant
to start

so the installation went trough with some warning and error which passed to fast and got access :9000

doing a reboot of the Pi4 
and still ok, 
but Portainer itself whish to update to 2.0 should i do it or go over the the iotstack tool?

then, went to portainer to the list and just for fun selected all container to restart and all gave error which just poped up and disapeared later - the container list is empty now!

about the installation, most of the installation on raspberry pi is done with a start sudo -i and then just working installing like this. It is correct to follow the steps https://sensorsiot.github.io/IOTstack/Getting-Started/ not as root?

Going to containers there is a published ports list, but they all start with 0.0.0.0 would it be possible to edit for all to the IP adresse portainer is running?

and telefgraf need mosquitto i hope Eclipse-Mosquitto is the same!?

have a nice day
vinc

Paraphraser, 2020-12-03:
If you want Portainer 2.0 then run the menu, choose build stack, de-select "portainer", and select "portainer-ce". On a first install, your portainer configuration will be copied to portainer-ce so your password will come across. Portainer uses port 9000 but for IOTstack portainer-ce uses port 9002. That was done so both versions could run together but you can change the port definition if you wish.

By the way, this is all [documented](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer-ce.md).

<hr>

On the topic of root and IOTstack, the best "rule" to follow is to never use `sudo` unless it is necessary. You find out when it is "necessary" by trying all commands without `sudo` first, and then only using `sudo` if you get a permissions error.

All of IOTstack's scripts invoke `sudo` inside the scripts when it is necessary. That, by the way, is another useful hint when working in any Unix environment. If a `grep "sudo" SCRIPT` tells you that SCRIPT contains "sudo" then you shouldn't use `sudo` to run the SCRIPT.

If you have used `sudo` on **any** of the initial `git clone` or `./menu.sh` or `docker-compose up -d` then the best advice I can give you is to start again from a clean image.

Once you have installed everything **without** using `sudo` then the only places you are likely to need `sudo` to interact with IOTstack are inside the `~/IOTstack/volumes` and `/IOTstack/backups/influxdb` folders.

If you're wondering why you should avoid using `sudo` unless it is demonstrably necessary, try running this command, first without `sudo` and then with `sudo`:

```
$ bash -c 'echo $HOME'
```

Without `sudo` the answer is `/home/pi`. With `sudo` the answer is `/root`. In other words, the definition of "~" changes so a path like `~/IOTstack` embedded inside a script that isn't intended to be run with `sudo` creates and looks for things in the wrong place. You get a mess. I hope this makes sense.

<hr>

On the topic of 0.0.0.0, please see [Setting the public IP address for your end-point](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Portainer-ce.md#setting-the-public-ip-address-for-your-end-point).

syswombat, 2020-12-04:
Hello @Paraphraser 
ok this helped, thanks a lot for your information. i did install all without sudo, so i am on the save side. 
changed to portainer-ce worked too, with the guide.
so i will dig deeper and watch more videos too
have a nice day
vinc

Slyke, 2020-12-20:
Closing this issue since it has been answered.

closed 2020-12-20 by Slyke