SensorsIot/IOTstack #190: IOTstack adding node-red -node-feedparser

syswombat, 2020-12-04:
Hello
i would like to be able to read some rss feed so i think this will do the job.
https://flows.nodered.org/node/node-red-node-feedparser

How to install it, 
do i just have to go to the node-red containter go to the commandline and add the command. 
Or do i have to take care on something because it is insyde the IOTstack?

have a nice day
vinc

Paraphraser, 2020-12-05:
Although you might quibble with my definition of "simple", the simplest approach is:

1. Open the Node-Red GUI (eg http://raspberrypi.local:1880/ )
2. From the main menu (three horizontal bars near the top, right of the window), choose "Manage Palette"
3. Click the "Install" tab.
4. In the search field, type "node-red-node-feedparser". After a small delay, the area below will populate with the details.
5. Click the "install" button. A dialog appears with another "install" button. Click that.
6. You'll see a note telling you to restart Node-Red.
7. Click the "Close" button.
8. In a CLI window (eg ssh session) on the Raspberry Pi:

	```
	$ cd ~/IOTstack
	$ docker-compose restart nodered
	```

Your work is done. If you ever have multiple nodes to install this way, do all the installations in the Node-Red GUI first and **then** finish off with the restart command in the CLI.

Nodes that are installed via menu.sh (which only offers a limited selection of nodes) wind up at the **internal** path:

```
/usr/src/node-red/node_modules
```

Nodes installed via Manage Palette wind up at the **external** path:

```
~/IOTstack/volumes/nodered/data/node_modules
```

In `docker-compose.yml` the volumes mapping:

```
./volumes/nodered/data:/data
```

implies that nodes installed via Manage Palette are made available to Node-Red (the process running inside the container) at the **internal** path:

```
/data/node_modules
```

I mention this so you understand why there are three paths and don't get confused by looking in the wrong place.

Nodes added via Manage Palette are also updated (or deleted) via Manage Palette. You finish-off any updates (or deletions) with a `docker-compose restart nodered`.

Because nodes installed via Manage Palette are in the `volumes` area, they get backed-up (assuming you're running backups). That means you don't have to keep track of what you've installed if you ever need to do a bare-metal restore - it will be in the backup.

Final point. If a node is installed in both places (by menu.sh and Manage Palette), Manage Palette take precedence. This is both good and bad. It gives you an easy way to update a menu.sh node to a new version (as an alternative to rebuilding the local image of nodered). However, the next time nodered is rebuilt, it may get an even later version of that node but that later version will be blocked by the one updated by Manage Palette. There's no easy way of figuring out what's going on.

Also, sometimes, nodes installed by menu.sh turn up in the persistent store area without you doing a thing to cause it. I have no idea why that happens but it creates a bit of a mess that has to be cleaned up by hand. I've been tinkering with a new gist which explains the how-to of this but it's not yet ready for publication.

Hope this helps.

Slyke, 2020-12-20:
Closing this issue since it's been answered. @syswombat Feel free to reopen if you believe this module should be included in the main stack.

closed 2020-12-20 by Slyke