SensorsIot/IOTstack #191: Portainer CE option not present

bburkett52, 2020-12-05:
I preface this with the admission that I am a pretty novice user, learning everyday!
That said, I have migrated to your fork of IOTstack, but when I run menu.sh, to go to PortainerCE, the PortainerCE option is not there. It is not present in the menu.sh script.
Not sure how to fix.
Would anyone offer help?
Thanks,
~Bob

Paraphraser, 2020-12-05:
You and me - both. In some ways the absolute dearth of knowledge is the most fun thing about IOTstack.

Let's try a couple of things. First:

```
$ cd ~/IOTstack
$ git remote -v
origin	https://github.com/SensorsIot/IOTstack.git (fetch)
origin	https://github.com/SensorsIot/IOTstack.git (push)
```

If you don't see those URLs then it might mean you're cloned from a different repo (either your own fork or gcgarner). You can try fixing that with:

```
$ cd ~/IOTstack
$ git remote set-url origin https://github.com/SensorsIot/IOTstack.git
$ git pull origin master
```

Sometimes just a `git pull` is needed instead of `git pull origin master`. A git expert could probably explain why but it's all a mystery to me.

If git moans about merge conflicts, undo the damage with:

```
$ git merge --abort
```

Next, make sure you're on the master branch:

```
$ cd ~/IOTstack
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

You're looking for "On branch master". If you see another branch name, try:

```
$ git checkout master
$ git pull
```

At various points along the way, you can check whether your clone is up-to-date by:

```
$ ls -1 ~/IOTstack/.templates | grep "portainer-ce"
$ grep "portainer-ce" ~/IOTstack/menu.sh
```

Once you're up-to-date, you should get one hit from the first command and two from the second command. If you see that, you should be good to go (in the sense of running `menu.sh`).

Another diagnostic is:

```
$ git log | head -8
commit 416091c4060ace4078d43e46a62e54e94393419d
Merge: 574b08d 6923c3f
Author: Slyke <steven.lawler777+github@gmail.com>
Date:   Fri Nov 13 15:19:57 2020 -0800

    Merge pull request #148 from Paraphraser/20201011-influxdb-env-template
    
    Use only known-good keys in InfluxDB environment template
```

That's the latest and greatest at the moment. If you open [SensorsIot/IOTstack](https://github.com/SensorsIot/IOTstack) in your browser and look at the top of the files area where you see:

```
Slyke Merge pull request #148 from Paraphraser/20201011-influxdb-env-template …
```

you'll see "416091c" which is the first few bytes of the commit signature above from `git log`.

Basically, if you see the same commit signature then you should be up-to-date.

Worst case: none of this helps. We can work around that too but I won't go into that until you tell me that you are still stuck.

bburkett52, 2020-12-07:
Phil,

Looks like I may have something messed up. See below for output I get when
I run "git checkout master":

pi@dockerpi:~/IOTstack $ git checkout master
M .bash_aliases
M .gitignore
M .native/rpieasy.sh
M LICENSE
M README.md
Already on 'master'
Your branch and 'origin/master' have diverged,
and have 11 and 198 different commits each, respectively.
  (use "git pull" to merge the remote branch into yours)

Subsequently, "git pull" reports:

pi@dockerpi:~/IOTstack $ git pull
error: Your local changes to the following files would be overwritten by
merge:
.gitignore
README.md
Please commit your changes or stash them before you merge.
Aborting

The grep commands suggested to not find portainer-ce

I am not sure how to attempt to correct it from here.

Thanks again for your help!

Regards,
Bob

Paraphraser, 2020-12-07:
This is what I would do:

1. Move everything that is causing conflicts out of the way. For example:

	```
	$ cd ~/IOTstack
	$ mv .gitignore .gitignore.off
	$ mv README.md README.md.off
	```

2. Then try another `git pull`. If more conflicts appear, repeat from step 1. Keep going until there are no more complaints.

3. Figure out whether the differences are significant. I mostly use `diff -y` for this because it puts up the old and new side-by side, as in:

	```
	$ diff -y .gitignore .gitignore.off
	```
	
	In that example, `.gitignore` will be "as downloaded from GitHub" while `.gitignore.off` is your version. If you see anything that needs to be preserved, either:
	
	- re-edit `.gitignore` (which will cause the problem to recur next time you want to update against GitHub), or
	- add the changes to `~/.gitignore_global` (this having the advantage it won't cause confliicts).

If you don't have a global ignores file, you'll find instructions on setting one up [here](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-local-computer-).

I can't imagine why you might have changed the README.md but if you've been adding notes etc then call it something like "MYREADME.md" and it will no longer be a problem.

bburkett52, 2020-12-07:
I have completed your mv suggestions above with the following results:

pi@dockerpi:~/IOTstack $ mv .gitignore .gitignore.off
pi@dockerpi:~/IOTstack $ mv README.md README.md.off

pi@dockerpi:~/IOTstack $ git pull
Auto-merging menu.sh
CONFLICT (content): Merge conflict in menu.sh
Automatic merge failed; fix conflicts and then commit the result.

I tried renaming the menu.sh file:

pi@dockerpi:~/IOTstack $ mv menu.sh menu.sh.off

pi@dockerpi:~/IOTstack $ git pull
error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.

I put the menu.sh file back as it was and get the same git pull error message:

pi@dockerpi:~/IOTstack $ mv menu.sh.off menu.sh

pi@dockerpi:~/IOTstack $ git pull
error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.

I don't know how to fix the merge issue and am concerned about crashing my installation.
Thanks for you time and patience in helping me!
~Bob

bburkett52, 2020-12-07:
Phil,

I have muddled my way thru getting thru all the above issues so I will close this issue!
Not really sure what fixed it, but I held my breath and tried several things. It looks like the menu.sh file was getting corrupted. I manually put a new copy in place and, after fixing some other errors, was able to get things correct.
I do have the Portainer-ce option now so I will be closing this issue!

I still have issues getting Portainer-ce to install without error, but I will try to sort that one out on my own.
I may be back! :)

Again, thank you for your patience and help!
It is greatly appreciated!
I can usually dig and get answers on my own, but with this I am treading in a lot of new territory!

Best regards,
~Bob

closed 2020-12-07 by bburkett52

Paraphraser, 2020-12-07:
I've also ben able to muddle through most Git problems (the "kick, curse, kick harder" method) but a couple of times I've had to give up and take more drastic action:

```
$ cd ~/IOTstack
$ docker-compose down
$ cd ..
$ mv IOTstack IOTstack.old
$ git clone https://github.com/SensorsIot/IOTstack.git IOTstack
$ mv IOTstack.old/docker-compose.yml IOTstack/
$ mv IOTstack.old/services IOTstack/
$ mv IOTstack.old/backups IOTstack/
$ sudo mv IOTstack.old/volumes IOTstack/
$ cd ~/IOTstack
$ docker-compose up -d 
```

In essence, this is sort of the corollary of the point I was making at [Preparing a Pull Request for IOTstack » Understand how IOTstack works » Key insights](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-key-insights-). A clean checkout creates a brand new "factory" and then you just mimic what a restore does moving your configuration into place.

If you have any other stuff lying around in the "old" directory, that can be moved too (obviously). I usually have a ".bak" for the compose file plus other scripts I'm tinkering with.

The only "move" you should need `sudo` for is the `volumes` directory. If you encounter the need for `sudo` with anything else then that *might* indicate a problem to be investigated. If you're running InfluxDB and have taken a backup, `~/IOTstack/backups/influxdb` will be owned by root but the parent should be owned by pi and therefore moveable without `sudo`.

> I'm harping on `sudo` because more than one problem with IOTstack misbehaviour has been traced to an unexpected side-effect of an unnecessary ancient `sudo`.

Once you're happy, you can clobber what remains of `~/IOTstack.old`.

bburkett52, 2020-12-07:
Thanks for the doomsday recovery plan!
~Bob