SensorsIot/IOTstack #193: ConbeeII (Conbee2) Handling dynamic USB reconnect/attach

bigbread59, 2020-12-09:
Please analyze and take into account what is reported in this link:

https://github.com/marthoc/docker-deconz/issues/298

to improve the connection of the device CONB 2 (DECONZ)

Paraphraser, 2020-12-09:
I think the general idea is that he/she who notices should do his/her own analysis and [prepare a Pull Request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d).

This is especially the case when there's extra hardware involved. For example, I have neither z-wave nor conbee so I couldn't test anything. I suspect many other IOTstackers would be in the same boat. We will have to rely on people who do have the hardware.

Nobody bites. We're all newbies at this. We all learn by doing.

bigbread59, 2020-12-09:
I apologize for my lack of skill.  
I don't think I can do what you suggest.  
I'm sorry

Paraphraser, 2020-12-09:
And neither does anybody else think they have the skills when they first get here (me included). You're already a fair way down the path. You've identified the problem, have a GitHub account and, presumably, the relevant hardware.

That said, I don't think this problem has a solution. At least, not yet.

Reading the material at the link you provided, the current issue seems to be the `device_cgroup_rules` option going away and coming back again, which version of what is supported by IOTstack, with the objective being to get to compose 3.7 or later. Yes?

If you open menu.sh in a text editor and hunt around, you'll find that `docker` gets installed via a `curl` piped to `sh` while `docker-compose` is installed via `apt install`. Thereafter, both are maintained by your own regular `apt update` / `apt upgrade` sequences. When I run `docker version` on my systems, my docker engines are 20.10.0. The [compatibility matrix](https://docs.docker.com/compose/compose-file/compose-versioning/#compatibility-matrix) says that compose file format 3.8 only needs 19.03.0 so, all other things being equal, I should be able to just change the docker-compose.yml header to 3.8 - which barfs. Same for you, right?

Googling the error led me to:

```
sudo curl -L "https://github.com/docker/compose/releases/download/1.24.1/docker-compose-$(uname -s)-$(uname -m)" -o /usr/local/bin/docker-compose
```

and, from that, to [releases](https://github.com/docker/compose/releases). I might be blind but I don't see any indication of 32-bit in there and, when I try that URL "as is" or with a recent version like 1.27.4, I get "not found".

What I **think** this means (as opposed to **know**) is that anyone (IOTstack or anything else) running Docker on a Raspberry Pi needs to wait for whoever it is that maintains the `apt` infrastructure to bring us more recent versions.

I'm going to stop at this point. If you want to follow these breadcrumbs back to the `apt` side of things and try to encourage some action, go to it.

bigbread59, 2020-12-10:
Hi Paraphraser, Thank you very much,  Thank you for your time and your help. I have read carefully what you wrote and  I have learned a lot !

Thanks again.
Pasquale

Paraphraser, 2020-12-10:
I did some more digging and realised that docker-compose is a Python script so there was no obvious reason why there should be any architecture dependency on either ARM or 32-bit. Indeed, that `curl` command above wouldn't work for docker-compose at 1.21.0 (the version we're running) either.

Try this:

```
$ sudo apt remove -y docker-compose
$ sudo apt install -y libffi-dev libssl-dev
$ sudo apt install -y python3-dev
$ sudo apt install -y python3 python3-pip
$ sudo pip3 install docker-compose
```

[image]

I was able to change 3.6 to 3.9 in my docker-compose.yml header and bring up my stack without any complaints from docker-compose. I didn't try the option that began this discussion - I'll leave that to you.

[This](https://devdojo.com/bobbyiliev/how-to-install-docker-and-docker-compose-on-raspberry-pi) is where I found the clues. See also [github.com/docker/compose](https://github.com/docker/compose].

menu.sh did its initial checks OK so, really, the only question remaining in my mind is whether the installation code in menu.sh should be changed to install docker-compose this way.

Paraphraser, 2020-12-10:
During the `apt install` above, I noticed messages about obsolete stuff and recommendations for `sudo apt autoremove`. I did that and it broke the new docker-compose. I fixed it by re-running the above `install` commands (not the `remove` - that's a one-time thing). I mention this so you don't fall into the same trap.