SensorsIot/IOTstack #196: Home Assistant stop working after 1st reboot

beecomgr, 2020-12-11:
Hi

Im working with IOTstack and Home Assistant from August 2020 without any issue. But last month something change and it stopped working.

Im running Raspios Buster with Raspberry PI 4 and im using IOTstack for installing HomeAssistant supervised.
Everything goes fine the first time. I'm able to connect to homeassistant and even I can download addons but with the first reboot of Raspberry only the Observer container running from the 7 containers that must be run.

Last time I used the same method without having problems was at first days of November.

Thx in advance

Nickous, 2020-12-11:
Maybe issue Docker 20.10 and Supervised ?
[https://alerts.home-assistant.io/#docker_2010.markdown](url)

beecomgr, 2020-12-11:
> Maybe issue Docker 20.10 and Supervised ?
> [https://alerts.home-assistant.io/#docker_2010.markdown](url)

Thank you Nickous

Docker 20.10 indeed was the problem.

For helping other people, I solved it by downgrading from version 20.10 to 19.03.14 by the command (for Raspian OS)
- sudo apt-get install docker-ce=5:19.03.14~3-0~raspbian-buster docker-ce-cli=5:19.03.14~3-0~raspbian-buster

You can check which version you have depending on your OS by
- apt-cache madison docker-ce

closed 2020-12-11 by beecomgr