SensorsIot/IOTstack #199: Home Assistant error missing Network-manager

syswombat, 2020-12-12:
hello 
i startet and wished to install  install Home Assistant
./menu.sh

Docker is installed so i select 
Install Home Assistant

select my Rasperpi 4 32bit

and then got a 

[error] missing: network-manager

why? How to install it

have a nice day
vinc

Paraphraser, 2020-12-12:
See [this gist](https://gist.github.com/Paraphraser/55643bd660db46716f6c81266905cae8#hassio-support-optional).

Some of the same information is at [the HA doco page](https://github.com/SensorsIot/IOTstack/blob/master/docs/Containers/Home-Assistant.md) but I agree it's incomplete. I also agree that it would be a nice idea if (a) the doco was complete and (b) the menu took care of all these details but I doubt that that will happen until someone who understands HA takes a look at both the doco and menu, and puts some effort into a Pull Request.

syswombat, 2020-12-12:
Hello @Paraphraser Phill
thanks for your fast feedback. i have installed but should it - at the end not be visible in portainer too?

have a nice day Down Under
vinc

Paraphraser, 2020-12-12:
Sorry, I really can't help with any of this. I don't run "Home Assistant" or "Hass.io" (I'm not even sure whether those are the same thing or not - the whole topic seems encrusted with "home" this, that and the other thing that I do not grok).

I do have Portainer-ce installed but the only time I use it is if someone complains about a Portainer issue. Quite honestly, I've never seen the point of Portainer. That may be because I'm a CLI person.

But, as to whether anything to do with HA should show up in Portainer ... (crickets)

What do you mean, lost some data inside Node-Red? I have two data sources for Node-Red. Mosquitto for about 90% of incoming data, HTTP for a PurpleAir air-quality monitor, and I also have HTTP endpoints open for trigboards to check for software updates which are served by Node-Red.

Other than **explicable** situations (such as restarting the Mosquitto container preventing Node-Red from subscribing, or restarting the InfluxDB container preventing Node-Red from recording an event), I have yet to lose any data inside Node-Red. If the MQTT packet gets to Mosquitto, or the HTTP air-quality data gets to Node-Red, it gets into my database. 100% reliable. So, I'm really curious if there are scenarios where that doesn't happen.

\<soapbox\>

Mind you, I distrust network-manager big time. Even if I did see a use-case for Home Assistant, its dependence on network-manager would see me search for alternative solutions. It'll be a really frigid day in Hades before network-manager ever gets near any of my systems. If it turns out that network-manager futzing with the RPi's data-comms is implicated in your data loss, that'll just be one more reason for me to salt the ground network-manager goes anywhere near.

\</soapbox\>

syswombat, 2020-12-13:
Ok it is Interesting to read.
but maybe we should mail to ask more question, it seems it is more a philosophical aspect ;-)
have a nice day
vinc