SensorsIot/IOTstack #200: Thingy52 detected but fails to connect

saket424, 2020-12-13:
I am able to scan and find the bluetooth mac address of the nearby thingy52 within node-red but am unable to get it to connect -- it says "missing" when I try to wire it up to the ble-generic-input node

Anyone have any success with this thingy52 and can pair and read the sensors ? I am using a new iotstack install

saket424, 2020-12-14:
I figured out how to get both the thingy52 and the thundersense2 devices into a connected state via the ble-generic-node after i disable scanning but each time i restart the flow, the two nodes get scanning enabled again and i have to manually go and turn it off and get it back to connected state

Unfortunately, even in the "connected" state, I am not getting any data flowing to the debug node. I can confirm that the two devices are connected but i do not see any sensor data in node-red debug node but i do see some rssi data in bluetoothctl from the eddystone beacon in the thingy52

[CHG] Device F1:BB:80:AC:AD:E4 RSSI: -77
[CHG] Device F1:BB:80:AC:AD:E4 ServiceData Key: 0000feaa-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E4 ServiceData Value:
  10 ee 03 67 6f 6f 2e 67 6c 2f 70 49 57 64 69 72  ...goo.gl/pIWdir

saket424, 2020-12-14:
@Paraphraser
Thanks to your lovely work, I have got this far but now I actually want to see some data from the sensor and not just the scannable mac addresses and the "connected" message. Any ideas on why I am not seeing debug output with the bluetooth payload data from the debug node?

Paraphraser, 2020-12-14:
Sorry but no. I do not actually have any Bluetooth devices that I am using to talk to NodeRed. I did order a couple of HC-06 RS232 to Bluetooth modules that I was intending to try out but, by the time they arrived, I had moved on. I have yet to find a project that suggests one of those modules as part of the solution.

Originally, my focus was on getting BLE working to solve a problem that someone else was having. The basic problems were the NodeRed image not having the prerequisites and what seemed (to me) to be a great deal of confusion over whether host mode was or wasn't required or whether it achieved anything (no and no). I pretty much stopped once I started to see MAC addresses appearing in NodeRed. I assumed that if NodeRed could see a device then the device could probably see the RPi and the rest was detail. Your experience suggests I might have been wrong.

Based on experience working with LoRa, I think the question I would ask myself is how certain I was that the sending station was actually sending a payload? Do you have anything else you can use to prove that it's actually sending? Then, if I had a spare RPi lying around, I'd run up a clean image and install NodeRed (non Docker) all by its lonesome and see if I could get it to receive. Once I had a known-good flow, I'd move that to the containerised version and start nosing through the Docker-Compose doco, seeing if I could find likely options and tinkering with them.

Other than that...

Sorry.

saket424, 2020-12-14:
@Paraphraser
Thanks for your response . I am certain i am establishing a connection but not so certain if the gatt data is flowing. There is a thingy52 snap from canonical that seems to work better ( https://github.com/anonymouse64/thingy52-snap/ ) but not fully reliably. Maybe this stuff is abandonware and a deadend
If someone else wants to weigh in, I am all ears

https://community.openenergymonitor.org/t/alternative-means-of-measuring-room-temperature/9009/8

I think some of this bluetooth node_modules stuff is abandonware 

thingy52
usage: thingy52 [-h] [-n COUNT] [-t T] [--temperature] [--pressure]
                [--humidity] [--gas] [--color] [--keypress] [--tap]
                [--orientation] [--quaternion] [--stepcnt] [--rawdata]
                [--euler] [--rotation] [--heading] [--gravity] [--battery]
                [--speaker] [--microphone]
                mac_address

Here are the gatt characteristics of thingy52 that the ble-generic node and bluetoothctl identifies for us

<Unnamed> (8ec90003f3154f609fb8838830daea50)Write (Custom Type)
<Unnamed> (ef6801099b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801089b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801079b3549339b1052ffa9740042)Read (Custom Type)
<Unnamed> (ef6801069b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801059b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801049b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801029b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6801019b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6802069b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6802059b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6802049b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6802039b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6802029b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6802019b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6803039b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6803019b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6803029b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef68040a9b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804099b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804089b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804079b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804069b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804059b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804049b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804039b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804029b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6804019b3549339b1052ffa9740042)ReadWrite (Custom Type)
<Unnamed> (ef6805049b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6805039b3549339b1052ffa9740042)Notify (Custom Type)
<Unnamed> (ef6805029b3549339b1052ffa9740042) (Custom Type)
<Unnamed> (ef6805019b3549339b1052ffa9740042)ReadWrite (Custom Type)

[NEW] Device F1:BB:80:AC:AD:E2 Thingy12
[CHG] Device F1:BB:80:AC:AD:E4 RSSI: -76
[CHG] Device F1:BB:80:AC:AD:E4 ServiceData Key: 0000feaa-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E4 ServiceData Value:
  10 ee 03 67 6f 6f 2e 67 6c 2f 70 49 57 64 69 72  ...goo.gl/pIWdir
[NEW] Primary Service
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service000a
	00001801-0000-1000-8000-00805f9b34fb
	Generic Attribute Profile
[NEW] Characteristic
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service000a/char000b
	00002a05-0000-1000-8000-00805f9b34fb
	Service Changed
[NEW] Descriptor
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service000a/char000b/desc000d
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[NEW] Primary Service
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service000e
[... 209 lines cut ...]
	Client Characteristic Configuration
[NEW] Primary Service
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service0066
	0000fe59-0000-1000-8000-00805f9b34fb
	Nordic Semiconductor ASA
[NEW] Characteristic
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service0066/char0067
	8ec90003-f315-4f60-9fb8-838830daea50
	Vendor specific
[NEW] Descriptor
	/org/bluez/hci0/dev_F1_BB_80_AC_AD_E2/service0066/char0067/desc0069
	00002902-0000-1000-8000-00805f9b34fb
	Client Characteristic Configuration
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: 0000fe59-0000-1000-8000-00805f9b34fb
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: ef680100-9b35-4933-9b10-52ffa9740042
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: ef680200-9b35-4933-9b10-52ffa9740042
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: ef680300-9b35-4933-9b10-52ffa9740042
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: ef680400-9b35-4933-9b10-52ffa9740042
[CHG] Device F1:BB:80:AC:AD:E2 UUIDs: ef680500-9b35-4933-9b10-52ffa9740042
[CHG] Device F1:BB:80:AC:AD:E2 ServicesResolved: yes
[CHG] Device F1:BB:80:AC:AD:E2 Name: Thingy1223
[CHG] Device F1:BB:80:AC:AD:E2 Alias: Thingy1223

saket424, 2020-12-15:
I managed to get a node-red flow working with thingy52 on FW 2.2 

https://nordicsemiconductor.github.io/Nordic-Thingy52-FW/documentation/firmware_architecture.html

Here is the sample flow using nodered-contrib-generic-ble

[{"id":"f454226b.1163c","type":"Generic BLE in","z":"8e787ad.6aa1688","name":"","genericBle":"2d8aedfc.d96572","useString":false,"notification":true,"x":640,"y":760,"wires":[["7a8ccae7.d4c604"]]},{"id":"7a8ccae7.d4c604","type":"debug","z":"8e787ad.6aa1688","name":"","active":true,"console":"false","complete":"false","x":870,"y":760,"wires":[]},{"id":"9ea31633.0311d8","type":"Generic BLE out","z":"8e787ad.6aa1688","name":"","genericBle":"2d8aedfc.d96572","x":860,"y":580,"wires":[]},{"id":"d948a714.3cf518","type":"inject","z":"8e787ad.6aa1688","name":"Get All Readable Attributes","props":[{"p":"payload","v":"","vt":"date"},{"p":"topic","v":"","vt":"string"}],"repeat":"","crontab":"","once":false,"topic":"","payload":"","payloadType":"date","x":290,"y":600,"wires":[["f454226b.1163c"]]},{"id":"663540e5.d4a24","type":"inject","z":"8e787ad.6aa1688","name":"Start Temperature Notify Events for 5 seconds","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"ef6802019b3549339b1052ffa9740042","payload":"{\"notify\":true,\"period\":5000}","payloadType":"json","x":290,"y":740,"wires":[["f454226b.1163c"]]},{"id":"c61ef0f3.00725","type":"inject","z":"8e787ad.6aa1688","name":"Connect a Thingy52 SensorTag","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"connect","payload":"","payloadType":"date","x":590,"y":500,"wires":[["9ea31633.0311d8"]]},{"id":"759911ab.f04a2","type":"inject","z":"8e787ad.6aa1688","name":"Disconnect a Thingy52 SensorTag","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":"","topic":"disconnect","payload":"","payloadType":"date","x":320,"y":840,"wires":[["f454226b.1163c"]]},{"id":"97b94248.e63c5","type":"comment","z":"8e787ad.6aa1688","name":"Perform after connected","info":"","x":290,"y":560,"wires":[]},{"id":"daf4b471.ce3888","type":"comment","z":"8e787ad.6aa1688","name":"Perform after Temperature Measurement Started","info":"","x":360,"y":660,"wires":[]},{"id":"60da101e.5c62d","type":"comment","z":"8e787ad.6aa1688","name":"Disconnect a peripheral device","info":"","x":310,"y":800,"wires":[]},{"id":"2d8aedfc.d96572","type":"Generic BLE","localName":"Thingy12","address":"f1:bb:80:ac:ad:e2","uuid":"f1bb80acade2","characteristics":[{"uuid":"2a05","name":"Service Changed","type":"org.bluetooth.characteristic.gatt.service_changed","notifiable":false,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"8ec90003f3154f609fb8838830daea50","name":"<Unnamed>","type":"(Custom Type)","notifiable":false,"readable":false,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801099b3549339b1052ffa9740042","name":"NFC Tag content characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801089b3549339b1052ffa9740042","name":"MTU Request","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801079b3549339b1052ffa9740042","name":"FW version","type":"(Custom Type)","notifiable":false,"readable":true,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6801069b3549339b1052ffa9740042","name":"Cloud token","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801059b3549339b1052ffa9740042","name":"Eddystone URL","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801049b3549339b1052ffa9740042","name":"Connection param characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801029b3549339b1052ffa9740042","name":"Advertising param characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6801019b3549339b1052ffa9740042","name":"Device name characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6802069b3549339b1052ffa9740042","name":"Environmental:Configuration characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false},{"uuid":"ef6802059b3549339b1052ffa9740042","name":"Environmental:Color charateristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6802049b3549339b1052ffa9740042","name":"Environmental:Gas (Air Quality) characteristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6802039b3549339b1052ffa9740042","name":"Environmental:Humidity characteristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6802029b3549339b1052ffa9740042","name":"Environmental:Pressure characteristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6802019b3549339b1052ffa9740042","name":"Environmental:Temperature characteristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"write [... cut ...]  characteristic","type":"(Custom Type)","notifiable":true,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6805029b3549339b1052ffa9740042","name":"Sound:Speaker data characteristic","type":"(Custom Type)","notifiable":false,"readable":false,"writable":false,"writeWithoutResponse":false},{"uuid":"ef6805019b3549339b1052ffa9740042","name":"Sound:Config characteristic","type":"(Custom Type)","notifiable":false,"readable":true,"writable":true,"writeWithoutResponse":false}]}]

saket424, 2020-12-15:
@Paraphraser ,
The one last remaining issue is hhow to prevent the BLE scan checkbox from automatically being checked when the flow is restarted. Any ideas?

Paraphraser, 2020-12-15:
Having trouble importing that flow

[image]

Paraphraser, 2020-12-15:
Perhaps put it on pastebin. Or wrap it in triple backticks, as in

\`\`\`
Paste here
\`\`\`

saket424, 2020-12-15:
Fixed. Waht happens is when the flow is imported, it automatically causes the scan checkbox to be checked which keeps the node in the "missing" state and prevents it to go into the default "disconnected" state until the scan box is unchecked and re-deployed. Each new deploy causess teh scan box to get checked again even though it was previously unchecked.

saket424, 2020-12-15:
@Paraphraser 

https://github.com/CANDY-LINE/node-red-contrib-generic-ble/commit/d4c583cd4565234efda410bdfe672c849ff7fa71

There is now a way to to off scanning inside the flow by issuing an inject request of scanStop to msg.topic since July 2020

I think I will close the issue. Thanks for your work getting bluetooth working in IOTStack dockerized node-red

closed 2020-12-15 by saket424