SensorsIot/IOTstack #204: docker is not being added to groups ??

ithinkido, 2020-12-20:
After a fresh install of raspbian as well as IOTstack I was greeted with thi error
`Error getting docker version. Received permission denied error`
I on a guess I added docker to groups with
`sudo usermod -aG docker $USER`
This seems to have solved the issue

Slyke, 2020-12-21:
This is related to https://github.com/SensorsIot/IOTstack/issues/205 investigating.

Slyke, 2020-12-21:
Hey @ithinkido did this error occur after restarting?

ithinkido, 2020-12-21:
Yes, and i can confirm that docker was not being added to groups.

Slyke, 2020-12-22:
I have a potential fix for this on the `experimental` branch. I want to test it out on a fresh RPi before merging to master in case it breaks anything.

chris2286266, 2020-12-30:
> After a fresh install of Raspbian as well as IOTstack I was greeted with this error
> `Error getting docker version. Received permission denied error`
> On a guess I added docker to groups with
> `sudo usermod -aG docker $USER`

Actually the mentioned command does not add docker to groups, it adds the current user (e.g. pi) to group docker, which solves the issue.

ithinkido, 2021-01-02:
> Actually the mentioned command does not add docker to groups, it adds the current user (e.g. pi) to group docker, which solves the issue.

OK

ukkopahis, 2022-06-23:
I'll assume the long inactivity indicates menu.sh now solves this.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke