SensorsIot/IOTstack #205: Fail to install IOTstack

rrobinet, 2020-12-20:
Hi,
Few days ago I did install IOTstack with no problem.
To day I did start the installation from a clean raspberry installation, and tons of messages are generated while starting the menu, no readable menu is displayed and no image installation is successful

1. First attempt
```
pi@HomePiServer:~/IOTstack $ ./menu.sh
Checking for project update
From https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
User is NOT in 'bluetooth' group
User is NOT in 'docker' group
!! You might experience issues with docker or bluetooth. To fix run: ./menu.sh --run-env-setup
Python Version: 'Python 3.7.3'. Python is up to date.
ruamel.yaml Version: 'Unknown'. ruamel.yaml is outdated.
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-dev is already the newest version (3.7.3-1).
python3-dev set to manually installed.
python3-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ruamel.yaml==0.16.12
  Downloading https://files.pythonhosted.org/packages/7e/39/186f14f3836ac5d2a6a042c8de69988770e8b9abb537610edc429e4914aa/ruamel.yaml-0.16.12-py2.py3-none-any.whl (111kB)
    100% |████████████████████████████████| 112kB 1.7MB/s 
Collecting blessed
  Downloading https://files.pythonhosted.org/packages/88/34/61e670039aefca011b5e6fb1a73de18165ef6d016ac16df423b20d719e64/blessed-1.17.12-py2.py3-none-any.whl (76kB)
    100% |████████████████████████████████| 81kB 3.1MB/s 
Collecting ruamel.yaml.clib>=0.1.2; platform_python_implementation == "CPython" and python_version < "3.9" (from ruamel.yaml==0.16.12)
  Downloading https://www.piwheels.org/simple/ruamel-yaml-clib/ruamel.yaml.clib-0.2.2-cp37-cp37m-linux_armv7l.whl (550kB)
    100% |████████████████████████████████| 552kB 760kB/s 
Requirement already satisfied, skipping upgrade: wcwidth>=0.1.4 in /usr/lib/python3/dist-packages (from blessed) (0.1.7)
Requirement already satisfied, skipping upgrade: six>=1.9.0 in /usr/lib/python3/dist-packages (from blessed) (1.12.0)
Installing collected packages: ruamel.yaml.clib, ruamel.yaml, blessed
Successfully installed blessed-1.17.12 ruamel.yaml-0.16.12 ruamel.yaml.clib-0.2.2
Please enter sudo pasword if prompted
Docker not installed
Setting up environment:
User is NOT in 'bluetooth' group. Adding:
sudo usermod -G bluetooth -a pi
You will need to restart your system before the changes take effect.
User is NOT in 'docker' group. Adding:
sudo usermod -G docker -a pi
You will need to restart your system before the changes take effect.
usermod: group 'docker' does not exist
Install Docker
# Executing docker install script, commit: 3d8fe77c2c46c5b7571f94b42793905e5b3e42e4
+ sh -c apt-get update -qq >/dev/null
+ sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
+ sh -c curl -fsSL "https://download.docker.com/linux/raspbian/gpg" | apt-key add -qq - >/dev/null
Warning: apt-key output should not be parsed (stdout is not a terminal)
+ sh -c echo "deb [arch=armhf] https://download.docker.com/linux/raspbian buster stable" > /etc/apt/sources.list.d/docker.list
+ sh -c apt-get update -qq >/dev/null
+ [ -n  ]
+ sh -c apt-get install -y -qq --no-install-recommends docker-ce >/dev/null
+ sh -c docker version
Client: Docker Engine - Community
 Version:           20.10.1
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        831ebea
 Built:             Tue Dec 15 04:35:27 2020
[... 195 lines cut ...]
                                                                          ║      [Right] for options for containers that support them                      ║                                                                          
                                                                          ║      [Tab] Expand or collapse build menu size                                  ║                                                                          
                                                                          ║      [H] Show/hide this text                                                   ║                                                                          
                                                                          ║      [Enter] to begin build                                                    ║                                                                          
                                                                          ║      [Escape] to cancel build                                                  ║                                                                          
                                                                          ║                                                                                ║                                                                          
                                                                          ║                                                                                ║                                                                          
                                                                          ║                                                                                ║                                                                          
                                                                          ╚════════════════════════════════════════════════════════════════════════════════╝                                                                          
                                                                                                                                                                                                                                      
                                                                                                                                                                                                                                      
                                                                                                                                                                                                                                      
                                            ╔══════ Build Issues ═══════════════════════════════════════════════════════════════════════════════════════════════════════════════════════╗                                             
                                            ║                                                                                                                                           ║                                             
                                            ║ nodered (fileIssues)            - ['./services/nodered/addons_list.yml does not exist. Build addons file to fix.']                        ║                                             
                                            ║                                                                                                                                           ║                                             
                                            ╚═══════════════════════════════════════════════════════════════════════════════════════════════════════════════════════════════════════════╝                                             

and finally 
 docker-compose up -d   
WARNING: Some networks were defined but are not used by any service: nextcloud_internal, vpn_nw, iotstack_nw_internal
ERROR: Couldn't connect to Docker daemon at http+docker://localhost - is it running?

If it's at a non-standard location, specify the URL with the DOCKER_HOST environment variable.
pi@HomePiServer:~/IOTstack $

ithinkido, 2020-12-20:
There are some issue with the new IOT stack
If you need to use the old menu, run these commands in the meantime
```
cd ~/IOTstack
git pull origin master
git checkout
./menu.sh

```

Slyke, 2020-12-21:
Hey @rrobinet it appears there's a few different issues. First of all, it appears that the `docker` group isn't being created. I'll need to investigate this. I usually test with a fresh instance of Raspberry Pi OS. Can you check if the `docker` and `bluetooth` groups exist? You can run `sudo cat /etc/group` to view all groups on your system. The docker installation script should create this group automatically. You can also try running: `sudo bash menu.sh --run-env-setup` and exiting once it finishes its thing and see if that fixes it.

Secondly, the weird ASCII characters you see are from not having enough horizontal space to render the menu correctly, or enough vertical space to render the help and issues text.

Slyke, 2020-12-21:
Adding the user to the `docker` group is the last thing the installation script does: https://github.com/SensorsIot/IOTstack/blob/master/install.sh#L306 The only 2 reasons I can think of that the user isn't being added is either the script is exiting early, or the group doesn't exist.

rrobinet, 2020-12-21:
@Slyke 
Yes after point 3. the docker issue solved, however still not solve the installation
Using the "old menu" or starting the installation from (after deleting the IOTstack directory)
 `git clone https://github.com/gcgarner/IOTstack.git ~/IOTstack`
Allow me to access a displayable menu, however I still have 2 new problems
1. While trying to install docker I go to an endless loop requested to reboot the system, that doesn't help
2. While composing docker with noder-red the program is looping on:
```
  found 7 vulnerabilities (4 low, 3 moderate)
  run `npm audit fix` to fix them, or `npm audit` for details
```
`npm audit fix `the problem
So for;
 point 1. I suppose that the some rubbish is still installed from  previous installations, so I will start again with a new Raspberry installation and using the code from gcgarner (which I suppose is the same as the so called old one")
point 2. I suppose that there is an issue with the latest update of node-red, maybe a previous version will help

I will let you know the results
Thanks for your support
Robert

rrobinet, 2020-12-21:
@Slyke @ithinkido 
After fresh raspberry installation and as expected everything works file with the version
`git clone https://github.com/gcgarner/IOTstack.git ~/IOTstack `
Endless message regarding to reboot the system while installing docker seems to be normal, when docker is already installed :-$ (no warning that it is already installed)
mariadb, mosquitto and openhab are installed without problems
node-red still generates vulnerability issues that can't be solved with `npm audit fix`, but no endless loop anymore and the installation is successful, so nothing to do with IOTstack.
So I believe there is a serious bug in the "?new" SenorsIot version.
At least I can continue my installation now.
 Regards
Robert

Slyke, 2020-12-22:
I have a potential fix for the groups issue on the `experimental` branch. I want to test it out on a fresh RPi before merging to `master` in case it breaks anything.

Yeah, the nodered vulnerability are from nodered, or some installed dependency itself. The script installs the latest version of the dependency that's publish to npm. We need the author to update their dependency before it'll remove those warnings on ours.

itsslamdunx, 2020-12-24:
Somewhat similarly, the new menu writes a docker-compose.yml that throws an error if you choose prometheus. It leaves in a blank "depends-on:" which previously would have been populated by   
    - cadvisor-arm
    - node-exporter
Leads to "NoneType error".
Resolved with `git checkout new-menu`.

itsslamdunx, 2020-12-24:
Also I see that docker-compose.yml has the following added to most services.
```
    logging:
      options:
        max-size: "5m"
        max-file: "3"
```
This needs to be removed for grafana, mariadb to be installed.
For home_assistant the value for ports: must be removed because it conflicts with network_mode: host.

Slyke, 2020-12-24:
Looks like prometheus' template file had nothing there for `depends_on`, so it was removed. Will have this pushed in an hour or so.

Slyke, 2020-12-24:
@itsslamdunx I've been installing grafana and mariadb with logging enabled. Do you have journald logging enabled? I recently updated the [Getting Started](https://sensorsiot.github.io/IOTstack/Getting-Started/#logging-journald-errors) guide on how to fix this.

The ports were intentionally left in, even though network mode is `host`. The reason is that there is no way for the menu to know which port the service would bind to, and so it can't determine if there'll be a conflict or not.

Slyke, 2020-12-24:
Another thing is that prometheus has not been reported to the new menu system, since it was recently added. At the moment, it will just inject the service template file into docker compose. It will not create any volumes or service directories, since it doesn't have a python `build.py` file. This goes for `dashmachine`, `heimdall` and `homer` too.

alberto11651, 2020-12-28:
Same issue here. Im not able to install it on a fresh RPi. 
Have you get a chance to test the experimental branch and push the changes to the main branch?

rrobinet, 2021-01-04:
Well I started all over and over again and here are my findings
1. Automatic installation: `curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash`
a. Menu claims that Python is not at level; however installed version is 3.7.3 which is above 3.6.9
```
   $python3 --version
    Python 3.7.3
```
b. After pseudo upgrade and reboot it looks OK beside  a funny menu that doesn't fit a terminal window (see my first dump)
c. For info: Group looks OK
```
   $ groups
     pi adm dialout cdrom sudo audio video plugdev games users input netdev bluetooth lpadmin **docker** gpio i2c spi
   $ sudo groups
     root lpadmin
```

2. Manual installation: `git clone https://github.com/SensorsIot/IOTstack.git IOTstack`
a. As above menu claims that Python is not at level however installed version is 3.7.3 which is above 3.6.9
```
   $python3 --version
    Python 3.7.3
```
b. After pseudo upgrade, Running `./menu.sh`; always error message on Docker group and menu says Docker not installed 
c. For info Group 
```
   $ groups
     pi adm dialout cdrom sudo audio video plugdev games users input netdev bluetooth lpadmin gpio i2c spi
   $ sudo groups
     root lpadmin **docker**
   $ docker version -f "{{.Server.Version}}"
     Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/    v1.24/version: dial unix /var/run/docker.sock: connect: permission denied

```
d. So because docker is in the super user group it need to be ran as `sudo ./menu.sh` this looks now OK beside  a funny menu that doesn't fit a terminal window as above

3. My conclusions
IOTstack is a nice start, if the current funny stuff are/will be bypassed, however if other applications (e.g.: samba, minidlna, ...) not included to the stack menu need to be added manually to the docker-compose file, menu doesn't allow to append new features other than the ones selected  and the docker-compose file is overwritten it  (may be a tip for an newer version )
4. Finally, to keep it nice and clean after the current standard installation is; do:
```
     $ git pull origin master
     $ git checkout old-menu
```
and then from the menu do an "Update IOTstak"

   Gives a nice menu for which I can't see the difference with the funny one and is a good basis for future docker add-ons, but to be done interactively.

Paraphraser, 2021-01-04:
With all possible respect given to Slyke's work with the new menu, I'm afraid that I completely agree.

There's a lot of chatter on Discord (a bit less in Issues), which mostly winds up disabling or overriding things done by the new menu. Personally, I find the changes to networking and logging particularly troubling. They are the root cause of many problems. I turned all that off in my own experimentation before concluding that the new menu was not where I wanted to be - yet.

If you want a quiet life, old-menu is definitely the way to go.

I've documented how I install IOTstack [in a gist](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#scripting-iotstack-installations). To summarise:

```
$ sudo apt install -y git curl
$ git clone -b old-menu https://github.com/SensorsIot/IOTstack.git ~/IOTstack 
$ curl -fsSL https://get.docker.com | sh
$ sudo usermod -G docker -a $USER
$ sudo usermod -G bluetooth -a $USER
$ sudo pip3 install docker-compose
$ sudo reboot
```

In words:

* Make sure git is installed (does nothing if it is)
* check out the old-menu branch from this repository
* install docker (the same way the menu does it)
* add user "pi" to the required groups (the same way the menu does it)
* install the up-to-date version of docker-compose (so we can use current features)
* reboot (needed to make the `usermod` commands stick).

Once you've done that, you can either restore a backup and bring up your stack or run the menu and build your stack.

I also recommend treating the menu (old or new) as a one-shot. Once you've set up your stack, **learn how it works** and maintain it from the command line. Using the menu (old or new) to maintain your stack is a fool's errand. Forget the compose-override file. Just edit `docker-compose.yml` as and when you need to and get on with your life.

The gist explains also how to [uninstall docker and docker-compose](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#-uninstalling-docker-and-docker-compose-). It's quicker than starting over with a clean Raspbian image.