SensorsIot/IOTstack #206: Docker not installed

Iftikhar-Omar, 2020-12-21:
curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash
IOTstack Installation
Enter in the sudo password when prompted, to install dependencies
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
Cloning into 'IOTstack'...
remote: Enumerating objects: 167, done.
remote: Counting objects: 100% (167/167), done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 3752 (delta 75), reused 129 (delta 55), pack-reused 3585
Receiving objects: 100% (3752/3752), 1.52 MiB | 947.00 KiB/s, done.
Resolving deltas: 100% (2253/2253), done.
IOTstack cloned
Python Version: 'Python 3.7.3'. Python is up to date.
Docker not installed
Setting up environment:
User is NOT in 'bluetooth' group. Adding:
sudo usermod -G bluetooth -a pi
User is NOT in 'docker' group. Adding:
sudo usermod -G docker -a pi
usermod: group 'docker' does not exist
Install Docker
\# Executing docker install script, commit: 3d8fe77c2c46c5b7571f94b42793905e5b3e42e4
\+ sudo -E sh -c apt-get update -qq >/dev/null
\+ sudo -E sh -c DEBIAN_FRONTEND=noninteractive apt-get install -y -qq apt-transport-https ca-certificates curl >/dev/null
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
usermod: group 'docker' does not exist
Install docker-compose
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?

You should now restart your system
Setting up environment:
User is NOT in 'docker' group. Adding:
sudo usermod -G docker -a pi
You will need to restart your system before the changes take effect.
usermod: group 'docker' does not exist
User is NOT in 'docker' group
!! You might experience issues with docker or bluetooth. To fix run: ./menu.sh --run-env-setup

Iftikhar-Omar, 2020-12-21:
it was solved with:
1. Run: curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash
2. Reboot
3. Run: cd ~/IOTstack && ./menu.sh
4. Reboot
3. Run: sudo ./menu.sh --run-env-setup

And of course agree occasionally when something is required to install

closed 2020-12-21 by Iftikhar-Omar

Slyke, 2020-12-21:
Just for reference, I think this was due to apt not correctly removing a lock file:
```
E: Could not get lock /var/lib/dpkg/lock-frontend - open (11: Resource temporarily unavailable)
E: Unable to acquire the dpkg frontend lock (/var/lib/dpkg/lock-frontend), is another process using it?
```