SensorsIot/IOTstack #210: Heimdall.env and homer.env no default files, must mkdir and touch them manually, so docker-compose up -d werks 😥

StohanzlMart, 2020-12-23:


Slyke, 2020-12-24:
Hey @StohanzlMart that's correct. These services were recently added, after the new menu was completed. None of the new menu logic has been ported to these services, so while they'll be added to the menu itself, no logic will be triggered for them until they have a `build.py` file.

kkoletz, 2021-01-21:
Hi @StohanzlMart 
How should we proceed in that case?

Thank you!

Paraphraser, 2021-01-21:
The quickest thing to do to get going is:

## Heimdall

```
$ cd ~/IOTstack
$ mkdir -p ./services/heimdall
$ cp ./.templates/heimdall/* ./services/heimdall
$ docker-compose up -d
```

## Homer

```
$ cd ~/IOTstack
$ mkdir -p ./services/homer
$ cp ./.templates/homer/* ./services/homer
$ docker-compose up -d
```

In each case, the "mkdir" is non-destructive.

The relevant folders in `.templates` only contain a `service.yml` plus the missing environment file so even if a `service.yml` is already in place in the relevant subfolder of `./services`, it'll just re-copy the file.

Paraphraser, 2021-01-21:
I've just submitted Pull Requests for both Heimdall and Homer. A few hours ago I also submitted a related Pull Request for Domoticz:

* Domoticz [PR249](https://github.com/SensorsIot/IOTstack/pull/249)
* Heimdall [PR250](https://github.com/SensorsIot/IOTstack/pull/250)
* Homer [PR251](https://github.com/SensorsIot/IOTstack/pull/251)

Now we just have to wait for those Pull Requests to be approved and make it into the master branch.

There are some instructions on how to apply pending pull requests at [this gist](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-). You don't **have** to follow those instructions. I'm just pointing out there's an option if you want to apply these Pull Requests to your local IOTstack before they get approved.

If you just want to apply one of those (eg Heimdall), do:

```
$ cd ~/IOTstack
$ git checkout master
$ git pull origin master
$ git fetch origin pull/250/head:PR250
$ git checkout PR250
```

If you want to apply all three:

```
$ cd ~/IOTstack
$ git checkout master
$ git pull origin master
$ git fetch origin pull/249/head:PR249
$ git fetch origin pull/250/head:PR250
$ git fetch origin pull/251/head:PR251
$ git checkout -b merge-pull-requests 
$ git merge -m "local merge branch" PR249
$ git merge -m "local merge branch" PR250
$ git merge -m "local merge branch" PR251
```

Then you can re-run the menu. The gist also explains how to undo all this once these PRs make it into the master branch (assuming they're approved, that is).

If you've already done the earlier work of copying files out of the template, that doesn't matter. You can do this as well. All that happens is that:

1. All of the environment variables that **were** in the `.env` files are now in the `service.yml` files;
2. References to the environment files that were in the various `service.yml` files have been removed; and
3. The `.env` files in the **templates** are deleted.

I emphasised **templates**. For example:

```
~/IOTstack/.templates/heimdall/
```

These pending PRs won't affect anything in a **services** directory. For example:

```
~/IOTstack/services/heimdall/
```

Once you have applied these pending PRs and run the menu, the changes to the various `service.yml` files should come through to the services directories. For example, this file:
	
```
~/IOTstack/.templates/heimdall/service.yml
```
	
gets copied to:
	
```
~/IOTstack/services/heimdall/service.yml
```

and, from there, into `docker-compose.yml`.

Any `.env` file already inside a services directory, example:

```
~/IOTstack/services/heimdall/heimdall.env
```

won't go anywhere but it also won't do anything.

It's up to you whether you want to remove such a redundant file. If it were my system, I would remove it, to avoid confusing myself down the track when I tried to modify an environment variable in a file that wasn't doing anything.

kkoletz, 2021-01-22:
@Paraphraser Thank you very much. It worked like a charm!

closed 2022-09-11 by Slyke