SensorsIot/IOTstack #211: Mosquitto is not working on fresh install

ingopmeyer, 2020-12-24:
Hi, I've installed a fresh IOTStack on a fresh Rasdberry OS installation.
First I got many strange errors, but as on issue #205 the old menu semms to be working
with "git checkout old-menu"
After that I could build and start the stack.
I installed mosquitto, openhab, portainer, influxdb and grafana.

But the mosquitto is not working. There are no logs, also not after enabling the file log.
I cannot connect to the console using portainer. It opens and directly close.
And mosquitto is not working at all.
As I got no error message and no other chance to see what is going on, I'm stuck now.
As I'm beginner in this world I have no idea what to do next.
Any help is highly appreciated.

Thanks and have a nice christmas,
~Ingo

Paraphraser, 2020-12-24:
My guess is it will be a permissions issue. Mosquitto (the Docker container and the process) is special in that it does not run as root (it runs as user ID 1883). If you run `docker ps` several times, you'll probably see from the pattern that it is constantly restarting the container - and that explains why you can't see anything from Portainer because it tries to connect to the container which is a moving target.

Assuming my diagnosis is correct ...

You said you were on old-menu. Good choice. So am I. The problem will be how much got set up and how much got done in the new menu and can be assumed to be available in the old menu.

You could **try** this:

```
$ cd ~/IOTstack
$ docker-compose down
$ ./services/mosquitto/directoryfix.sh
$ docker-compose up -d
```

If the third command won't work because the file isn't there, it's an indication that something is seriously hosed. If you like living dangerously you could also try

```
$ ./.templates/mosquitto/directoryfix.sh
```

but, in all honesty, you're probably better off starting over like this:

```
$ cd ~/IOTstack
$ docker-compose down
$ cd ..
$ mv IOTstack IOTstack.failed
$ git clone -b old-menu https://github.com/SensorsIot/IOTstack.git IOTstack
$ cd IOTstack
$ ./menu.sh
```

Your stack running, albeit without Mosquitto, implies docker and docker-compose installed so you can probably skip that step and move to building the stack.

The reason for keeping the old folder around is in case there are any files or folders you need to use to speed up the recovery. Just don't use anything from Mosquitto.

Hope this helps.

Slyke, 2020-12-24:
New menu now applies permissions from inside the `build.py` file: https://github.com/SensorsIot/IOTstack/blob/master/.templates/mosquitto/build.py#L115

@ingopmeyer are you getting these prompts?
```
Need to set owner on mosquitto directories with command: chown -R 1883:1883 MOSQUITTO_SERVICE_PATH
Set user 1883 on MOSQUITTO_SERVICE_PATH (Y/n):
sudo chown -R 1883:1883 MOSQUITTO_SERVICE_PATH
```

But I added back the `directoryfix.sh` for now so that those commands will work.

As you can see from the code, it'll only run the directory fix logic if it needs to create any files or folders for mosquitto. I should add in an option in mosquitto's menu that'll also run the permission fixing logic.

ingopmeyer, 2020-12-25:
Hi,

first many thanks for your help. Unfortunately I'm totaly stuck now. I tried several thing and both of your ideas, the old and new menu. As far as I could find out, mosquitto is still not working, same problem.
Running "./services/mosquitto/directoryfix.sh" gives something like "...found user 1883" and no error.
With the latest IOTStack I got the prompt @Slyke mentioned about the chown command and I answered with yes.
But now I stuck at another problem: I could not login into portainer as it is not asking for new admin password on fresh install.
I always deleted the OITstack directory on each try.
With the old-menu portainer-ce is not coming up, no web UI, so I startet again with old-menu and portainer. And totally strange after I checked out a fresh clone of IOTstack, the portainer-ce volume directory was there. I deleted manually but no help.
Semms that git is adding something on fresh clone of IOTStack, the volumes folder is always there with portainer password file.
When I clone into another directory then IOTstack i cannot start docker as it gives a lot of errors:
Creating grafana   ... error
in use by container "14cd65fe7cfb3e32d858d4de87c54f1a6d9376d9c8afbe27b55476f6af4838de". You have to remove (or rename) tha...

So either git is messed up or docker or both. Maybe I need a fresh reasberry installation...
I will try that and report.

ingopmeyer, 2020-12-25:
Oh, I found out that when restarting the rpi after deleting the folder IOTstack it is back with ALL volumes of the containers I ever used. So are "portainer" and "portainer-ce" with the portainer.key files. Wtf, from where do the coming from?? I already removed docker totally from the system.

Happy christmas
~Ingo

ingopmeyer, 2020-12-25:
Ok, the above nonesense came because I missed a "docker-compose down" and removed the directory. This is amost not recoverable.
Starting with a fresh rasbian OS I can say that mosquito is not running in either the new or old menu.
I have done nothing but:
`
git clone -b old-menu https://github.com/SensorsIot/IOTstack.git IOTstack
// or -> git clone https://github.com/SensorsIot/IOTstack.git IOTstack
cd ~/IOTstack
./menu.sh
docker-compose up -d
`
Btw, with the old-menu portainer-ce is not running,
with the new-menu openhab:latest is not running, which I assume is the 3.0.0
OH3 is the reason why I try this at all and exactly what I want in the end.

So any other ideas?

Thanks a lot
~Ingo

Paraphraser, 2020-12-26:
Hi,

I can't see what you're seeing (obviously) so I'm limited in the conclusions I can draw but when a container restarts because of a configuration problem, the [up, fault, restart] iteration time is usually in the order of a few seconds, possibly a minute, but rarely much longer. When I see a 28-minute gap I think "something else is going on."

Next point. The good thing about containers is that, when a container terminates and gets rebuilt, it's always a clean slate. The bad thing is that a clean slate implies loss of logs. In other words, for a log to not be there is expected, not unusual. About the most you can do is:

```
$ docker logs -f «container»
```

Then you'll (mostly) get the last log entry before the container terminates and starts over. I do remember at one point having a problem that I couldn't track down with `docker logs` so I figured out where the process running inside the container was writing its logs and added that folder as a mapping into the volumes folder in `docker-compose.yml`. That made the log persistent. It was over a year ago and the details are sketchy in what passes for my memory but I think it was Node-Red and had something to do with one of the flows I was trying to import.

Having said all that, if there's no other indication in Mosquitto's log than what you've included here, I don't see much future in chasing the log. I come back to 28 minutes being "unusual" all by itself and want to look for other reasons.

<hr>

I thought I'd try to attack the problem from a different direction by doing a clean build of my own, satisfying myself that it was working, and then sharing my results. That'd give you a "known good" reference.

Basics:

* Raspberry Pi 4B, 4GB
* Samsung T5 500GB SSD
* 2020-12-02-raspios-buster-armhf.zip

The RPi4 is new but it **has** gone through the steps of burning the above image to an SD, booting from that SD, a full upgrade, and making sure everything was set up to boot from SSD.

It has also been running from an identical Samsung T5 500GB SSD with IOTstack old-menu installed, plus a restore of a backup from my "live" Raspberry Pi 4.

For this exercise, I took a brand new Samsung T5 500GB SSD out of its box, imaged that, connected it to the RPi and applied power (no SD in the slot).

A complete blow-by-blow of the installation is on [pastebin.com/VtaF9pYd](https://pastebin.com/VtaF9pYd).

Probably the most significant difference is that I did not include "openhab", mainly because I don't run it normally, have no idea what it is, wouldn't be able to say whether it was working or not, wouldn't be able to fix it if it wasn't and, if it ultimately proves to be the culprit in whatever ails your system, I wouldn't have any means of reaching such a conclusion.

The end-point of the Pastebin is that I was about to fire up the browser to connect to Portainer-CE. I did that and saw:

[image]

which, to me, equals "working".

However, I did not login. Some time later (after preparing the Paste and as I was typing the above, I did another `docker ps` and saw that Portainer-CE had a created time of "about an hour ago" but a status of "up 15 minutes". The logical conclusion is that it was restarting. I **assumed** that that might be because I had not created an admin account so I did that, then I went back to check the logs and found:

```
2020/12/26 00:33:32 No administrator account was created after 5 min. Shutting down the Portainer instance for security reasons.
```

So "good guess" on my part!

I don't normally use `docker ps` as-is because it's far too wordy. I have this alias defined:

```
alias DPS='docker ps --format "table {{.Names}}\t{{.RunningFor}}\t{{.Status}}"'
```

the output from which is:

```
NAMES          CREATED       STATUS
portainer-ce   2 hours ago   Up 31 minutes
nodered        2 hours ago   Up 2 hours (healthy)
influxdb       2 hours ago   Up 2 hours
grafana        2 hours ago   Up 2 hours
mosquitto      2 hours ago   Up 2 hours
```

So, other than Portainer-CE where the restart is explained, everything else is as happy as the proverbial Larry.

The Paste also has proof that mosquitto (both the container and the service) is working so, basically, I ain't seein' what you're seein'. 

<hr> 

One final point. Yes, I can imagine a container having conniptions if you rip out its persistent store while it is running. **BUT**, under the old-menu at least, recovery is always straightforward. There are only two patterns:

1. Where `directoryfix.sh` is needed to preset the structures.
2. Where `directoryfix.sh` is not needed.

How do you know? Answer: look in the container's folder in `~/IOTstack/services` to see if a `directoryfix.sh` is present. Resetting any «container» to scratch is:

```
$ cd ~/IOTstack
$ docker-compose stop «container»
$ docker-compose rm -f «container»
$ sudo rm -rf ./volumes/«container»
$ ./services/«container»/directoryfix.sh
$ docker-compose up -d «container»
```

with the running of `directoryfix.sh` dependent on whether it exists.

The `stop` and `rm -f` commands are what happens if you run `docker-compose down`. A `down` applies to all containers whereas `stop` then `rm -f` means you can be selective.

Basically, if the persistent storage area in the `volumes` directory needs to be set up a special way then `directoryfix.sh` does that. Everything else gets initialised by `docker-compose` as it processes the left hand sides of the `volumes` statements. Then, as each container comes up it finds (generally) empty folders and populates them with defaults.

Paraphraser, 2020-12-26:
As an aside to Monsieur @Slyke, if I understand you correctly, the functionality of `directoryfix.sh` files is now embedded in `build.py` files - yes?

The fact that `directoryfix.sh` files went away and came back implies that you think of `directoryfix.sh` files as deprecated - yes?

Under old-menu, a user could get a new clean slate for any «container» by:

```
$ cd ~/IOTstack
$ docker-compose stop «container»
$ docker-compose rm -f «container»
$ sudo rm -rf ./volumes/«container»
$ [ -e services/«container»/directoryfix.sh ] && ./services/«container»/directoryfix.sh
$ docker-compose up -d «container»
```

Question. Can any user to replace the `directoryfix` step with:

```
$ ./.templates/«container»/build.py
```

Unless I'm missing something, my casual experimentation (with the mosquitto container) suggests that, right now, the answer is "no".

I see this as a problem. I don't think it's reasonable to expect the average user to drill into build.py files to see whether they are doing any directoryfix-like steps and interpolate from the Python to shell commands.

Absent an easy "clean slate" process for a container, I think what this is likely to mean in practice is that "start from a clean Raspberry Pi image" is increasingly going to become the default solution for any problem. Opinions may differ but I don't think that's either user-friendly or particularly good for IOTstack's reputation as a go-to solution for IoT support.

I don't see that it's particularly useful (in maintenance terms) to have both `build.py` and `directoryfix.sh` for a given container doing the same thing. My 0.2¢ is that there are two obvious solutions:

1. `build.py` should invoke `directoryfix.sh` if it exists but `build.py` should never replicate any of the functionality of `directoryfix.sh`. In other words, any user should be free to run a container's `directoryfix.sh` to establish "clean slate" state, knowing that it is exactly what `menu.sh` does on a first install. 

2. Running `build.py` without any arguments should establish "clean state" state for the container (ie either functionally identical to what was previously done in `directoryfix.sh`, or a no-op where `directoryfix.sh` wasn't required). We could then just include "run `build.py`" in the steps for getting a clean slate.

Thoughts?

> As an alternative to option 2, maybe running a `build.py` without arguments should chuck up "usage" text and exit. The usage might then include a hint for something like a `--initialize-persistent-storage` option.

ingopmeyer, 2020-12-27:
Hi,

very cool help. thanks you very much!
At the end I'm not sure what it was. I got it running with the new menu and portainer-ce and switching the log of the mosquitto to filebased, then stdout, then back to filebased. And then it worked. Strange, but so are computers...

Your help was really cool, at least also to learn a lot about that project.

I'm now a far step further to get my Openhab 3 instance running.
The next problem comes in a new issue to not mess this up.

~Ingo

closed 2020-12-27 by ingopmeyer