SensorsIot/IOTstack #213: "Install log2ram" does not work from the menu

fcarella, 2020-12-24:
line 46 in  misc_commands.py is
subprocess.call("bash disable_swap.sh", shell=True)

doesnt look right, I think should be
subprocess.call("./scripts/disable_swap.sh ...", shell=True)

with some argument ..., not sure what though

Slyke, 2020-12-24:
This is probably from your other bug https://github.com/SensorsIot/IOTstack/issues/212 Give it a try now, and let me know if you still have troubles with it.

fcarella, 2020-12-24:
Thanks for the quick responses.  Will try out fixes when I get a chance.

fcarella, 2020-12-25:
Still doesnt work.

line 46 in misc_commands.py is
  subprocess.call("bash disable_swap.sh", shell=True)

I still think the syntax is wrong but even so, if you look at "disable_swap.sh", theres nothing in there about log2ram.
The old IOTstack code does this (see line 410 @ https://github.com/gcgarner/IOTstack/blob/master/menu.sh):
...
git clone https://github.com/azlux/log2ram.git ~/log2ram
chmod +x ~/log2ram/install.sh
...

Hope this helps...

Slyke, 2020-12-25:
Hey @fcarella , Looks like I forgot to actually commit the file and rename the command. Copy paste mistake! I just pushed another fix, and tested that begins to install it (but have not yet tested if install completes since I don't have access to a RPi right now).

Slyke, 2020-12-25:
You can view the changes in this file: https://github.com/SensorsIot/IOTstack/commit/123f35cffcbf72ba63ec654abf5007c9946301a8, I'll close this issue, but please recomment if you continue to have issues with it.

closed 2020-12-25 by Slyke

fcarella, 2020-12-25:
> You can view the changes in this file: 123f35c

thanks.  i'll try this out, let you know if there's any probs.  And Merry
Christmas!