SensorsIot/IOTstack #214: Hassio changing Docker LogingDriver - Can not build up my stack

Padm59, 2020-12-25:
Hey,

I installed Hassio normally and then I tried to install the other containers with docker-compose. Unfortunately I get the Error:

ERROR: for mosquitto  Cannot create container for service mosquitto: unknown log opt 'max-size' for journald log driver

ERROR: for nodered  Cannot create container for service nodered: unknown log opt 'max-size' for journald log driver

After a quick research I tried to change the logging Driver to the original default json-file. Hassio however didn't like that at all and wanted me to change it back. 

In the Docker-compose.yml was this created:

logging:
      options:
        max-size: "5m"
        max-file: "3"

this is leading to the error because journald log driver doesn't support this options.

Please fix the Stack Builder so that there are no Errors occurring in connection with hassio.

scorpinus, 2021-01-01:
Having the exact same issue

breemts, 2021-01-04:
I kept the journald and set max sizes in the /etc/systemd/journald.conf
```
SystemMaxUse=100m 
SystemMaxFileSize=5m 
RuntimeMaxUse=100m 
RuntimeMaxFileSize=5m 
```
according to [this wiki](https://wiki.archlinux.org/index.php/Systemd/Journal#Journal_size_limit) more options [here](https://www.freedesktop.org/software/systemd/man/journald.conf.html)

I couldn't figure out how to remove the logging options with the [compose-override.yml](https://sensorsiot.github.io/IOTstack/Custom/). So I removed the lines manually in docker-compose.yml. 
Anyone any Idea how to do that, so the change will persist the next update?

Slyke, 2021-01-05:
Hello,

You can update your docker logging driver by following the commands in the Getting Started Guide: https://sensorsiot.github.io/IOTstack/Getting-Started/#starting-your-iotstack

You can also enforce no logging by removing the logging entries in the docker-compose file, by using docker compose overrides for each service (see: https://sensorsiot.github.io/IOTstack/Custom/).

I have a solution for toggling logging on and off in the menu for each service.

ukkopahis, 2022-06-23:
Native/supervised Home Assistant is no longer supported by IOTstack, marking issue to be closed.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke