SensorsIot/IOTstack #215: nodered (fileIssues)            - ['./services/nodered/addons_list.yml does not exist. Build addons file to fix.']

alex1362, 2020-12-25:


a67878813, 2020-12-27:
I follow standard install note and found an ERROR when doing "docker-compose up -d"
        -Building nodered
        -ERROR: Cannot locate specified Dockerfile: Dockerfile

Paraphraser, 2020-12-27:
Re-run the menu. Use down arrow until the cursor is on top of Node Red then press the right arrow to go into the options. Accept the defaults and complete the steps. This creates the Dockerfile, after which another `docker-compose up -d` should work.

You can, if you wish, choose other nodes (ie more than the defaults). The two you should consider are SQLite and Bluetooth. If you **need** either of those, they **must** be built by the Dockerfile process but everything else is optional.

Personally, I prefer to limit what gets included at Dockerfile time to the defaults and things that must be built by Dockerfile, using Manage Palette in the GUI for everything else. It doesn't really matter how a node gets added, it all gets backed-up (assuming you run backups). Nodes added by Dockerfile are implicitly backed up when the Dockerfile is backed up. Nodes added via Manage Palette are in the persistent store in the volumes directory and get backed up from there.

Slyke, 2021-01-05:
I'll close this issue. Please reopen if you have issues with nodered.

closed 2021-01-05 by Slyke