SensorsIot/IOTstack #216: IOTstack installs unsupported home assistant system

Mesyu, 2020-12-27:
I wanted to install the Crownstone integration (Intelligent dimmer/switches) but it seems not available on my system. HA version from 2020.12.7 has been installed on a RPI4/ 4Gb/250Gb Kingspec HD using the latest IOTstack. 
In supervisor/hassio/system under supervisor it messages "you are running an unsupported installation". 
So, I wonder should I better install another version of HA that works with those Crownstone devices?

Bettelmage, 2021-01-02:
That shouldn't be the issue here. The reason your supervisor is showing this is that Home Assistant Supervised is considered an "alternative install".

The official installation methods are "Home Assistant OS", which as the name implies, brings its own OS and is meant for dedicated Home Assistant machines. And "Home Assistant Container" which runs in Docker but limits your possibilities, e.g. you are not able to use AddOns. The second may be an option for you. But in my opinion the "unsupported" supervised install is still the way to go if you want to use Home Assistant along with IOTstack.

More information here: https://www.home-assistant.io/docs/installation/#alternative-installs

Mesyu, 2021-01-02:
Thanks for your support. It makes a few things clear, but still leaves me confused about these systems. I have not found a clear description of the differences, apart from the way they are installed. Afterall, the version installed through the IOTstack also results in a container based HA: portainer shows no less than 7 HA containers! 
Also, the interface of all versions looks the same. 
A dedicated machine is no problem for me, I do't need to install any other non-home-assistant-related applications. I think I'm going to install all three versions on separate SSD cards and see which fits best my needs! 

Kind Regards

Mesyu

+31 625263136

ukkopahis, 2022-06-23:
Native/supervised Home Assistant is no longer supported by IOTstack, marking issue to be closed.

PR #576 (2022-09-11): docs/Changelog: update to reflect latest changes
Update the Changelog page to reflect previous changes and currently open pull-requests.
Preview available at: https://ukkopahis.github.io/IOTstack/Updates/Changelog/
Also add Changlog link to the Wiki homepage.

Other contributors are welcome to make [change suggestions](https://github.com/ukkopahis/IOTstack/edit/doc-changelog/docs/Updates/Changelog.md) to the Changelog after creating a pull request.

Using this pull-request to close old resolved/abandoned issues:
* Empty / Not actionable
  * Closes #374
  * Closes #288
  * Closes #111
* Duplica [... cut]
closed 2022-09-11 by Slyke