SensorsIot/IOTstack #219: RPi4 + native-boot SSD sometimes fails to boot from USB3 port, and WiFi freezes when IOTstack comes up

Paraphraser, 2020-12-29:
I don't think this is really an IOTstack issue as such. I'm describing it here in case anyone else is experiencing problems with an RPi4 + native boot SSD + IOTstack causing odd behaviour like:

* failures to boot from SSD
* WiFi issues as IOTstack comes up.

I don't think IOTstack **causes** either problem, but it may be involved in triggering underlying causes.

<hr>

I have three 4GB RPi4s, named "iot-hub", "sec-dev" and "new-dev".

Each has an SSD attached to a USB3 port. iot-hub has an [OWC](https://eshop.macsales.com/shop/ssd/owc) SSD while the other two have [Samsung](https://www.samsung.com/au/memory-storage/portable-ssd/portable-ssd-t5-2tb-black-mu-pa2t0b-ww/) T5 SSDs.

iot-hub and sec-dev were purchased last year so they boot from SD but run from SSD (built as per [this gist](https://gist.github.com/Paraphraser/55643bd660db46716f6c81266905cae8)). new-dev is a recent purchase and I went through the steps to make it "SSD bootable" but otherwise followed the same gist for add-on packages etc.

All three are Ethernet connected.

All SSDs are bus-powered.

The Raspberry Pi world seems to be full of people who offer "power" as the universal answer to almost any RPi problem. I'd like to address that topic because I'm quite sure "power" isn't the issue:

* I've confirmed (several times) that the **measured** draw going into each RPi4 is well within the spec of the **official power supplies**.
* I've swapped power supplies to see if the problems follow the supply. They don't. The problems follow new-dev.
* I live in country New South Wales, Australia where mains power is not exactly "reliable" (a single line doth not a "grid" make). Everything is behind beefy APC UPSes. The Pis don't "see" brownouts or spikes from the mains. If the mains goes out for an extended period, I do orderly shutdowns and then turn off the UPSes.
* If "power" was the explanation for the observed problems on new-dev, it wouldn't explain why the identical hardware on sec-dev didn't exhibit identical problems.

There **are** software differences:

* iot-hub was built from 2020-05-27-raspios-buster-armhf
* sec-dev was rebuilt from 2020-08-20-raspios-buster-armhf after an `apt upgrade` broke something ([issue reported here](https://github.com/raspberrypi/linux/issues/3922))
* new-dev was built from 2020-12-02-raspios-buster-armhf

In terms of `uname -a`:

```
Linux iot-hub 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Linux sec-dev 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
Linux new-dev 5.10.3-v7l+ #1386 SMP Sun Dec 27 14:05:40 GMT 2020 armv7l GNU/Linux
```

I decided not to risk a repeat of `apt upgrade` breaking iot-hub because that's my "live" system, which is why it's stuck at 5.4.51. But, other than that, the add-on packages are the same and IOTstack is the same.

> When I run-up a new machine, I `git clone` IOTstack (nowadays as the `old-menu` branch), let `menu.sh` install docker and docker-compose, then restore a backup from my live machine, and let it rip. Sure, this fetches current images from Dockerhub but I keep all my machines up-to-date with regular "pulls" so there are rarely going to be any differences. 

Aside from the `apt upgrade` breakage mentioned above, iot-hub and sec-dev have been rock solid since the get-go (over 12 months). The only real issue of note is the WiFi interfaces occasionally going walkabout, the cure for which is documented at [this gist](https://gist.github.com/Paraphraser/305f7c70e798a844d25293d496916e77).

But I am having terrible trouble with new-dev. Frequently (like maybe every 5th reboot) it refuses to boot, even after a power-cycle. When I put the SD back in, boot from that and nose around, the SSD just isn't there. I can't see it in `lsblk` and I can't mount either logical volume.

> Put the SSD in the Mac - up comes the boot partition (macOS doesn't see the ext4 partition - but it's visible in other tools)

I normally use the upper USB3 port (no particular reason). Sometimes, when new-dev refuses to "see" the SSD, moving the SSD to the lower USB3 port will make it come good. Only "sometimes". Definitely not "always". The most reliable way of recovering is to attach the SSD to either one of the USB2 ports. That works every time!

All my RPi4s have [inline power switches](https://core-electronics.com.au/micro-usb-c-in-line-power-switch-cable-for-raspberry-pi-4.html). Any time I'm going to insert/remove the SD, or move the SSD between ports, I turn off power before making the change and turn power back on afterwards. Because of that, I discount possibilities like:

* whether RPi USB is or isn't hot-swappable? It doesn't matter if you never do it!
* whether pulling a USB plug from a hot system might've corrupted the SSD? Leaving to one side that it didn't happen, why would "corruption" prevent booting via the upper USB3 port but not the lower USB3 port (sometimes) or either USB2 port (always)? The SSD is either corrupt or it isn't.
* whether pulling a USB plug from a hot system might "hang" the USB3 interface? If removing and re-applying power would **always** reset the USB3 interface then both ports should always work. Why does the lower port sometimes work? Why do USB2 ports always work? What combination of power-cycles and/or successful boots from the other ports cause the upper USB3 port to become viable again?

It's downright weird.

Now, here's another wrinkle. I mentioned that iot-hub and sec-dev had occasional WiFi issues. With new-dev, I can pretty much kill WiFi simply by:

1. SSD connected to USB3 (either port)
2. Boot (either apply power or a CLI reboot)
3. Bring up IOTstack.

By "kill", I mean that:

* ICMP-echo replies stop arriving in response to `pings` aimed at the WiFi interface from another machine; and
* `ifconfig wlan0` on new-dev has no IP address (either v4 or v6).

The obvious variation on the theme is:

1. RPi4, SSD connected to USB3, IOTstack up
2. Reboot.

The ICMP echo reply pattern is replies come back during the shutdown, and briefly as the machine starts to come back up, but stop pretty much as soon as Docker tries to resume the stack.

> It doesn't matter whether I'm running IOTstack on the master branch or old-menu. Tried both.

If, instead of using `docker-compose up -d`, I bring up one container at a time (eg `docker-compose up -d mosquitto`) then WiFi will generally stay up. But WiFi dies quite reliably if the whole stack comes up at once.

The problem goes away completely if I connect the SSD to USB2. I can reboot with IOTstack up or down, or take the stack up or down while the system is running. WiFi remains up and stable.

Hypothesis: it's a load problem - IOTstack is overloading the Pi. Well:

* I've eliminated "new menu" vs "old menu" (on new-dev) as a possibility so it's not that.
* sec-dev and new-dev are identical hardware so why not identical behaviour under identical load?
* sec-dev and new-dev have different Raspbians, and sec-dev boots from SD and runs from SSD while new-dev boots and runs from SSD. But why does the problem go away if I connect new-dev's SSD to USB2?

Another thing that makes my eyebrows twitch is that the guts of the gist on WiFi issues mentioned earlier is `sudo dhclient wlan0`. On iot-hub and sec-dev, this command returns immediately and has the effect of resetting the interface which immediately comes good. When WiFi has gone walkabout on new-dev, the command takes over a minute to return and doesn't cure whatever has caused WiFi to go away. If it was a "load" issue then surely the command would work once IOTstack was up and stable? Why does bringing containers up one at a time sometimes avoid a WiFi freeze?

I really do not know what to make of all this. For the life of me, I can't see why booting SSD+USB3 = problems while SSD+USB2 does not. The only real difference is somewhat slower access. A timing issue?

I've rebuilt the system three times, starting from a clean image with Balena Etcher. That hasn't helped.

I bought new-dev so I'd have a test vehicle for native-boot SSD (after which I was going to redeploy it in another project). I have no idea whether the new-dev hardware is a lemon, whether SSD native boot isn't all it's meant to be, whether this is a Raspbian difference between 2020-08-20 and 2020-12-02, or what. I don't think it's the Samsung SSD because I've got another one of those too and have already tried a swap: the problems follow new-dev.

I'm going to try re-building new-dev as an SD/SSD combo from 2020-12-02, then from 2020-08-20. I don't want to try moving sec-dev to SSD native boot in case this turns out to be a firmware issue and I wind up with two misbehaving RPi4s.

If anyone reading this has any suggestions for things to try out, I'm all ears!

2020 - a year sent to try our patience! With any luck, the expression "2020 hindsight" will take on a whole new meaning.

Roberto6969, 2020-12-29:
I believe this is not directly related, but just to be careful with latest firmware ieeprom-2020-12-11.bin in combinations with Kingspec SSD. Before upgrade firmware, I had no problems with booting. But keep in mind, the Raspberrys worked fine for a months with only a few boots.
In my case after upgrade two Raspberry Pi 4 (one with IOTstack) with Kingspec SSD won't boot at all. The only solution to boot is to connect SSD disk later after power on (you have to catch the right moment). It seems latest firmware has problems, so better not to upgrade to the latest version. 
Opened issue: https://github.com/raspberrypi/firmware/issues/1515

piwi447, 2020-12-29:
Hello 
I am a newbie to RPI and I experienced similar problem when trying to install IOstack on My RPI4 Model B 4 Gb.
I am not sure I can give you all the details on my configuration so easily.
This RPI was bought in March 2020 and was not able to boot from USB.
My first idea was to run both openhab and codesys on it.
But at that time I didn’t know the docker possibility . So I bought another RPI3 model B+ that would run codesys while RPI4 would run openhab based on openhabian config.
During this Xmas Holiday, I had more time to setup PI4 with docker in order to run openhab with more flexibility and without the need of good Linux knowledge .
( I am more lazy than Andreas Spiess)
For now, I am booting from USB 3.0 Kingston data traveler 16Gb.
Configuration from a Rasbian burned last Friday on SD card and then copied with RPi clone to USB stick.
Then I installed IOstack from menu.
Finally got it running on Monday and use my new config for one day without problem but without rebooting
Sunday evening I try to shutdown to install a new ice tower cooling fan and the problems begun.
I try finalizing my configuration yesterday adding PI VPN .
When trying to backup , my IOT stack never started and all docker commands are stucked without any error message...

The more reboot, the more problems ...

My symptoms are,
reboot very slow... and sometimes did not succeed 
Then reboot most of times don’t work first time, need to power off and then it restart
Sometimes after restart the wifi is not enabled but rebooting a second time helps.

I was thinking that my data traveller could have been damaged due to forced power shutdown.

I discovered yesterday evening that the green light blinking periodically is a kind of Morse code for boot trouble.

What is clear is that I could never get my iostack running again since even « Docker ps » is stucked

trying to remove and install Docker did not help...

So next step is try to boot from USB2
For this I made a shutdown , mode stick to top usb2 and power on.
It started without trouble at first time, this is a confirmation of similar situation to yours.

but my docker config is still wrong...

I will trying install docker on a SD card with a minimum container config (piHole, openhab, nodered, influxDB , grafana ) to be able to get my PI4 running waiting for help from the community.

Paraphraser, 2020-12-29:
I'm going on the information [here](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) but I really don't know what I'm doing.

This is what I see on the three RPis:

```
$ uname -a; vcgencmd bootloader_version; sudo rpi-eeprom-update
Linux iot-hub 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

```
$ uname -a; vcgencmd bootloader_version; sudo rpi-eeprom-update
Linux sec-dev 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

```
$ uname -a; vcgencmd bootloader_version; sudo rpi-eeprom-update
Linux new-dev 5.10.3-v7l+ #1386 SMP Sun Dec 27 14:05:40 GMT 2020 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

While there are subtle formatting differences, they all seem to agree on the version signature and timestamp-in-seconds, as well as the "Current" and "Latest" values.

I've been interpreting that to mean that, even though the three Pis are running different base builds and updates-since, they have the same EEPROM, and that all the updates have been applied, silently, behind the scenes.

Do you think I'm misinterpreting? Should I be looking somewhere else?

glenn3451, 2020-12-29:
i dont know if it was the same problem, but i just now saw this because someone said something on the discord and paraphraser responded with this. anywho... one day after 2 months of running well i noticed my tasmota switches werent working, which led to me discovering the pi was not working. vnc'd in and os was running and responding but not the stack. i rebooted (seemed like the easiest way to restart everything) and found that the pi would not boot off of ssd anymore. i eventually decided to try a fresh sd card instead of usb ssd. booted fine. i then proceeded to rebuild everything this time being sure to backup a whole image to the sd card after i got to the point i liked how things were running. now my suspicions are some sort of corrupt boot partition, but honestly i dont have any expertise in this area with linux. i guess my thought was if all i did was clone an sd card back onto the ssd with a clean image then its probably not hardware related. since the firmware wasnt changed and seemed to boot fine with a new image im assuming this has something to do with a corrupted boot partition

piwi447, 2020-12-30:
Hello, 

yesterday morning 
I changed my firmware to 2020-09-03. As explained by Roberto , Boot was ok but docker was still blocked .
Yesterday, I received my brand new Samsung SSD T7 500 Gb
And also some ubiquity unify access point and one manageable switch .
In the evening I retry to configure my SSD with docker.

my first trial was to copy a well working se card a got with my RPI4 originally .
This is working well on my rpi 3 for month without any boot problem .
This SD card was configured with noobs I think because there is some recovery and setting partition on it ..
Unfortunately, after copy with Rpi-clone, I could not boot but got message saying some trouble with  recovery.
I am new to GitHub too and have to check how to share picture... 
The message said that an upgrade is needed...
But I did not know how to upgrade the SSD without any change on the SD that I wanted to reuse on my RPi3..
So I tried copying boot directory from SD to SSD but that did not work better.
Then I made a full upgrade of the Raspi an on the SD and rpiclone again without more success
Then I installed a rasbian image from scratch directly on the SSD ( very fast compared to the data traveler ) .
It boot without problem directly from SSD and I made the full installation process.
For some reason, the wifi refused to associate to my wifi...it was perfectly working using SD card one our before.
The complete iotstack install was faster ( except the Nodered  nom that is so slow and giving so many error message) .
My docker is working well... and reboot is ok.
Just don’t understand why portainer is installed but don’t know my docker config...
Next status tomorrow .

I need to check how add unify controller container in iostack as it is not in the menu...

Pierre

Paraphraser, 2021-01-15:
For anyone following this issue, on Jan 6 I rebuilt "new-dev" from the ground up, to boot and run from SSD (ie no SD).

I started with the SSD attached to the upper USB3 port. Apart from one reboot during the initial setup which hung and needed a power-cycle, the RPi has performed flawlessly ever since.

> In terms of sequencing, there's the burn of the SSD using BalenaEtcher, then a first boot followed by an apt update & full-upgrade and applying settings via raspi-config, then a reboot. Those were cool. Then installing packages I like to have (eg wget), cloning IOTstack, installing docker and docker-compose, and then another reboot. That's the reboot that hung but it came back after the power-cycle without having to move the SSD from USB3.

Then I restored an IOTstack backup and brought up the stack. Earlier experience was that this was where the WiFi would freeze. WiFi was OK and has stayed OK.

Previous experience was that an apt update / upgrade followed by a reboot while the stack was up was likely to cause a freeze. I've done a two of those in the intervening period without incident but the third one a few moments ago hung and needed two power-cycles plus moving the SSD to USB2 before the RPi came up again.

> Interestingly, the "up times" in `docker ps` suggested the stack had been up for five minutes. The only way that makes sense is if the stack went down cleanly on the "reboot" command, then the machine actually came up with the SSD connected to USB3 but the machine just wasn't responding to pings on either Ethernet or WiFi, then I turned off the power, moved the SSD to USB2, applied power, and waited for the machine to respond to pings.

The OS/EEPROM version situation now is:

```
$ uname -a; vcgencmd bootloader_version; sudo rpi-eeprom-update
Linux new-dev 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

Summary: the OS version has changed from 5.10.3-v7l+ to 5.4.83-v7l+ but the EEPROM remains the same.

> I do not get the OS numbering across the the two posts. The post I made on December 30th was copy-and-paste just like the one here. I didn't invent "5.10.3-v7l+" any more than I invented "5.4.83-v7l+". The numbering looks like the versioning has gone backwards. I have no idea how to explain that.

But, now I'm in the situation where:

* if the SSD is connected to USB3 **and** IOTstack is running when I do a `sudo reboot`, the machine will hang. My guess is the hang is "on the way up". The ping pattern is I get 7 replies after issuing the reboot (which I have always assumed is the time it takes for IOTstack to go down, followed by other shutdown-time housekeeping), then nothing.
* if the SSD is connected to USB2 **and** IOTstack is running when I do a `sudo reboot`, the machine comes back. 7 seconds of ping replies, 30 seconds of no replies, then replies start.
* if the SSD is connected to USB2 **and** the machine is running **and** I "down" IOTstack, `sudo shutdown -h now`, remove power, switch the SSD to USB3, and then apply power, the machine will boot OK. But if I bring up IOTstack and try to reboot while the SSD is on USB3, I'm stuck.

So, something about the latest OS update...

... which hasn't also affected sec-dev which is also at the same OS release.

I'm in two minds about whether to:

1. rebuild from scratch again on the SSD; or
2. revert to the boot SD + run SSD paradigm

Option 1 implies that the initial apt update & full-upgrade will bring the machine to the same version it is at now, before docker and docker-compose are installed. If that works on USB3 then it kinda implies it is something about applying system updates over the top.

Option 2 might (eventually) help isolate whatever is causing this to either the hardware of the new-dev RPi or the boot method, with SD+SSD being "more immune" than SSD only.

Still very strange. Still very confused.

Paraphraser, 2021-01-22:
See also [Issue 253](https://github.com/SensorsIot/IOTstack/issues/253).

I've just implemented the fix described there on new-dev, put the SSD back on the USB3 port, done several reboots with/without the stack running and, thus far, it all seems happy.

The ping-reply pattern from the `sudo reboot` on new-dev is 7 replies while going down, 30 seconds no replies, replies start.

Granted that 3 tests are not statistically significant but it's two more than I could achieve before this patch.

Anyone else facing the same problem might want to give it a whirl!!!

Paraphraser, 2021-01-22:
I've implemented the [Issue 253](https://github.com/SensorsIot/IOTstack/issues/253) fix on all of my Pis. That's:

* iot-hub boot SD, run SSD
* sec-dev boot SD, run SSD
* octopi boot and run from SD
* new-dev boot and run from SSD (where I was having the most hassle).

All systems and associated IOTstack instances as happy as the proverbial Larry.

For the record, what I've **actually** been doing is:

1. Edit `/etc/dhcpcd.conf` to append the line:

	```
	allowinterfaces eth0,wlan0
	```

2. A tail to make sure it all looks OK (ie I didn't make a mess during the edit)

	```
	$ tail -2 /etc/dhcpcd.conf
	```

3. Tell the OS to notice the change:

	```
	$ sudo systemctl daemon-reload
	$ sudo systemctl restart dhcpcd
	```

4. That causes a small stutter in ssh responsiveness so I give it a minute in case anything else needs time to react, then:

	```
	$ sudo reboot
	```

Paraphraser, 2021-01-25:
It's day 4 (give or take). I'm up to about 30 reboots on new-dev scattered across the period, interspersed with two separate apt update / apt upgrades followed by an immediate reboot, always with IOTstack up and running, the whole time with the SSD on the upper USB3 port. The thing hasn't missed a beat.

I've also done a couple of reboots of iot-hub and sec-dev for good measure. They're happy too.

I think we're on a winner here. @gbsmith ➡️ IOTstack Hall Of Fame! 👏👏👏

My standard build script now includes:

```
sudo bash -c '[ $(egrep -c "^allowinterfaces eth0,wlan0" /etc/dhcpcd.conf) -eq 0 ] && echo "allowinterfaces eth0,wlan0" >> /etc/dhcpcd.conf'
```

Paraphraser, 2021-01-29:
Since installing the "allowinterfaces eth0,wlan0", I've had one firing of `isc-dhcp-fix.sh` (see [Does your Raspberry Pi's Wireless Interface freeze?](https://gist.github.com/Paraphraser/305f7c70e798a844d25293d496916e77))

`isc-dhcp-fix.sh` was an earlier attempt to deal with WiFi going walkabout for no apparent reason. Given that they are both playing in the same sandpit (DHCP) I had been thinking that the "allowinterfaces eth0,wlan0" might've cured the walkabout problem too. But it seems not. Pity.

Anyway, it's a week since "allowinterfaces eth0,wlan0" and I haven't had a single freeze anywhere.

lbaca, 2021-02-08:
Thanks for keeping us updated, @Paraphraser.

Back on 23 January I mentioned on Discord that I would report back, and here I am fulfilling my (belated) promise.

Since my RPi4 is in a rather awkward position under a desk, I didn't feel like crawling under there to do a series of rigorous tests, but I do have some good news. When I first brought the issue up on Discord, I had the SSD plugged into the top USB3. You suggested I try the bottom one, so I did. _At the same time_, I did an `apt full-upgrade`, including installing the latest EEPROM bootloader. One or both of those two changes has resulted in me being able to reboot to my heart's content, and never again experiencing the issue whereby I was forced to unplug and replug the SSD to get it to boot.

From my perspective, the problem is solved!

Paraphraser, 2021-03-01:
Here we are - March 1st already. How time flies!

I wanted to confirm that I have not had a single freeze, at boot time or at any other time, on any RPi since applying the "allowinterfaces eth0,wlan0" patch.

I have rebuilt new-dev at least twice (not to solve any real problem, just refining and testing my build scripts). The most-recent build was on Feb 10.

sec-dev used to boot SD, run SSD but I rebuilt that on Feb 23 to be native boot from SSD.

For the record, the OS version and EEPROM status is:

```
Linux iot-hub 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
BCM2711 detected
Dedicated VL805 EEPROM detected
BOOTLOADER: up-to-date
CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 LATEST: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
 
Linux sec-dev 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
BCM2711 detected
Dedicated VL805 EEPROM detected
Checking for updates in /lib/firmware/raspberrypi/bootloader/default
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
RELEASE: default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
 
Linux new-dev 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
Sep  3 2020 13:11:43
version c305221a6d7e532693cc7ff57fddfc8649def167 (release)
timestamp 1599135103
update-time 0
capabilities 0x00000000
BCM2711 detected
Dedicated VL805 EEPROM detected
Checking for updates in /lib/firmware/raspberrypi/bootloader/default
Use raspi-config to select either the default-production release or latest update.
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
RELEASE: default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

closed 2021-06-01 by Paraphraser