SensorsIot/IOTstack #222: Unable to install when following quick start : "Please run as root"

TommySharpNZ, 2021-01-02:
I'm just following through the Getting started guide and at the point where it says...

$ cd ~/IOTstack
$ ./menu.sh
Select "Native Installs"
Select "Install Docker and Docker-Compose"

When I do that I get told to "Please run as root"...

With the lengthy discussion on "sudo" in the guide I just didn't want to go trying lots of things...

Anyone got some ideas on what needs to be done?

Slyke, 2021-01-05:
Hey @TommySharpNZ can you try to install everything with the automatic installer?
```
curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash
```

obergerdavid, 2021-04-20:
Hi @TommySharpNZ 

there is a missing sudo in the install script.
As a temporary workaround, please edit the file `scripts/install_docker.sh` and change line 22 from:
`curl -fsSL https://get.docker.com | sh`
to
`curl -fsSL https://get.docker.com | sudo sh`

[PR 325](https://github.com/SensorsIot/IOTstack/pull/325) is already created.

Paraphraser, 2021-04-20:
@obergerdavid - I must admit that your post (and the related pull request) gave me a fright. I thought, "surely I can't have been **that** dumb?!" and "surely I copied and pasted what I **actually** use and, if not, what on earth did I use as my source when I stuck that in the doco?" And so on. As you do... 

However, I think something else must be going on that is causing the underlying problem for @TommySharpNZ

Please study this screen shot:

[image]

In words:

1. Does the docker install script mention sudo - yes but then wraps it as `sh_c`
2. How many times does the docker install script use `sh_c` (ie wrapped invocations of `sudo`)? 26
3. What form of invocation is used by the IOTstack install script? It's sudoless.
4. What form of invocation is used by new menu? It's sudoless.
5. What form of invocation is used by old menu? It's sudoless.

I have four RPis (one 3B+ and three 4s) every single one of which has been built starting from BalenaEtcher and a series of "canned" scripts which include:

```
curl -fsSL https://get.docker.com | sh
sudo usermod -G docker -a $USER
sudo usermod -G bluetooth -a $USER
sudo apt install -y python3-pip python3-dev
sudo pip3 install -U docker-compose
sudo pip3 install -U ruamel.yaml==0.16.12 blessed
```

Again, no `sudo` on the pipe to shell. By "canned" I mean that I just call a series of scripts in order so I get a predictable, repeatable and auditable result.

So, between the observable facts that the Docker script invokes `sudo` when needed, and all the IOTstack-supplied scripts do not use `sudo`, plus my practical experience of not using `sudo` in my own install scripts, that's a **lot** of evidence in favour of the conclusion that `sudo` is not a actually needed to get Docker installed on a Raspberry Pi running Raspberry Pi OS.

That's why I think something else must be going on. Like maybe not a Raspberry Pi. Or maybe not Raspberry Pi OS. Or maybe something else so seriously hosed under the hood that a clean installation is called for.

The only thing that gives me pause is [the Docker doco](https://docs.docker.com/engine/install/debian/#install-using-the-convenience-script) which **does** use `sudo` in its example. Axiomatically, if a script contains calls to `sudo` then the script itself should not need to be invoked using `sudo` and will often produce unintended results if it is.

I haven't looked at that Docker doco page in a while but, now that I have, I'm wondering whether we should still be doing our installs via curl piped to shell?

Paraphraser, 2021-04-21:
> I haven't looked at that Docker doco page in a while but, now that I have, I'm wondering whether we should still be doing our installs via curl piped to shell?

And after reading it more carefully, we seem to be stuck with the "convenience scripts" on Raspbian:

[image]

I'm glad I re-read that. Saved me the time in doing what I was planning - to test the "apt" method after starting from a clean BalenaEtcher image.

Paraphraser, 2021-04-21:
The Docker web documentation (earlier link which does include `sudo`) is also at variance with no less an authority than the script itself:

[image]

I'm thinking of filing a bug report to get this issue clarified.

obergerdavid, 2021-04-21:
Hi @Paraphraser,

thank you very much for your input and your thoughts!
I just took a look at the documentation and in the first bullet point of the section [Install using the convenience script](https://docs.docker.com/engine/install/debian/#install-using-the-convenience-script) there is written:

> The scripts require `root` or `sudo` privileges to run. Therefore, you should carefully examine and audit the scripts before running them.

So I think this sudo call was missing all the time. But maybe we can resolve this when they answer your bug report.

Paraphraser, 2021-04-21:
Well, to my mind, saying that a script needs root privileges is not *quite* the same as saying "use sudo to execute this script". I interpret "needs root privileges" as saying that the account running the script needs to be in the sudoers list.

However, seeing as the web doco has `sudo` in front of its examples, I agree that I'm splitting some exceedingly fine hairs.

<hr>

Putting that to one side, I still can't reconcile these facts:

* The entire evolution of IOTstack (from gcgarner in 2019 through old menu and new menu in SensorsIot) has used the sudoless form of curl piped to shell. To conclude that `sudo` *should always have been there* is at odds with all that history. Or do you disagree?
* Internally, the `get.docker.com` script invokes `sudo` *where needed* so sticking `sudo` on the front actually does nothing other than make everything in the script run with elevated privileges. That's not exactly an application of the principle of least privilege.
* Tommy opens this issue but is, I hope you will agree, exceptionally sparse on specifics. No description of the hardware or OS. No `uname -a` output. No screen grabs that might have led us to other conclusions.
* Tommy does not respond to Slyke's attempt to help. No "that fixed the problem" or "it turned out to be something else". Just silence. We don't even know if it is an ongoing problem.

A bit more drilling down. Let's go back to the original complaint. Tommy wrote:

> When I do that I get told to "Please run as root"...

Can we find the source of that message:

```
$ cd ~/IOTstack
$ find . -name "*.sh" -exec grep -Hi "please run as root" {} \;
./scripts/install_docker.sh:  then echo "Please run as root"
```

Tommy said he was in the menu and had selected "Install Docker and Docker-Compose" so that fits. But let's make sure `get.docker.com` can't produce the same message:

```
$ curl -fsSL https://get.docker.com | grep -i "please run as root"
$ 
```

No. `install_docker.sh` seems to be the only source. What is the context in that script where the message is produced?

```
$ grep -B 1 -A 2 "Please run as root" scripts/install_docker.sh 
if [ "$EUID" -ne 0 ]
  then echo "Please run as root"
  exit
fi
```

It's an explicit check, by `install_docker.sh`, to see whether it (`install_docker.sh`, not the `get.docker.com` script) has been invoked with `sudo`.

Taken at face value, it is `install_docker.sh` that is not being launched via `sudo`. Tommy said he was in the menu. How does the menu launch `install_docker.sh`?

```
$ grep "install_docker.sh" menu.sh
	sudo bash ./scripts/install_docker.sh install
	sudo bash ./scripts/install_docker.sh upgrade
```

Tommy's issue was opened on Jan 2nd. Maybe there was a bug in the menu back then which has since been fixed:

```
$ git checkout 'master@{2021-01-02 00:00:00}'
pi@new-dev:~/IOTstack$ grep "install_docker.sh" menu.sh
	sudo bash ./scripts/install_docker.sh install
	sudo bash ./scripts/install_docker.sh upgrade
```

No difference so that's not the explanation.

I hope you can see why I find the idea of adding `sudo` after the pipe a bit troubling. The solution doesn't actually seem to be directed at the original problem. Indeed, it is not clear **how** Tommy was able to trigger the "Please run as root" message in the first place. The menu is using `sudo` to invoke `install_docker.sh` which, in turn, means that **everything** inside the script is running with elevated privileges **including** the `sh` on the other side of the pipe.

The explanation **MUST** lie elsewhere.

My money would be on Tommy's system either not being a Raspberry Pi, or a Pi but running something other than Raspbian, or a Pi running Raspbian but a user other than "pi" which is not a member of sudoers.

But we won't actually get anywhere until we hear from Tommy.

projectsdotie, 2021-04-23:
Hi all, not sure if this helps but, I ran into problems today using the latest build where the ./menu.sh containers menu is not displaying correctly and goes into a loop, I thought it could have been because I made a mistake so I removed 'IOTStack' entirely and reinstalled. While my menu looping problem is still there I also noted that when I tried the 'Native Install', of 'Docker and Docket-Compose' I get a similar (if not the same) message 'Tommy' outlined above as follows:

[image]

"Install docker
Install docker-compose
bash ./scripts/install_docker.sh install
Please run as root

Process terminated. Press [Enter] to show menu and continue."

Not sure if it helps but ...

Paraphraser, 2021-04-24:
The weird thing about this is how your screen grab can show `install_docker.sh` being invoked without `sudo` when the `grep` output from `menu.sh` (in my earlier post) shows it is invoked with `sudo`. 

Let's see if we can drill into this.

Please read the [list of assumptions](https://github.com/SensorsIot/IOTstack/blob/master/docs/Getting-Started.md#-assumptions-) and tell me whether you meet all of those or, if not, how you vary. Perhaps include `uname -a` output in your reply.

Maybe try running the same `grep "install_docker.sh" ~/IOTstack/menu.sh` to confirm that there is a sudo on the front of the install_docker.sh calls. I can't, for the life of me, see why it would be otherwise but it never hurts to be certain.

Maybe try a `ls -al $(which -a sudo)`. There are two instances on each of my systems (including a 3B+). They have the same time stamp, size and sha256 hash so I assume your system will follow the same pattern. Perhaps an `echo $PATH` as well.

Maybe try something like a `sudo ls` just to make sure sudo is actually working. I once got a "sudo not found" when it seemed to be there. There was no reason I could find why it would not run. I couldn't even reboot and the same happened after a power off/on, so I gave up and reinstalled.

Then, with a view to getting you past this problem, perhaps see [this gist](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f) and, more specifically:

* do the steps in [uninstalling docker and docker-compose](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#-uninstalling-docker-and-docker-compose-) in case anything is somehow getting in the way
* then either remove or rename the existing `~/IOTstack` folder (removing the folder *might* need `sudo` which is OK)
* then read **all** of [scripting installations](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#scripting-iotstack-installations) so you understand you need to make a decision about old menu (run the commands "as written") or new menu (omit the `-b old-menu` from the `git clone`)
* then execute the "scripting installations" commands in the order shown
* then try re-running the menu again.

Slyke, 2021-05-05:
I'm unable to reproduce this on `master` and `old-menu`.

On the docker docs page it does say that
> The scripts require root or sudo privileges to run. Therefore, you should carefully examine and audit the scripts before running them.

This implies that the script requires sudo privileges, but it doesn't imply that it needs to be run with sudo (as in, it will request sudo when it needs it).

However, in their example they do use sudo:
```
 curl -fsSL https://get.docker.com -o get-docker.sh
 sudo sh get-docker.sh
```

But it also mentions [post-installation steps for Linux](https://docs.docker.com/engine/install/linux-postinstall/#manage-docker-as-a-non-root-user) which is the method IOTstack uses (adds current user [`pi`] to the `docker` group) for installing and running docker without sudo.

As a security precaution, you shouldn't be running scripts from the internet with sudo blindly, and it certainly wouldn't be a good idea to have the installer do that without the user knowing. It only takes 1 DNS hijacking and they have arbitrary sudo execution on the system.