SensorsIot/IOTstack #223: deconz devices settings not working

breemts, 2021-01-03:
### Setup:
Raspberry Pi 3
with Raspbian 2020-12-02
RaspBee on /dev/ttyS0
IOTStack commit 564e244

### issue
In the menu.sh I configured deconz to use the device /dev/ttyS0. Resulted in the following configuration:
~/IOTstack/services/deconz/build_settings.yml
```version: '1'
application: IOTstack
service: deconz
comment: Build Settings
hardware:
- "/dev/ttyS0"
databasePasswordOption: Do nothing
```
~/IOTstack/docker-compose.yml
```deconz:
    image: marthoc/deconz
    container_name: deconz
    restart: unless-stopped
    ports:
    - "8090:80"
    - "443:443"
    - "5901:5900"
    volumes:
    - ./volumes/deconz/:/root/.local/share/dresden-elektronik/deCONZ
    devices:
      # This list is replaced during the build process. Modify the list in "build_settings.yml" to change it.
    - /dev/null
    environment:
    - DECONZ_VNC_MODE=1
    - DECONZ_VNC_PASSWORD=IOtSt4ckDec0nZ     # %randomPassword% is overwritten during preBuild hook
    - DEBUG_INFO=1
    - DEBUG_APS=0
    - DEBUG_ZCL=0
    - DEBUG_ZDP=0
    - DEBUG_OTAU=0
    networks:
    - iotstack_nw
    logging:
      options:
        max-size: "5m"
        max-file: "3"
```

With this configuration I was not able to get a connection to the RaspBee device.

But it worked find using this docker command:
```
docker run -d \
   --name=deconz \
   --net=host \
   --restart=always \
   -v /etc/localtime:/etc/localtime:ro \
   -v ~/.local/share/dresden-elektronik/deCONZ:/root/.local/share/dresden-elektronik/deCONZ \
   --device=/dev/ttyS0 \
   -e DECONZ_DEVICE=/dev/ttyS0 \
   -e DECONZ_VNC_MODE=1 \
   -e DECONZ_VNC_PORT=5901 \
   -e DECONZ_VNC_PASSWORD=IOtSt4ckDec0nZ \
   marthoc/deconz
```

So my assumption is that the device settings are not parsed correctly during the build.

### Workround

adding this in the docker-compose.yml worked for me
```
devices:
- /dev/ttyS0
volumes:
 - /etc/localtime:/etc/localtime:ro
environment:
 - DECONZ_DEVICE=/dev/ttyS0
```

I have not investigated where the issue is in the menu. I think for someone who wrote the code I should be feasible.

Schnuecks, 2021-01-03:
Hi I ran into this issue, a good help I found in the official repo https://github.com/marthoc/docker-deconz

breemts, 2021-01-03:
@Schnuecks I already know the official repo. And as described I have no issues with the marthoc/deconz image. It is working fine. As described only the configuration for docker compose generated from the IOTstack menu is not working.

Slyke, 2021-01-05:
Hello,

I just pushed a fix for this. Seems like the password option was overwriting the saved hardware selection. Give it a try now and let me know if you continue to have issues with it.

basekk, 2021-02-26:
Hi all, 

I'm just trying to get the **RaspBee I** running on a Raspberry 3B - without success.

I have the same issue as @breemts had: 
The connection to the Raspbee device on `/dev/ttyS0` fails. 

His work around, adding the lines into the docker-compose.yml file didn't help. 
The project is up to date. So I would assume the fix pushed by @Slyke is not working properly(?).

```
  deconz:
    image: marthoc/deconz
    container_name: deconz
    restart: unless-stopped
    ports:
      - '8090:80'
      - '443:443'
      - '5901:5900'
    volumes:
      - ./volumes/deconz/:/root/.local/share/dresden-elektronik/deCONZ
    # - /etc/localtime:/etc/localtime:ro
    devices:
      #ConBee - changed manually
      - /dev/ttyS0
    environment:
      - DECONZ_VNC_MODE=1
      - DECONZ_VNC_PASSWORD=changeme
      - DEBUG_INFO=1
      - DEBUG_APS=0
      - DEBUG_ZCL=0
      - DEBUG_ZDP=0
      - DEBUG_OTAU=0
      - DECONZ_DEVICE=/dev/ttyS0 #added as per @breemts 

```
Deconz Container is up and running.

**LOG Deconz**

```
00:27:56:047 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/,
00:27:56:080 CTRL. 3.27.200:27:56:202 dev /dev/ttyAMA0,
00:27:56:203 dev /dev/ttyS0,
00:27:56:204 COM: /dev/ttyS0 / serialno: , RaspBee,
00:27:56:205 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt,
00:27:56:526 parent process /bin/sh,
00:27:56:527 gw run mode: docker,
00:27:56:527 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version,
00:27:56:528 sd-card cid: 035344534333324780ffffffff013401,
00:27:56:531 DB sqlite version 3.27.2,
00:27:56:533 DB PRAGMA page_count: 33,
00:27:56:534 DB PRAGMA page_size: 4096,
00:27:56:534 DB PRAGMA freelist_count: 0,
00:27:56:534 DB file size 135168 bytes, free pages 0,
00:27:56:535 DB PRAGMA user_version: 7,
00:27:56:535 DB cleanup,
00:27:56:536 DB create temporary views,
00:27:56:548 started websocket server at port 443,
00:27:56:553 [INFO] - Found file containing button maps. Parsing data...,
00:27:56:559 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...,
00:27:56:575 [INFO] - Button maps loaded.,
00:27:56:578 dlg action: Read binding table,
00:27:56:578 found node plugin: libde_rest_plugin.so - REST API Plugin,
00:27:56:583 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin,
00:27:56:612 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin,
00:27:56:680 dev /dev/ttyAMA0,
00:27:56:680 dev /dev/ttyS0,
00:27:56:681 COM: /dev/ttyS0 / serialno: , RaspBee,
00:27:57:741 dev /dev/ttyAMA0,
00:27:57:742 dev /dev/ttyS0,
00:27:57:742 COM: /dev/ttyS0 / serialno: , RaspBee,
00:27:57:764 COM check bootloader,
00:27:59:770 COM timout query bootloader, assume application,
00:28:01:346 device state timeout ignored in state 2,
00:28:01:788 dev /dev/ttyAMA0,
00:28:01:788 dev /dev/ttyS0,
00:28:03:514 New websocket 192.168.178.20:64464 (state: 3) ,
00:28:04:484 device state timeout ignored in state 2,
00:28:07:484 device state timeout ignored in state 2,
00:28:10:484 device state timeout ignored in state 2,
00:28:11:572 dev /dev/ttyAMA0,
00:28:11:573 dev /dev/ttyS0,
00:28:11:804 discovery network reply error: Host phoscon.de not found,
...
00:28:54:483 wait reconnect 2 seconds,
00:28:55:484 wait reconnect 1 seconds,
00:28:55:544 dev /dev/ttyAMA0,
00:28:55:544 dev /dev/ttyS0,
```

Any help how to get this thing work is highly appreciated. 
Thanks in advance!

Basekk

Schnuecks, 2021-02-26:
Hi,

please get through this step by step 
https://phoscon.de/de/raspbee/install#docker
i think you need --device=/dev/ttyAMA0 that what i have in my docker-compose.yml too (but have an Conbee 2 Stick)

basekk, 2021-03-01:
Hi @Schnuecks , 

You were right. I beg your pardon -- my mistake. It works perfectly fine with `/dev/ttyS0` and a RaspBee I.