SensorsIot/IOTstack #233: Node-Red unable to connect to Mosquitto

larrybell, 2021-01-11:
New install on SSD. In the process of migrating from VMs running on VMWare to IOTStack. The only issue so far is Mosquitto. Devices are not able to talk to Mosquitto.

1610397239: mosquitto version 1.6.12 starting
1610397239: Config loaded from /mosquitto/config/mosquitto.conf.
1610397239: Error: Unable to open pwfile "/mosquitto/config/pwfile".
1610397239: Error opening password file "/mosquitto/config/pwfile".
 
This is a new error after migrating from the gcgarner repo. I was hoping to fix my first problem now i have this one.

Paraphraser, 2021-01-11:
It probably means what it says (basically, your pwfile is in the wrong place or might have the wrong permissions).

You didn't include any of the relevant files in your issue so I can't immediately pinpoint the actual problem.

I'll step through how I would diagnose a problem like this on my own system. With luck, something I say will help you figure out what's what on your own system.

Starting with the template:

```
$ grep "password" ~/IOTstack/.templates/mosquitto/mosquitto.conf
#Uncomment to enable passwords
#password_file /mosquitto/pwfile/pwfile

$ grep "pwfile" ~/IOTstack/.templates/mosquitto/service.yml
      - ./volumes/mosquitto/pwfile:/mosquitto/pwfile
```

After the menu run, the **active** `mosquitto.conf` will wind up at:

```
~/IOTstack/services/mosquitto/mosquitto.conf
```

If passwords are enabled by uncommenting the "#password_file" line in the **active** `mosquitto.conf` then, taken together with the `service.yml` definition:

* The **file** `pwfile` appears at the **internal** directory path `/mosquitto/pwfile`
* That **internal** directory path maps to the **external** directory path `./volumes/mosquitto/pwfile`, so
* The password file, if present, should be at the **external** path:

	```
	~/IOTstack/volumes/mosquitto/pwfile/pwfile
	```

If I were having a problem like this, I'd:

* confirm that the **active** `mosquitto.conf` was set as expected (ie it wasn't using a different internal path - I seem to recall that there was an issue with this at some point in the gcgarner->SensorsIot journey). I think you may well find that yours has the wrong path.
* check that my `pwfile` was actually at the **external** path:

	```
	~/IOTstack/volumes/mosquitto/pwfile/pwfile
	```

* verify that `./volumes/mosquitto` and everything below it, including the `pwfile`, had ownership of "1883:1883". The "1883:1883" is what mosquitto (both the container and the broker process) expects.
* double-check the contents of these three files:

	```
	~/IOTstack/.templates/mosquitto/service.yml [1]
	~/IOTstack/services/mosquitto/service.yml   [2]
	~/IOTstack/docker-compose.yml               [3]
	```

	to make sure that the critical line:

	```
	      - ./volumes/mosquitto/pwfile:/mosquitto/pwfile
	```
	
	is the same in all three. [1] is the definition. [2] is a copy of [1] created by a menu run. [3] is inserted from [2] by a menu run. If you're coming from gcgarner, [2] might be out of sync with [1].

If you need to move files around and/or change permissions within `./volumes/mosquitto`, it's probably best if you start with:

```
$ cd ~/IOTstack
$ docker-compose stop mosquitto
$ docker-compose rm -f mosquitto
```

then do the work in the file system, then:

```
$ docker-compose up -d mosquitto
```

That's the only way I know to be certain that the running container has been recreated and sees the external file system the way you see it. I used to think a simple `docker-compose restart mosquitto` got the job done but I learned (the hard way) that that isn't always true.

larrybell, 2021-01-12:
Thank you so much for the quick response. I was going to just give up and go back to running on VMware. I will give this a try as soon as I can.

Again, Thank you