SensorsIot/IOTstack #235: nextcloud_db mounts wrong directory

obergerdavid, 2021-01-13:
The wrong path is mounted in the nextcloud_db template (.templates/nextcloud/service.yml), therefore the database is not saved in the volumes/nextcloud/db directory.
```
nextcloud_db:
  volumes:
  - ./volumes/nextcloud/db:/var/lib/mysql
```

As done in .templates/mariadb/service.yml it should mount to the /config directory:
```
mariadb:
  volumes:
    - ./volumes/mariadb/config:/config
```

Paraphraser, 2021-01-13:
Perhaps see [Preparing a Pull Request for IOTstack](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d). You've done most of the work. I've found that preparing a PR is a good way of getting changes made. Once you've done the work, it's also easy to apply the pending PRs to your own local copy of the repo (explained in the gist too) so you, as the author, gain the immediate benefit.

obergerdavid, 2021-01-13:
PR #238 created.

closed 2021-01-27 by obergerdavid