SensorsIot/IOTstack #236: Homeassistant is deployed in a different network of IOTStack

f616, 2021-01-13:
Homeassistant is deployed in a different network of IOTStack.
How can it be deployed on the same network (iotstack_nw)?
This is causing issues on configuring MariaDB on Homeassistant because I cannot user mariadb dns and have to configure the internal mariadb IP, which sometimes change over the reboots.

f616, 2021-01-13:
As a workaround my temporary solution is to configure a static IP on docker-compose.yml for mariadb

```
    networks:
      iotstack_nw:
         ipv4_address: 10.77.60.99
```

ukkopahis, 2022-06-19:
Cross-container connections should use the container names. For instance to access mariadb, instead of it's internal IP, use the hostname `mariadb`. Docker will dynamically resolve this to the correct IP.

closed 2022-09-11 by Slyke