SensorsIot/IOTstack #237: Mosquitto :  failed  to save /mosquitto/data/mosquitto.db.new for writing

reba0, 2021-01-13:
Wrong  user  in  Volumes directory   :    default User instead of  "1883"

LOG Output 
today at 01:41  2021-01-12T22:11:01: Error saving in-memory database, unable to open /mosquitto/data/mosquitto.db.new for writing.
today at 01:41  2021-01-12T22:11:01: Error: Invalid argument.

pi@raspi-test:~/IOTstack/volumes $ ls -al
insgesamt 24
drwxr-xr-x  5 pi   pi   4096 Jan 12 16:17 mosquitto
.templates is a directoryfix.sh   
copied  directoryfix.sh to  /IOTSTACK/
copied   service.yml   to   IOTSTACK/Services/mosquitto 

executeted the  direcotory fix 

pi@raspi-test:~/IOTstack/volumes $ ls -al
drwxr-xr-x  5 1883 1883 4096 Jan 12 16:17 mosquitto

working

senarvi, 2021-02-06:
I had the same problem and had to do the same fix.

azakow, 2021-02-17:
same issue here; I understand to fix the issue I need to run directoryfix.sh; where can I get this file; sry I am completely new to such things here in github

azakow, 2021-02-17:
🤦 found it; sry

batamire, 2021-04-02:
For anyone stumbling, manual command is: `sudo chown -R 1883:1883 /home/pi/IOTstack/volumes/mosquitto`

ukkopahis, 2022-06-23:
Problem now should be prevented by #350.

closed 2022-09-11 by Slyke