SensorsIot/IOTstack #239: New install: domoticz fails to install, missing "IOTstack/services/domoticz/domoticz.env"

zeddski, 2021-01-13:
And that's about it... 
* Clean install
* ./menu
* > Build Stack
* {select domoticz}
* >start stack
Result:
`Start Stack:
docker-compose up -d --remove-orphans
ERROR: Couldn't find env file: /home/pi/IOTstack/services/domoticz/domoticz.env

Stack Started
Process terminated. Press [Enter] to show menu and continue.
`

kkoletz, 2021-01-21:
Unfortunatelly I have the same issue! Hopefully help is coming!

Paraphraser, 2021-01-21:
I have just prepared [Pull Request 249](https://github.com/SensorsIot/IOTstack/pull/249) for this. How long it will be before this change is applied is an open question.

In the meantime, you can **EITHER**:

* follow the instructions at [this gist](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d#-testing-pending-pull-requests-) to apply PR249 to your local copy, then run-run the menu; **OR**
* Edit both of the following files:

	```
	~/IOTstack/services/domoticz/service.yml
	~/IOTstack/docker-compose.yml
	```
	
	In each case, you will be looking for these two lines:
	
	```
	    env_file:
	      - ./services/domoticz/domoticz.env      
	```
	
	Delete both of those lines. Save the file. Then bring up the stack again with:
	
	```
	$ cd ~/IOTstack
	$ docker-compose up -d
	```

Paraphraser, 2021-01-21:
See also [Issue 210](https://github.com/SensorsIot/IOTstack/issues/210).