SensorsIot/IOTstack #243: Fresh install: MQTT not connecting to mosquitto,

Daneild, 2021-01-16:
Hi, fresh install on a raspberry 3 with Hass this time.
Not managing to connect or to find the right mqtt broker ip adress or name.
All conf files are default from installation.
I have a sensor connected i can see this in zigbee2mqtt logs that also confirm that eclipse mosquitto it's up and ok.

_`Zigbee2MQTT:info  2021-01-16 22:32:06: MQTT publish: topic 'zigbee2mqtt/0x00124b002268d332', payload '{"battery":100,"humidity":26.31,"linkquality":99,"temperature":17.9,"voltage":3100}'`_
 
I have tried using "mosquitto", my raspberry ip adress ,localhost, 0.0.0.0 even mosquitto container address  but no luck.

 ```
docker inspect -f '{{range.NetworkSettings.Networks}}{{.IPAddress}}{{end}}' mosquitto
10.77.60.4
```

Mosquitto log are giving those messages, 192.168.100.23 is Tasmota rellay that is working , another proof that mosquitto is ok, 10.77.60.1 are internal ip's of docker, not involve in this, just an installation fresh using the ./menu.sh

```
_2021-01-16T22:06:30: New client connected from 10.77.60.1 as 2izIF08rObBNN7DP5qc0GQ (p2, c1, k60).
2021-01-16T22:10:08: New connection from 192.168.100.23 on port 1883.
2021-01-16T22:10:08: New client connected from 192.168.100.23 as DVES_B893F7 (p2, c1, k30, u'DVES_USER').
2021-01-16T22:18:50: Saving in-memory database to /mosquitto/data/mosquitto.db.
2021-01-16T22:31:15: New connection from 10.77.60.1 on port 1883.
2021-01-16T22:31:15: New client connected from 10.77.60.1 as 7iTYSNyqGoLjsZ0Tu8UvbF (p1, c1, k60).
2021-01-16T22:31:15: Client 7iTYSNyqGoLjsZ0Tu8UvbF disconnected.
2021-01-16T22:31:15: New connection from 10.77.60.1 on port 1883._

```
Thanks

Daneild, 2021-01-16:
solved, another install without Hass, 
This approach its deeply wrong, on home assistant event with the above issues i'm able to see history of sensors!
In configuration devices there are any!
Maybe in the future will be better and i will be able to help.
bye

closed 2021-01-16 by Daneild

batamire, 2021-04-02:
How did you manage to set `mqtt: broker` hostname/ip? I am having exactly the same issue... fresh docker install with menu.sh.