SensorsIot/IOTstack #245: Networking under the new menu

Paraphraser, 2021-01-19:
@Slyke 

I would like to open a discussion on "new menu" networking.

> I've made no secret of my views on the changes, particularly how it adds complexity that mere mortals should not have to deal with. You can relax because I don't propose to re-hash any of that here.

I have just done a clean clone of SensorsIot/IOTstack and stepped through the contents of the `.templates` folder to build the following table:

[image]

Here are the things that jump out at me:

1. domoticz defines `network_mode: bridge`.

	I can't find any documentation to support that. See [compose spec](https://github.com/compose-spec/compose-spec/blob/master/spec.md#network_mode).
	
	Any idea what "bridge" means in this context? Any idea why it actually works when it seems to be an undocumented option? What seems to happen is that domoticz is the only container attached to a network called "bridge".
	
2. Containers with a blue highlight have no "networks" clause in their service definitions. This implies that:
	* they get attached to "iotstack_default";
	* have interoperability issues with containers that **are** attached to "iotstack_nw".
	
	Interoperability issues seem to be a fairly common complaint on Discord.
	
3. The right-most column ("IOTstack_Net_Internal") is not associated with any service definition. It seems to serve no purpose other than to produce warning messages at "up" time.

4. The rationale for the "IOTstack_NextCloud" network is not clear. I assume it's intended to be a dedicated back-end path for database communications, presumably based on an assumption that it will improve performance.

	Such an assumption might be valid were (a) real switching hardware involved and (b) the unicast load to/from the database engine sufficient to swamp unicast traffic reaching the front end but I question whether there's likely to be any measurable performance difference in a Docker environment running on a single RPi.

	That said, my only real objection is the way it imposes itself on your attention at "up" time when it isn't in use. At the very least, I think that needs to be fixed. Please keep reading!
	
5. I assume PiHole is attached to "IOTstack_VPN" to provide a DMZ-like function such that DNS resolution does not have to reach the internal network. If there's another reason then please elaborate.
	
Rolling all that together, here's what I think we should do:

1. Get rid of "IOTstack_Net_Internal". It isn't doing anything.
2. Reconsider "IOTstack_NextCloud".
3. Retain "IOTstack_VPN" if there's a good reason for keeping it.
4. Remove every single mention of "IOTstack_Net" in every single service definition so that, by default, every container is attached to "iotstack_default". By inference, the exceptions would be containers in host mode and the "IOTstack_NextCloud" and "IOTstack_VPN" cases if they continue.

I just bet your next question is something like:

> but how will PiHole work? It can't just be attached to "IOTstack_VPN" alone because then it won't be visible to other containers. It has to be attached to something like "IOTstack_Net" too and that, in turn, forces other containers to adopt "IOTstack_Net".

Not true!

```
$ cat docker-compose.yml
version: '3.6'

services:

  portainer-ce:
    container_name: portainer-ce
    image: portainer/portainer-ce
    restart: unless-stopped
    ports:
      - "8000:8000"
      - "9002:9000"
    volumes:
      - /var/run/docker.sock:/var/run/docker.sock
      - ./volumes/portainer-ce/data:/data

  nodered:
    container_name: nodered
    build: ./services/nodered/.
    restart: unless-stopped
    user: "0"
    env_file: ./services/nodered/nodered.env
[... 237 lines cut ...]
> The same approach would work for nextcloud if the back-end network is retained.

Seems to me this solves a bunch of problems:

1. No more half-in, half-out interoperability issues.
2. IOTstackers who aren't data-comms gurus can have a quiet life.
3. New containers can be added without having to consider comms **unless it is necessary**.
4. The default network goes back to Docker choosing the IP range.
	
Simplifying it down to just those containers that actually need specialised comms probably also creates the opportunity for you to revisit how network definitions get tacked onto docker-compose.yml, so that they are only included **when needed**.

Rather than the all-in-one `.templates/env.yml`, how about something like:

```
~/IOTstack/.templates
  Network_Definitions
    vpn_nw.yml
    nextcloud_internal.yml
```

A mention of a network name like "vpn_nw" in a service definition is the signal to include the "vpn_nw.yml" at the end of docker-compose.yml with the "networks:" header dependent on whether there is at least one extra network.

Including an empty "default.yml" would generalise the solution for the PiHole and NextCloud cases, and provide the opportunity to chuck out a warning if a service definition mentioned a network for which there was no corresponding NETWORK_NAME.yml file.

Thoughts?

senarvi, 2021-01-28:
I've only used IOTstack for a couple of days so I cannot comment on your proposal, but you're right the networking was confusing. zigbee2mqtt_assistant is by default configured to connect to mosquitto, but it won't find the host, because mosquitto is in iotstack_nw network and zigbee2mqtt_assistant is not. Also, it's not clear whether iotstack_nw ("exposed by your host") is the correct network to use, or whether iotstack_nw_internal ("for interservice communication, no access to outside") would be adequate.

Paraphraser, 2021-01-28:
@senarvi short answer: iotstack_nw

The basic tests are:

* does the process running inside the container need to communicate with the outside world?
* does something in the outside world need to communicate with the process running inside the container?
 
If the answer to either/both is "yes" then "iotstack_nw" is appropriate.

Because the vast majority of situations answer "yes" to both questions, it would be better if that was the default. The most appropriate default is so have no networking specified at all. Then docker-compose automatically sets up `iotstack_default` and attaches all containers to it. Nobody needs to think about networking at all. It just works.

Then, if there are specific situations where you need specialised networks, handle those case-by-case. That's what the proposal is all about.

Slyke, 2021-02-01:
All very good points. Most of the services require communicating with the outside world. IOTstack will allow you to select the networking mode, and the network(s) on a per container basis. It will not include any unused networks. The reason is that some users may not want some of their services only accessible via VPN and docker internal.
[image]

closed 2021-06-01 by Paraphraser

mats-nk, 2021-09-13:
There is a third option in the evaluation process:

_The basic tests are:

    1. does the process running inside the container need to communicate with the outside world?
    2. does something in the outside world need to communicate with the process running inside the container?

If the answer to either/both is "yes" then "iotstack_nw" is appropriate._

And that is:
3. Is any multicast needed in the container?

If yes then is `network_mode: host` needed for the container.