SensorsIot/IOTstack #247: Upgraded python packages not found when starting menue.sh

elaiel, 2021-01-20:
Hi there
I tried to switch from the original gcgarner/IOTstack to the Sensorlot/IOTstack, following the steps for the migration. Everthing worked till I had to run the menu.sh script. When the script was executed, I was asked if I wanted to upgrade the python packages: 

```
Python 3.6.9 or later (Current = 3.7.3), ruamel.yaml 0.16.12 or later      │
 │ (Current = Unknown), blessed and pip3 are required for IOTstack to         │
 │ function correctly. Install these now? 
```

I started the installation of the packages:

```
pi@pi4server:~/IOTstack $ ./menu.sh
Checking for project update
Von https://github.com/SensorsIot/IOTstack
 * branch            master     -> FETCH_HEAD
Project is up to date
Python Version: 'Python 3.7.3'. Python is up to date.
ruamel.yaml Version: 'Unknown'. ruamel.yaml is outdated.
OK:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
OK:2 http://archive.raspberrypi.org/debian buster InRelease
OK:3 https://download.docker.com/linux/raspbian buster InRelease
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Alle Pakete sind aktuell.
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
python3-dev ist schon die neueste Version (3.7.3-1).
python3-pip ist schon die neueste Version (18.1-5+rpt1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: ruamel.yaml==0.16.12 in /home/pi/.local/lib/python3.8/site-packages (0.16.12)
Requirement already satisfied: blessed in /home/pi/.local/lib/python3.8/site-packages (1.17.12)
Requirement already satisfied: ruamel.yaml.clib>=0.1.2 in /home/pi/.local/lib/python3.8/site-packages (from ruamel.yaml==0.16.12) (0.2.2)
Requirement already satisfied: wcwidth>=0.1.4 in /home/pi/.local/lib/python3.8/site-packages (from blessed) (0.2.5)
Requirement already satisfied: six>=1.9.0 in /usr/local/lib/python3.8/site-packages (from blessed) (1.15.0)
Please enter sudo pasword if prompted
Command: docker version -f "{{.Server.Version}}"
Docker version 20.10.2 >= 18.2.0. Docker is good to go.
Project dependencies up to date

Existing installation detected.
Traceback (most recent call last):
  File "./scripts/menu_main.py", line 3, in <module>
    from blessed import Terminal
ModuleNotFoundError: No module named 'blessed'
```

The module "blessed" should be available. I can run ./scripts/menu_main.py manually and it works. 
I als tried to manually install belssed as module using pip, pip3, as user and via sudo - same result.
Also I should have the most current version of the ruamel.yaml, not sure why the version is not detected correctly.
Any ideas how I solve this?

Paraphraser, 2021-01-20:
No ideas. Didn't happen to me. Just worked the first time I tried the new menu.

Did you try running `sudo pip3 install` with either/both the  `--upgrade` or `--force-reinstall` flags?

I'm not 100% sure of this but I **think** that if you run pip3 without sudo then it installs in ~/bin or ~/.local/bin, assuming one of those exists whereas, with sudo, it picks places like /usr/local/bin. Although I can't point to an example, it seems to me that if one uses a mixture of sudo/non sudo calls then one might potentially wind up in the situation where an old non-sudo version of something winds up blocking a newer sudo version of something. Maybe nose around your PATH and see what's where.

Slyke, 2021-01-24:
@elaiel there were 2 PRs that were just merged, update to the latest version and see if that fixes it.