SensorsIot/IOTstack #248: Container mozilla webthings deprecated, now is webthingsio

erickcampos50, 2021-01-20:
It's just a minor update, since container mozilla webthings now is deprecated, it should be fixed to webthingsio/gateway on menu.sh

Paraphraser, 2021-01-20:
It has been - see [SensorsIot/IOTstack/blob/master/.templates/webthings_gateway/service.yml](https://github.com/SensorsIot/IOTstack/blob/master/.templates/webthings_gateway/service.yml).

The problem is that this is only on the master branch. You won't see it in the old-menu branch which is, I'd be the first to agree, a bit of a pest. 

See also [gcgarner/IOTstack/issues/224](https://github.com/gcgarner/IOTstack/issues/224). Someone running on the old repo said the same thing 24 hours ago and I went through how to work around it.

Hope that helps.

erickcampos50, 2021-01-25:
Thank you @Paraphraser !

closed 2021-01-25 by erickcampos50