SensorsIot/IOTstack #253: docker/dhcpcd conflict

gbsmith, 2021-01-21:
Wanted to share this in case anyone else experiences the same issue. After installing the IOTStack and building out a bunch of containers, my RPi4 seemingly stopped booting. Turns it was booting - but the network was hosed, and nothing could get in or out. I finally found out that the docker.service was killing the dhcpcd service. Even if I set up dhcpcd to start **after** docker on boot, any restart of docker would take dhcpcd down. Simply adding 
```
allowinterfaces eth0,wlan0
```
at the end of `/etc/dhcpcd.conf` appears to have done the trick, confining DHCP client activities just to those two main interfaces and staying out of docker's way.

(this **and** falling back to the Sep 2020 bootloader to keep the SSD booting!)

Paraphraser, 2021-01-22:
Hey - a **HUGE** "thank you" for this.

See also [Issue 219](https://github.com/SensorsIot/IOTstack/issues/219). I've just implemented this, put the SSD back on the USB3 port, done several reboots with/without the stack running and, thus far, it all seems happy.

gbsmith, 2021-01-22:
Glad to hear it. Hears hoping this is the Right Solution, and it keeps working for us.

JMBMdes, 2021-02-21:
Hi all, I think I suffered the same problem you mention. If you create a large number of containers, it tries to create an interface for each of them and it seems some component of dhcpcd overflows. The sympton is the raspi starts but suddenli comms are down. I soved it adding 
`denyinterfaces veth*`
at the beginning of `/etc/dhcpcd.conf`

gbsmith, 2021-02-21:
@JMBMdes, that is roughly the same thing but from the deny side. `allowinterfaces eth0,wlan0` denies everything that is neither eth0 nor wlan0 (e.g., veth4). Not sure if either one is better. Depends if there is something other than eth0, wlan0 and veth* getting caught in the crossfire.

Blu-Eagle, 2021-07-19:
thank you, i thought I was doing something wrong, so I solved.
thx

Paraphraser, 2021-07-19:
Agreed. Haven't been able to make much progress on this topic. In the meantime, see if [PiBuilder](https://github.com/Paraphraser/PiBuilder) could help.

Incidentally, I've started using `allowinterfaces eth*,wlan*`. I was trying to set up a two-port transparent bridge using a couple of USB-to-Ethernet dongles and realised I was blocking `eth1` and `eth2`.

I did experiment with the "deny" approach but it didn't seem to work when I tried it. With the benefit of hindsight, that may have been the fault of one of the dongles which I have since come to suspect is a bit flaky. Nevertheless, wildcarding eth and wlan works for me so that's what I'm using.

Paraphraser, 2022-06-19:
Documented in getting started. PiBuilder does this automatically. I'd call this fixed.

closed 2022-09-11 by Slyke