SensorsIot/IOTstack #255: influxdb Configuration file can't be edited with nano inside of influxdb container

kkoletz, 2021-01-23:
Hi everyone,

I've tried to setup my influxdb database and it's rights in order to use it with grafana and iobroker.
Now the admin and user account setup is working fine, also creating the database is fine. 
Now I want to edit the influxdb config file to be able to access the newly created database, in my case called "iobroker" and also to set correctly the https certificates. 
 [http]  
 enabled = true  
 bind-address = ":8086"  
 auth-enabled = true
 log-enabled = true  
 write-tracing = false  
 pprof-enabled = false  
 https-enabled = false  
 https-certificate = "/etc/ssl/influxdb.pem"  

I want to edit the influxdb.conf file and it says that nano command not found. How can I open the influxdb.conf file and in case of IOTstack folder structure, in which directory is the influxdb.pem file?
root@3ddf65d363f3:~# cd /etc/influxdb/
root@3ddf65d363f3:/etc/influxdb# nano influxdb.conf 
bash: nano: command not found
root@3ddf65d363f3:/etc/influxdb# ls
influxdb.conf
root@3ddf65d363f3:/etc/influxdb# 

Unfortunately I could not finde a proper documentation for this issue so all my hopes are on this wonderful community.
Thank you.

Paraphraser, 2021-01-23:
The answer is that, when influxdb is running in Docker, you do **not** edit the config file.

I understand that this will feel "wrong" if you've been using influxdb outside of Docker but that's the way it actually is.

Every option you can set in a config file can also be set with an environment variable. See the [Influx documentation](https://docs.influxdata.com/influxdb/v2.0/reference/config-options/). For example:

```
[http]
auth-enabled = true
```

is:

```
INFLUXDB_HTTP_AUTH_ENABLED=true
```

If you are running IOTstack on the "old-menu" branch, you make changes to

```
~/IOTstack/services/influxdb/influxdb.env
```

If you are running the "master" branch (ie new menu) then you make the changes in the influxdb section of `docker-compose.yml` and, if you are diligent, you will also make parallel changes in

```
~/IOTstack/services/influxdb/service.yml
```

Whichever method you use, if you then

```
$ cd ~/IOTstack
$ docker-compose up -d influxdb
```

that will cause `docker-compose` to notice the environmental change and that, in turn, causes it to recreate the container with the new settings.

You might find [this gist](https://gist.github.com/Paraphraser/9f3aa21fddd5d4e8141a7ab03ffb4dd7) helpful.

mvadu, 2021-02-01:
Docker supports proper config file usage and we don't have to figure out a translation of config file -> compose environment variables. I execute `docker run --rm influxdb influxd config > influxdb.conf` to generate the default congfig file and remove all non mandatory stuff, and place under `./volumes/config`, and map it to `/etc/infludb/influxdb.conf` which the default place where Influx engine is looking to get its configuration from.

```yml
  influxdb:
    container_name: influxdb
    image: "influxdb:latest"
    restart: unless-stopped
    ports:
    - "8086:8086"
    volumes:
    - ./volumes/influxdb/data:/var/lib/influxdb
    - ./volumes/influxdb/config/influxdb.conf:/etc/influxdb/influxdb.conf
    - ```

Paraphraser, 2022-06-19:
Pilot error. Stale.

closed 2022-09-11 by Slyke