SensorsIot/IOTstack #259: Git install is wrong?

tablatronix, 2021-01-28:
```
Manual
Install git
$ cd ~/IOTstack
$ git remote set-url origin https://github.com/SensorsIot/IOTstack.git
$ git pull origin master
$ git checkout master
$ docker-compose down
$ ./menu.sh
$ docker-compose up -d
```

Why not just git clone  instead of 4 steps ?

Also missing git init
git commands will fail, you have to git init first

Paraphraser, 2021-01-28:
I think it is - the migration instructions from gcgarner.

A lot of work needs to be done on the documentation. The best way to get things like this fixed is to [Prepare an IOTstack Pull Request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d). Slyke is usually pretty quick to apply changes, as long as you keep each PR simple and focused on fixing one thing.

There are still people turning up on gcgarner and I keep pointing them here so, if you do rewrite the instructions, please don't assume there is no need for a few words about how to migrate between repos. Ideally, people coming from gcgarner would be told to checkout old-menu because that's the smallest incremental step. Then they can try the master branch and adopt it if nothing breaks.

Just my two cents...

Prowling123, 2021-03-16:
Ehm, I seem to have some problems because i used the gcgarner instructions. Are there specific ones i would have to use instead of them?

Slyke, 2021-03-16:
Hey @Prowling123 Those commands are for migrating an existing version of IOTstack from gcgarner's repo to Andreas'. If it's a new installation, you can do with 2 ways:
```
curl -fsSL https://raw.githubusercontent.com/SensorsIot/IOTstack/master/install.sh | bash
```
or
```
$ git clone https://github.com/SensorsIot/IOTstack.git
$ cd ~/IOTstack
$ ./menu.sh
```
I just noticed that the readme instructions for manual installation is incorrect. I'll fix that up later tonight after work.

Prowling123, 2021-03-17:
@Slyke yes I am trying to migrate, but it didnt work from the gcgarner instructions

Prowling123, 2021-03-17:
So I did git hard reset and now it's back to the original state where menu.sh gives me the migrate popup

Paraphraser, 2021-03-17:
@Prowling123 I don't know if it will help but I've just added more material to https://gist.github.com/Paraphraser/107888b746a65d5d8f0e95a681471ebd to give you another option, which is to move your existing IOTstack out of the way, do a clean clone of the repo, then merge docker-compose.yml, services and volumes from the old folder into the new.

Paraphraser, 2021-03-17:
@Prowling123 also, I'm yet to be convinced that the `install.sh` is a good way to get started. That comment is based on some of the questions that get asked on Discord which seem to imply the basic scaffolding that IOTstack needs isn't always set up correctly.

I don't have any evidence that directly implicates `install.sh`, Just suspicions. I might be worrying about nothing. However, I wrote another gist on the topic which you might find useful if you keep having problems [Installing Docker for IOTstack](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f).

Every time I rebuild a Pi (which I mainly do to test the "if my Pi goes up in smoke and I have to get going again in a hurry with minimum risk of mistake" scenario), I install IOTstack as per [scripting-iotstack-installations](https://gist.github.com/Paraphraser/d119ae81f9e60a94e1209986d8c9e42f#scripting-iotstack-installations). I don't go near the menu or `install.sh`. I just run those commands, then restore a backup (taken from [Paraphraser/IOTstackBackup](https://github.com/Paraphraser/IOTstackBackup)), then `docker-compose up -d`. Job done!

Paraphraser, 2022-06-19:
My opinion on this is unchanged. Basically, you get a much better user experience if you let PiBuilder do the work of installing Docker and Docker-compose, than any of the "installers" in this repo. We really should fix this sometime so I think of this as an open issue.

ukkopahis, 2022-07-05:
gcgarner migration is getting very old and untested, maybe it's time drop support for it?