SensorsIot/IOTstack #260: no INFLUX-WEB-UI found on port 8083

SMS0815, 2021-01-31:
hello community... quite new to the IOT-Stack-Project...  so detailed answers are very much appreciated ; )

I´m running a clean installation of IOT-Stack on a Raspi 4... 
After building the stack with menu.sh,  "$ docker-compose up -d" brings up all the containers... 

portainer-ce is up-to-date
grafana is up-to-date
influxdb is up-to-date
home_assistant is up-to-date
nodered is up-to-date
homebridge is up-to-date

nice! I´m impressed!

So far, I´ve not managed to connect to the Web-UI on Port 8083...  
The UDP Service on port 8086 seems to be there... but no WEB-UI... so I bet I forgot something important...

best regards... 
stephan

SMS0815, 2021-01-31:
... a hint, how to access the influx cli would help me as well..

jahbini, 2021-01-31:
same problem - new install. Docker up and running.  However nmap says there are no open ports, and there is no response from web access to the system.

Paraphraser, 2021-02-01:
My guess is that "the Web-UI on Port 8083" means you're hoping to find a GUI administrator interface of some kind. Short answer: I don't think there is one. A more hopeful answer: if you ever succeed in finding one, please let me in on the secret too!

To access the Influx CLI:

```
$ docker exec -it influxdb influx -precision=rfc3339
> exit
$
```
		
The first line runs the influx CLI inside the container. That's what's giving the `>` prompt. You can type commands like "show databases". Getting out is a matter of typing `exit`. The `-precision=rfc3339` tells influx to display human-readable timestamps. If you leave it off you get nanoseconds since 1/1/70.

You might find these helpful:

* [InfluxDB Ports documentation](https://docs.influxdata.com/influxdb/v1.8/administration/ports/)
* [A gist on setting up some useful aliases for IOTstack](https://gist.github.com/Paraphraser/7612d3c780d284a502bd1f158c5186e8). It includes an `influx` alias which gets you straight into the CLI:

	```
	$ influx
	> exit
	$
	```
	
* Some InfluxDB-related gists which might be helpful in learning a bit more about how InfluxDB and Docker and IOTstack hang together:
	* [Enabling authentication in InfluxDB](https://gist.github.com/Paraphraser/9f3aa21fddd5d4e8141a7ab03ffb4dd7)
	* [Enabling UDP support](https://gist.github.com/Paraphraser/15483641fc657a147dfae7b03d291bc0)
	* [From Mosquitto to NodeRed to InfluxDB](https://gist.github.com/Paraphraser/c9db25d131dd4c09848ffb353b69038f)

<hr>

Now I'd like to drill down into the "Port 8083" question because it's piqued my interest.

The IOTstack template for InfluxDB defines these ports:

```
    ports:
      - "8086:8086"
      - "8083:8083"
      - "2003:2003"
```

Those three port mappings have been in the template since the get-go with [gcgarner/IOTstack](https://github.com/gcgarner/IOTstack/blob/master/.templates/influxdb/service.yml) and I doubt that anyone (including me) has given it all that much thought since.

Port 8086 we know about. Anything that wants to talk to InfluxDB (eg Node-Red doing database inserts or Grafana running queries) will use HTTP over TCP port 8086. If UDP support is enabled, that occurs over UDP port 8086.

According to the doco link above, port 2003 *runs the Graphite service*. I have zero idea what that is. In any event, it's disabled by default.

Port 8083 doesn't get a mention in the InfluxDB documentation. Googling the topic didn't get me any further. Do you have anything definitive you can point me at? If not, maybe it's just a mistake in the original IOTstack template.

On my "live" RPi (and DNET is an alias from the earlier gist):

```
$ DNET|grep -e "PORTS" -e "influxdb"
NAMES               PORTS
influxdb            0.0.0.0:2003->2003/tcp, 0.0.0.0:8083->8083/tcp, 0.0.0.0:8086->8086/tcp
```

So, Docker is reflecting what's in service.yml→docker-compose.yml. A small test script:

```
#!/bin/bash

if [ $# -eq 0 ] ; then
   echo "Usage: testPort port {ipaddr}"
   echo "       (ipaddr defaults to 127.0.0.1)"
   exit -1
fi

if nc -w 1 "${2:-127.0.0.1}" "$1" ; then
   echo "Port $1 responding"
else
   echo "Port $1 rejected"
fi
```

Use that to check the three external ports:

```
$ testPort 2003; testPort 8083; testPort 8086
Port 2003 responding
Port 8083 responding
Port 8086 responding
```

All responding. But all that proves is that Docker has implemented the port mappings. It has bound to those external ports and is listening, ready to forward any traffic to the corresponding internal ports. The question is, is anything listening inside the container?

Let's repeat the test from inside container-space. I have found the Node-Red container to be the most useful for network testing so I'll use that. Step 1 is to copy the script into the running container:

```
$ docker cp .local/bin/testPort nodered:/usr/src/node-red/testPort
```

Then go into the container (another alias) and repeat the test (the "influxdb" argument being the host name for the container, which looks-up the container's IP address on the internal network):

```
$ NODERED_SHELL
# ./testPort 2003 influxdb; ./testPort 8083 influxdb; ./testPort 8086 influxdb
Port 2003 rejected
Port 8083 rejected
Port 8086 responding
# exit
$
```

So, the processes running in the InfluxDB container are only paying attention to port 8086. That matches the InfluxDB documentation. It also matches the thinking of whoever designed the image for InfluxDB on DockerHub:

```
$ docker inspect influxdb:latest | jq -r .[0].ContainerConfig.ExposedPorts
{
  "8086/tcp": {}
}
```

Granted, that's only a *hint*. It doesn't govern what is or isn't exposed.

Bottom line: I can't find any evidence that InfluxDB implements port 8083 for any service.

What makes you think it does? I'm really curious!

jahbini, 2021-02-01:
> same problem - new install. Docker up and running. However nmap says there are no open ports, and there is no response from web access to the system.

after this, I went back to the old menu system, and tried again.  Still no connection to ports. --
Next, I moved the IOTstack out of the way and tried installing the whole thing over again.  This time it worked mostly as advertised.

Bottom line: some residual information in the original directory was fouling up the network.  If at first you don't succeed, blow the whole directory away and start over.

mvadu, 2021-02-01:
InfluxDB stopped supporting webUI long back. [There is no webUI anymore](https://docs.influxdata.com/chronograf/v1.7/administration/transition-web-admin-interface/).  You can however use [Influx Admin](https://grafana.com/grafana/plugins/natel-influx-admin-panel) which provides a nice UI to execute queries inside grafana directly. Don't be afraid its not updated in last 3 years, Influx query language hasn't changed in 3 years, so nothing to update there.

SMS0815, 2021-02-01:
... this is true,... Influx stopped supporting the AdminUI in v1.3. Seems I´ve been away from IOT-Stuff for too long.

but the hint for the CLI helped me a lot... it works fine.
As a cleanup, the 8083 Port could be removed from the docker-compose file?

THX for your quick response

MostHated, 2021-02-04:
I never saw what the "old" admin panel had in terms of configuration, but there is [Chronograf](https://www.influxdata.com/time-series-platform/chronograf/)
On the Chronograf site, it has the following quote:

> What is Chronograf?
> Chronograf is the user interface and administrative component of the InfluxDB 1.x platform.

So perhaps it might have some of the functionality you are looking for?

Paraphraser, 2022-06-19:
A combination of pilot error and misunderstanding that influxdb 1.8 doesn't have a web UI while influxdb 2 does. Stale.

closed 2022-09-11 by Slyke