SensorsIot/IOTstack #263: Exec Console: No bash in some containers

snoozer, 2021-02-02:
Hi,

I have tried to use Portainer to access the CLI of the Mosquitto Container without success. I have left the defaults /bin/bash and root. This flickered a black background very quick and brought me beck to where I started. Since I did not see any error message I tried the CLI approach on the host system.

`docker exec -it mosquitto /bin/bash`

This gave me the following error:
`OCI runtime exec failed: exec failed: container_linux.go:370: starting container process caused: exec: "/bin/bash": stat /bin/bash: no such file or directory: unknown`

Using `/bin/sh` instead of /bin/bash then worked, is that intended not to have a bash in the container or is that an oversight ? Is there a way that I can add the bash ?

I have noticed the same issue in the Adminer container as well.

Kind Regards
Jan P.

Paraphraser, 2021-02-02:
I agree that it can be annoying. You can add Portainer-CE to your list; neither bash nor sh.

However, whether an image pulled down from Dockerhub has a shell program (or, indeed, any other program) installed inside the image is not something that IOTstack controls. Decisions about what is, or isn't, inside a container are made by whoever builds the image. We IOTstack users just run the image.

If you think a good case can be made for any container that doesn't have a shell program to have that added to its DockerHub image, you need to raise that with whoever maintains the DockerHub image.

I hope that makes sense.

Paraphraser, 2021-02-03:
I've realised that I didn't actually answer this part of your question:

> Is there a way that I can add the bash

The answer is "yes".

1. Create a Dockerfile. These two lines:

	```
	FROM eclipse-mosquitto
	RUN apk update && apk upgrade && apk add --no-cache bash
	```

	need to be in a file at this path:
	
	```
	~/IOTstack/services/mosquitto/Dockerfile
	```
	
	You'll notice `bash` on the end of RUN command. That's adding `bash` to Mosquitto.

2. If Mosquitto is running, take it down:

	```
	$ cd ~/IOTstack
	$ docker-compose stop mosquitto
	$ docker-compose rm -f mosquitto
	```
	
	That's the equivalent of `docker-compose down` but it only applies to Mosquitto.

3. Edit docker-compose.yml. At the moment, the first part of the mosquitto chunk looks like this:

	```
	  mosquitto:
	    container_name: mosquitto
	    image: eclipse-mosquitto
	    restart: unless-stopped
	```

	Make it look like this:

	```
	  mosquitto:
	    container_name: mosquitto
	    build: ./services/mosquitto/.
	#   image: eclipse-mosquitto
	    restart: unless-stopped
	```

4. Bring up mosquitto with the `--build` flag:

	```
	$ cd ~/IOTstack
	$ docker-compose up -d --build mosquitto
	```

	You'll see a bunch of guff and, towards the end, a mention of installing `bash`.
	
5. Test the result:

	```
	$ docker exec -it mosquitto bash
	bash-5.0$ exit
	$
	```
	
	Mosquitto now has `bash`!

6. Understand what is going on:

	```
	$ docker images | grep -e "REPOSITORY" -e "mosquitto"
	REPOSITORY               TAG         IMAGE ID       CREATED          SIZE
	iotstack_mosquitto       latest      44bcffe814bc   2 minutes ago    11.3MB
	eclipse-mosquitto        latest      c83ed7264442   6 weeks ago      8.08MB
	```
	
	The "iotstack_mosquitto" line is the "local image". It is what has just been built. It only exists on your RPi. It is this image that becomes the running container. The local image has `bash` and so does the running container.
	
	The "eclipse-mosquitto" line is the "base image" from DockerHub. That image used to become the running container but it's still needed because the changes we've just made are applied on top.

7. Understand the maintenance problem.

	For most containers you can get updates from DockerHub just by doing this:
	
	```
	$ cd ~/IOTstack
	$ docker-compose pull
	$ docker-compose up -d
	$ docker system prune
	```
	
	The `pull` will check for any updates and download any later images, then the `up` re-creates any containers where the images have been updated. The `prune` cleans up the mess left behind by any old images.
	
	Those commands won't work for containers that are built with a local Dockerfile. Node-Red is another example of this problem. If you know that Mosquitto has been updated on DockerHub (and the only way I know of is to go to DockerHub and check), then you need to force it like this:
	
	```
	$ cd ~/IOTstack
	$ docker-compose build --no-cache --pull mosquitto
	$ docker-compose up -d mosquitto
	$ docker system prune
	```

	The "build" with those arguments forces the base image to be re-downloaded from DockerHub, whether it has changed or not. If the base image has actually changed then that, in turn, forces the Dockerfile run and you get a new local image.

Thus, the price you pay for customising an image is not getting automatic updates from Dockerhub.

The question for you is, "is it worth all this just to get `bash` inside Mosquitto?"

Please don't hear any negative overtones in that question because this is the kind of thing I do on my own IOTstack. I focus on [customising Node-Red](https://gist.github.com/Paraphraser/c8939213faf2de8a10f2a1f67452b0c1) but that's because:

* it is already built from a Dockerfile so it adds no additional maintenance overheads to use it as the basis of my own customisations; and
* Node-Red is in the middle of everything I do so, if a **flow** doesn't work, my first question is "what can the **container** see?". Answering those questions needs add-ons like the mosquitto clients, tcpdump, DNS tools, and so on.

I work from the command line (I don't see the point of Portainer) so I heavily customise the command-line environment with shell scripts and a [scheme of aliases](https://gist.github.com/Paraphraser/7612d3c780d284a502bd1f158c5186e8). If I want to get into the Node-Red or Mosquitto shells, it's:

```
$ NODERED_SHELL
$ MOSQUITTO_SHELL
```

The fact that the first invokes `bash` while the second invokes `sh` is kinda unimportant. But I **would** take this Dockerfile approach if I really needed `bash` inside Mosquitto.

None of this is to suggest that my way is the "right way" or anything like that. It works for me and it **might** work for you too.

Hope this helps.   

<hr>

If you follow along and then decide you want to undo the changes:

1. Take Mosquitto down.
2. Edit docker-compose.yml to undo the edits.
3. You can leave the Dockerfile where it is or remove it - your choice.
4. Bring Mosquitto up. That will go back to the base image.
5. Run the commands:

	```
	$ docker rmi iotstack_mosquitto
	$ docker system prune
	```

You should then be back to where you were.

Paraphraser, 2022-06-19:
I'd call this a misunderstanding (not every container has a shell, not every shell is bash). Stale.

closed 2022-09-11 by Slyke