SensorsIot/IOTstack #271: "502 Bad Gateway" when running duck.sh from crontab

senarvi, 2021-02-12:
I noticed that when I run `IOTstack/duck/duck.sh` in crontab, it often (not always) produces the following error in `/var/log/duck.log`:

```
<html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
</body>
</html>
```

By accident, I noticed also that this doesn't happen if I add a five second delay before the curl command. Have others duckdns users experienced this?

Sounds like this has to be a general issue with Linux networking or cron, but I did a fair bit of searching and didn't find anything. I can create a pull request that adds the five second delay, if I'm not the only one experiencing this. Of course it would be nice to also understand what's causing the problem.

I'm running IOTstack on Raspberry Pi OS.

Paraphraser, 2021-02-13:
A 502 is usually something beyond your control but, sometimes, these turn out to be because of DNS timeouts. `curl` asks the local resolver to find the IP address for "www.duckdns.org", some part of the chain takes too long to respond, times-out and you get a mess. This "mess" is not necessarily on your system.

I can't explain why a 5-second delay would help anything but what I've found can sometimes be effective - emphasis on **sometimes** - is to try to prime the DNS.

Start by trying this command:

``` 
$ dig +short www.duckdns.org
```

If that doesn't work, you'll need to install the dnsutils package

```
$ sudo apt install -y dnsutils
```

then retry the command. The output will look something like this:

```
DuckDNSAppELB-570522007.us-west-2.elb.amazonaws.com.
35.160.123.194
44.230.159.203
```

Next, try changing the script like this:

```
#!/bin/bash
# Your comma-separated domains list
DOMAINS="YOUR_DOMAINS"
# Your DuckDNS Token
DUCKDNS_TOKEN="YOUR_DUCKDNS_TOKEN"
# where DuckDNS should store its log
DUCKDNS_LOG="/var/log/duck.log"

# create the log file if it does not exist
if [ ! -f "$DUCKDNS_LOG" ] ; then
   sudo touch "$DUCKDNS_LOG"
   sudo chgrp "$USER" "$DUCKDNS_LOG"
   sudo chmod 664 "$DUCKDNS_LOG"
fi

# prime the DNS resolver cache
dig +short www.duckdns.org >>"$DUCKDNS_LOG"

# update DuckDNS
curl -k -o "$DUCKDNS_LOG" "https://www.duckdns.org/update?domains=${DOMAINS}&token=${DUCKDNS_TOKEN}&ip="
```

You'll notice I also go to the trouble of setting up the log file if it isn't there. I don't use DuckDNS myself so I have no idea how the log file gets created in IOTstack. It just seems to me to make more sense to include "auto-repair" in any script.

If adding the `dig` solves the problem for you then by all means propose a PR to include it. It does no harm and doesn't incur any delays over and above those implicit in the DNS lookup `curl` would still have to do if the local resolver did not have the answer in its cache. However, some thought will have to be given to checking whether `dig` is available. It's part of my standard install sequence so it's not an issue I'd face.

Another approach (the one I use) is to see if your router has Dynamic DNS functionality. A lot do. Your router knows when its WAN IP changes so it's really a better judge of when and how often to send updates/retries/refreshes to your chosen Dynamic DNS service provider. During the last 15 years (give or take) the only time I have not been able to reach my home network via its dynamic DNS name is when there was no viable path (eg ISP link down, or something like a routing loop or a global DNS malfunction wrecking everything for everyone). I've never had to worry about whether a 5-second delay will make any difference.

senarvi, 2021-02-16:
Thanks for taking the time to answer, @paraphraser! My router supports a couple of Dynamic DNS providers, but they cost. Duck DNS is free and works fine. The five second delay makes no difference. I was mostly curious whether others can confirm this strange behavior.

A DNS problem was my first thought too, since I've recently played around with a local DNS server. However, I shouldn't have any non-standard DNS configuration anymore. The first thing I tried is what you proposed too, although I used `nslookup`.
I never saw any problems resolving the IP address. Just to be sure I now tried with dig too, just like you proposed. It resolves the IP address, but curl fails (unless I add the delay).

In this case the "bad gateway" problem is clearly not beyond my control, because it only happens when the script is run by cron. It's not unrelated to duckdns either. I tried to retrieve a bunch of other web pages in the beginning of the cron script and never saw any problems with any other web page.

It sounds unbelievable that only www.duckdns.org would fail, and only in the beginning of a cron job. The only thing that comes to my mind is whether it could be related to duckdns detecting my gateway IP. I tried to retrieve the duckdns landing page and this failed too, if it was done fast enough.

So the simplest way to reproduce the bug is to create a cron job that runs `curl "https://www.duckdns.org/"` every five minutes. Cron will mail you the output (or if you don't have postfix installed, you can direct it to a file). It should be `OK`, but usually I see

```
<html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
</body>
</html>
```

Paraphraser, 2021-02-16:
Oh boy! I have **definitely** been able to replicate your experience on my RPi.

Here's the weird thing. Like an ever-increasing number of people on the planet, I have a solar system which logs metrics to the manufacturer's cloud service. About five years ago I wrote a script that pulls down "yesterday's" CSV from the cloud, and loads it into a database. It runs at 3am as a cron job. It uses curl a couple of times, once to set up a cookie, and then to pull down the data. The only time it misses a beat is if something happens beyond my network (server down, transmission interrupted). Never once in close-on 2000 invocations have I seen a 502 error. Not once.

BUT: the script I'm talking about runs on my Mac whereas we're talking about RPi. Maybe it's a macOS vs Raspbian thing?

<hr>

Talk about puzzling!

I also see what you mean about "delays". Pretty much anything I add before the `curl` seems to affect the outcome but it's still as intermittent as all get-out.

I thought to myself, "if at first `curl` won't succeed, why not try `wget`?" The DuckDNS doco only talks about the URL. It is silent as to how it's actually invoked. 

crontab:

```
SHELL=/bin/bash
HOME=/home/pi
PATH=/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
USER=pi

#min	hour    mday    month   wday    command

# report system temperature every 5 minutes 
*/5	*	*	*	*	curlTest >>./Logs/curlTest.log 2>&1
*/5	*	*	*	*	wgetTest >>./Logs/wgetTest.log 2>&1
```

curlTest:

```
#!/usr/bin/bash

echo "Running at $(date)"
/usr/bin/curl -G https://www.duckdns.org/
echo "Exit code $?"
```

wgetTest:

```
#!/usr/bin/bash

echo "Running at $(date)"
/usr/bin/wget -O - https://www.duckdns.org/
echo "Exit code $?"
```

The output from curlTest is as you described. I'll just note that the exit code always returns zero whether it succeeds or not.

The first time wgetTest ran, it succeeded. I thought "now we're cookin'" but the second time it gave me:

```
Running at Tue 16 Feb 2021 10:05:01 PM AEDT
--2021-02-16 22:05:01--  https://www.duckdns.org/
Resolving www.duckdns.org (www.duckdns.org)... 44.230.159.203, 35.160.123.194
Connecting to www.duckdns.org (www.duckdns.org)|44.230.159.203|:443... connected.
HTTP request sent, awaiting response... 502 Bad Gateway
2021-02-16 22:05:03 ERROR 502: Bad Gateway.

Exit code 8
```

So, we have a commonality in the 502 and a difference in a sensible exit code, which `man wget` describes as:

```
8   Server issued an error response.
```

That may offer a way forward - iterate the request if non-zero.

<hr>

In the past I've found `cron` differences can often be explained by environment differences but I couldn't spot any likely candidates.

<hr>

After letting both scripts run for a while:

```
$ grep -c "^Exit code" curlTest.log wgetTest.log
curlTest.log:6
wgetTest.log:5
$ grep -c -e "<title>502 Bad Gateway</title>" -e "ERROR 502: Bad Gateway" curlTest.log wgetTest.log
curlTest.log:5
wgetTest.log:2
```

So, 5 out of 6 `curl` fails, 2 out of 5 `wget` fails. Almost as bad as a government service!

I reset the logs, changed the URL (in both scripts) to a completely different website, and let it run for a while:

```
$ grep -c "^Exit code" curlTest.log wgetTest.log
curlTest.log:6
wgetTest.log:6
$ grep -c -e "<title>502 Bad Gateway</title>" -e "ERROR 502: Bad Gateway" curlTest.log wgetTest.log
curlTest.log:0
wgetTest.log:0
```

6 attempts for both and no failures. `wget` assigning blame to the server. Also keep in mind that, possible macOS vs Raspbian differences aside, the script I talked about at the start has been running for over five years without hitting a 502.

I reckon the preponderance of evidence is strongly in favour of this all being down to "DuckDNS".

I put "DuckDNS" in quotes because I have an open mind about whether it's AWS or DuckDNS but I find myself wondering whether someone has implemented a "random disconnect" kind of algorithm which is intended to mitigate DDoS attacks and the like.

It's also possible that "502" is DuckDNS's way of quickly disposing of updates that serve no purpose (ie WAN IP has not changed) and/or preventing any given WAN IP from sending updates too frequently. 

Thoughts?

ps: Up to 9 tries, no fails for the non-DuckDNS URL. Time to turn it off.

Paraphraser, 2021-02-16:
Something else occurred to me. I stumbled across this as an effective way for a script to ask "what's my WAN IP address?"

```
$ wget -q -O - ifconfig.me
```

> I have absolutely no idea why that works. If you know, please explain.

Then there's asking the DNS the "what IP is associated with my DuckDNS name?"

```
$ dig +short example.duckdns.org
```

or, maybe the better form:

```
$ dig +short @ns1.duckdns.org example.duckdns.org
```

So, we can imagine a script with something like this:

```
WANIP=$(wget -q -O - ifconfig.me)
DNSIP=$(dig +short @ns1.duckdns.org example.duckdns.org)
if [ "$WANIP" = "$DNSIP" ] ; then
   echo "WAN.IP and DuckDNS agree at $(date) - no update required"
else
   echo "WAN.IP $WANIP differs from DuckDNS $DNSIP at $(date) - sending update"
   while [ $? -ne 0 ] ; do
      wget -q -O - https://www.duckdns.org/update?...
   done
fi
```

Fleshing it out might stick a delay before the wget so it doesn't "hammer" the service, or maybe combo:

```
   echo "WAN.IP $WANIP differs from DuckDNS $DNSIP at $(date) - sending update"
   wget -q -O - https://www.duckdns.org/update?...
   if [ $? -eq 8 ] ; then
      echo "502 error - retrying once in 5 seconds:
      sleep 5
      wget -q -O - https://www.duckdns.org/update?...
   fi
```

and then leave it to cron to retry.

If it were me, I'd probably just have a single attempt per cron cycle, and leave the implied retry to the next cron cycle.

Sound like a plan?

senarvi, 2021-02-16:
Something occurred to me. I thought that maybe it's not about whether we invoke curl in the beginning of the cron job, but whether we invoke it at the exact moment of time. Now we invoke it exactly on the hour, 5 minutes past and so on. So I change the cron job to run every minute. I can see that those invocations that happen on the hour, 5 minutes past and so on will still fail, but the invocations between them won't fail! My hypothesis is that the duckdns server is also running a cron job every 5 minutes that stops the service for a couple of seconds. :D

senarvi, 2021-02-16:
To answer your question about ifconfig.me, it's a web server that returns the client's IP address in the HTTP response. Your gateway is doing network address translation, so your WAN IP is what the server sees.

senarvi, 2021-02-16:
It would be nice to check first whether the IP address has changed. My only concern is, if we use ifconfig.me for obtaining the external IP address, when enough people start using the script, are we going to see similar issues with ifconfig.me?

Paraphraser, 2021-02-17:
Good points. I must've been living under a rock because I'd never heard of the .me TLD so I didn't recognise what I was looking at. I thought it was some wacky special case.

rckrchrdsn, 2021-03-28:
Just setting up my Rpi with DuckDNS and found the same issue, with a 502 about 3 out of 5.   Interesting to read the conversation.

rckrchrdsn, 2021-03-28:
Just added this to CRON and I get no 502's

#Delay the job randomly between 10 and 40 seconds.
DLY_SITE_CHK="RANDOM % 30 + 10"
*/5 * * * * d=$(($DLY_SITE_CHK)) ; sleep ${d}s ; ~/IOTstack/duck/duck.sh >/dev/null 2>&1

Than senarvi  and Paraphraser for the discussion and the direction!

senarvi, 2021-03-29:
I made a suggestion how to change the script: https://github.com/SensorsIot/IOTstack/pull/298/files

What do you think?

rckrchrdsn, 2021-03-30:
Yours is better than my implementation. I now officially hate you. 

HOWEVER, in the discussion on Reddit with jjohnp he pointed out that we can also add the starting time in the hour and the interval so -59/5 in the first position would start minute 3 then fire at 8, 13, 18... by showing n-59/5 with n = 1,2,3,4,5 we could spread th the load to 5 * 60 slots in or 300 would divide the load. 

Your is a very simple change. (Again, I am jealous and hate you.) and the change to the instructions for the CRON to replace the first number 

Thanks for showing me this. I have already changed my code and I plan to bring your change up and I will put your the link you sent me in.

Paraphraser, 2022-06-19:
I am pretty sure DuckDNS now works as advertised (certainly never gives me any trouble) so this is probably fixed.

closed 2022-09-11 by Slyke