SensorsIot/IOTstack #276: unable to stop and remove Hassio containers in IOTstack

mohankrishna-g, 2021-02-18:
unable to stop and remove Hassio containers in IOTstack

i install hassio through menu.sh in baremettle ubuntu desktop from @gcgarner/IOTstack.
as it is outside IOTstack. im unable to stop hassio containers. tried remove docker and stopping containers and update the policy to restart unless stopped.but nothing worked. need to get the location details of hassio container and i want to remove hassio containers as im not using HA.

didn't worked,
Stop all running containers
docker stop $(docker ps -aq)
Remove all containers
docker rm $(docker ps -aq)
Remove all images
docker rmi $(docker images -q)

tried updating this flag tooo
docker update --restart unless-stopped $(docker ps -q)

Paraphraser, 2021-03-02:
I assume you also tried them one at a time rather than in the form shown. Does hassio actually show up when you do a `docker ps`? At the `docker rm` step did you try adding the `-f` flag to force removal?

mohankrishna-g, 2021-03-02:
@pharaphaser Yes I tried all the possibilities to stop the containers.
As the container running I. Always start mode even do containers stoped for a sec by forcing them again they started..so I posted above issue. After while searching I found above documentation which helped me stop haasil containeers

closed 2021-03-02 by mohankrishna-g