SensorsIot/IOTstack #278: Openhab

RubyBailey, 2021-02-22:
After installing Openhab I noticed that it's ports were not 8080 and 8443, per the default settings.
The [IOTstack documentation for Openhab](https://sensorsiot.github.io/IOTstack/Containers/openHAB/) states that the port bindings are 8080 and 8443, whereas the [service.yml file](https://github.com/SensorsIot/IOTstack/blob/master/.templates/openhab/service.yml) is set up for 4050 and 4051.
Wondering why?

RubyBailey, 2021-02-23:
After some searching I noticed the changes from ports 8080 and 8443, to 4050 and 4051 in the service.yml file were committed July 20, 2020 by @Slyke 
Any reason why ports were changed from their standard configuration? I cannot find notes or documentation for the changes. 
Thanks

Slyke, 2021-02-24:
Hey @RubyBailey Many IOTstack services use those same ports, so had to change them in case a user selects services that have conflicting ports.

I've listed the HTTP ports on this page: https://sensorsiot.github.io/IOTstack/Default-Configs/

You can override the ports that are placed into the `docker-compose.yml` file by using the `compose-override.yml` & rebuilding the stack. Instructions here: https://sensorsiot.github.io/IOTstack/Custom/#overriding-default-settings

Slyke, 2021-02-24:
I will update the incorrect documentation shortly, so that there's no confusion. Thanks for finding that!

RubyBailey, 2021-02-24:
Thanks for the reply.
I understand what your trying to accomplish, this has been a challenge for some time from a distribution perspective.
Yet I disagree with this approach for a number of reasons:
1) The chances of port conflicts is only dependent on the services someone installs and therefore should be managed by that specific user. That's when overrides should be used, not when someone wants to get back to the default settings.  
2) Arbitrarily assigning new ports, HTTP and HTTPS is a breaking change for many systems. Something that should be flagged by menu.sh. 
3) It would be better if menu.sh picked up that a user had selected services that use the same ports (as their defaults) and warned them to make changes. Not sure if that's possible, but a better approach in my opinion.
4) Personally,  I think it will be far easier to manage the docker-compose files for the IOTstack services if they are as close as possible to those published on the docker hub. Otherwise IOTstack will become difficult to manage as updates are made to the individual services/packages on Docker hub.
5) Lastly, the document you referenced only lists the HTTP ports. For completeness it should also list HTTPS ports. 

I think this approach makes it more difficult to manage systems, particularly because nothing is broken or wrong with the default settings. Conflicts will only happen when a user selects services with conflicting ports and that's when those ports/services should be managed or overridden.
Thank again for getting back to me. Like to get your feedback on the above points.

ukkopahis, 2022-06-18:
> The chances of port conflicts is only dependent on the services someone installs and therefore should be managed by that specific user

The IOTstack project aims for a beginner friendly experience where everything "just works" out of the box. Manually having to adjust ports would defeat its purpose.

> Arbitrarily assigning new ports, HTTP and HTTPS is a breaking change for many systems

Have a look at all the services provided. There are like seven different services that by default would use port 8080. Citation needed for claim about breaking change.