SensorsIot/IOTstack #279: InfluxDB install gives error: "ERROR: no matching manifest for linux/arm/v7 in the manifest list entries"

NickXML, 2021-02-25:
I installed Grafana, influxdb, node-red and portainer-ce without issue just last week.

I tried to do a new install today and influxdb seems to break everything.

I see the following error:
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries

When I checked the Influx docker page, it seems they may have changed settings to use the 2.0 version of Influxdb.

I edited the docker-compose.yml line for influxdb, changing: 

image: "influxdb:latest"

to

image: "influxdb:1.8"

Which resolved the issue.

Apologies if I have not provided all of the correct information -- Docker and IOTStack are both new to me as of last week.

Paraphraser, 2021-02-28:
Please see [Issue 283](https://github.com/SensorsIot/IOTstack/issues/283).

tablatronix, 2021-04-08:
I just did the same, seems 2.x required a setup or init to migrate or auth changed also will check later whats involved

Paraphraser, 2021-04-08:
Repeating what I said on Discord, the underlying problem is that InfluxDB 2.x has not been built for armv7.

You can stare at the [tags page](https://hub.docker.com/_/influxdb?tab=tags&page=1&ordering=last_updated) as much as you like but the first mention of armv7 is against 1.8.4.

When 2.0 was first released we thought this might be an omission and an armv7 version would eventually appear on DockerHub. It's been a long time with no armv7 so my guess is it wasn't a mistake and InfluxDB 2.x will never be available for armv7. 

That's what this pattern means:

```
Pulling influxdb (influxdb:latest)...
latest: Pulling from library/influxdb
ERROR: no matching manifest for linux/arm/v7 in the manifest list entries
$ uname -a
Linux new-dev 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
```

Docker is looking at the "armv7l" we see in the `uname` output and is trying to find the "influxdb:latest" built for "Linux/arm/v7", and is not getting anywhere. The only solution for us, at least for now, is to pin to 1.8.4 as per [issue 283](https://github.com/SensorsIot/IOTstack/issues/283).

Although I have not been able to find this written down anywhere (eg an "InfluxDB minimum system requirements" list) my guess is that this is a 64-bit thing. My understanding is that our Pi 4s are ARMv8 which are 64-bit chips. It's Raspbian that's 32-bit. When [Raspbian 64-bit](https://raspberrytips.com/raspberry-pi-os-64-bits-vs-32-bits/) becomes generally available then, presumably, Influx will support it and we'll be able to get our hands on it. And **then** we will have to worry about "migration" and so on. 😏

I hope this helps.

tablatronix, 2021-04-08:
I am running 64bit rasbian

Paraphraser, 2021-04-08:
You're braver than I am. I'm not going near 64-bit until it's a recommended OS.

You can pin to 1.8.4:

```
  influxdb:
    container_name: influxdb
    image: "influxdb:1.8.4"
```

What seems to be happening is that, each time a bug is found, InfluxDB is rebuilt as 1.8.4. Under the hood it might be 1.8.4a, 1.8.4b, and so on, but it matches the 1.8.4 suffix. So, we do get updates on a "pull", even though the version number doesn't change.

<hr>

On the flip side, once you work out how to migrate to 2.0 then please write it down. Maybe create a gist.

tablatronix, 2021-04-08:
if you pin to 1.8.4 you wont get minor updates, that was what I was saying, also do you know what the `meta` tags are?

Paraphraser, 2021-04-08:
But that simply is not true! I've been pinned to 1.8.4 since Feb 28 and in the time since, a `docker-compose pull` has downloaded at least three influxdb updates. They are all tagged "1.8.4" in `docker images`.

See this?

[image]

Only 8 days old. Updated **this month**. Nowhere near Feb 28 when I added the pin (and it must've been older than Feb 28 at the time). But still 1.8.4.

As to meta tags, I don't understand the question. Something from `docker image inspect`?

tablatronix, 2021-04-09:
Thats bizarre.. and confusing. Heh

there are tags like 1.x-meta

PR #567 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3
Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
commit 3239e9b4 (2022-06-12): 20220602 InfluxDB 2 - master branch - PR 1 of 3

Adds InfluxDB 2 container and documentation.

Fixes #455, Fixes #289, Fixes #279, Fixes #169, Fixes #132
closed 2022-06-12 by Slyke