SensorsIot/IOTstack #282: Adding Smokeping

theDiverDK, 2021-02-25:
It would be really nice, if someone could add:
https://hub.docker.com/r/linuxserver/smokeping

To the setup, so we can check if our internet connection is working well.

Paraphraser, 2021-03-03:
IMO, there's nothing worse than opening an issue and getting nothing but silence.

The first thing I'll say is that, if you want to add a new container to IOTstack, the best way to go about it is to [prepare a Pull Request](https://gist.github.com/Paraphraser/818bf54faf5d3b3ed08d16281f32297d). Part of my intention in responding is to explain "where I got up to" to give you a starting point for preparing a Pull Request (always assuming, of course, that you needed or wanted such a starting point and that your ultimate goal is to have SmokePing available in IOTstack).

As you read the rest, please keep in mind that I'm not "in charge" of anything. The feedback I'm giving you is not a **decision** because I have no power to make decisions. I'm just a randomly-self-selected representative IOTstacker giving it a whirl and offering a two-cents'-worth instant reaction to what I see. I don't have any prior knowledge or expectations about SmokePing to guide me and, to be honest, I suspect that a very large component of the somewhat negative tone you're going to sense in what follows has its origins in my complete inability to figure out just what the heck SmokePing is meant to do.

That's actually the first "petty annoyance" I'd like to lay on the table. Most spruikers of a product have some kind of "elevator pitch" which tells people not-in-the-know (like me) a bit about what the thing is and why it might deserve a place in their IOTstack. Nothing speaks to me at either:

* [DockerHub](https://hub.docker.com/r/linuxserver/smokeping); or
* [GitHub](https://github.com/linuxserver/docker-smokeping)

What I take to be the [home page](https://oss.oetiker.ch/smokeping/) still leaves me without an answer to the very basic question of, "why would I actually *want* it?" Maybe Andreas Spiess will do one of his excellent videos on the topic and the scales will fall from my eyes.

> I might also be tempted to ask why I might need it in addition to my [SamKnows Whitebox](https://samknows.one/hc/en-gb/articles/360000451757) but let's put a pin in that. 

However, assuming that the true purpose of any even-halfway-useful application will immediately become self-evident on running it up, in with a penny…

Starting position (docker-compose.yml fragment).

```
  smokeping:
    image: ghcr.io/linuxserver/smokeping
    container_name: smokeping
    environment:
      - PUID=1000
      - PGID=1000
      - TZ=Australia/Sydney
    volumes:
      - ./volumes/smokeping/config:/config
      - ./volumes/smokeping/data:/data
#   network_mode: host
    ports:
      - 8085:80
    restart: unless-stopped
```

Spin it up.

```
$ docker-compose up -d smokeping
Pulling smokeping (ghcr.io/linuxserver/smokeping:)...
latest: Pulling from linuxserver/smokeping
cc95a97ca69a: Pull complete
3ef44a04eed3: Pull complete
3512d30de144: Pull complete
7677a1b5ca99: Pull complete
28f927e73300: Pull complete
47c76ce4ec87: Pull complete
d79e72f76089: Pull complete
3a831d5c0ea5: Pull complete
47d7650b6932: Pull complete
Digest: sha256:fcd26daee97fe767d6a608b20153a769aec140edb5442344ee8986eba68ee77b
Status: Downloaded newer image for ghcr.io/linuxserver/smokeping:latest
Creating smokeping ... done
```

Seems to be working. What got created in the persistent storage area:

```
$ tree -pu ./volumes/smokeping/
./volumes/smokeping/
├── [drwxr-xr-x pi      ]  config
│   ├── [-rw-r--r-- pi      ]  Alerts
│   ├── [-rw-r--r-- pi      ]  Database
│   ├── [-rw-r--r-- pi      ]  General
│   ├── [-rw-r--r-- pi      ]  httpd.conf
│   ├── [-rw-r--r-- pi      ]  pathnames
│   ├── [-rw-r--r-- pi      ]  Presentation
│   ├── [-rw-r--r-- pi      ]  Probes
│   ├── [drwxr-xr-x pi      ]  site-confs
[... 179 lines cut ...]

[cont-init.d] 10-adduser: exited 0.
[cont-init.d] 30-config: executing... 
[cont-init.d] 30-config: exited 0.
[cont-init.d] 99-custom-files: executing... 
[custom-init] no custom files found exiting...
[cont-init.d] 99-custom-files: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
WARNING: Hostname 'ipv6.google.com' does currently not resolve to an IPv4 address
```
No obvious reaction in the log, one way or the other.

Connect a browser on Port 80. Wait an hour. No discernible difference.

Maybe it's a Safari thing - I'll try Chrome. No discernible difference.

I'm sorry but I give up. My guess is that more config work needs to be done under the hood, after which it will spring into life.

Compare/contrast Node-Red. Install the container. Spin it up. Creates its persistent storage area with sensible defaults. Next stop is the GUI where you can start creating flows that actually work. Ditto Grafana. That's how it's done!

So, if you decide to go down the path of preparing a Pull Request, please try to consider the problems I've laid out here. To be useful in IOTstack, a container really needs to **work** right out of the box. Otherwise it simply generates endless issues and Discord questions. That's got to be a key goal (after solving the dependency problem). And a bit of doco saying how SmokePing will solve all the world's problems would go some way to filling the "eh?".

If you have solutions to any of this that you want me to try out then I'm happy to help. If the solution involves a Dockerfile to build a custom version so that IOTstack-relevant sensible defaults are bundled into the image, I can help with that too.

ooono-developer, 2021-03-03:
Hi There

Thanks for your reply, and guide.

I will see if I can get the time to provide a pull request myself.

Thanks for your time.

Have a great day :)

theDiverDK, 2021-03-03:
Hi There

Thanks for your reply, and guide.

I will see if I can get the time to provide a pull request myself.

Thanks for your time.

Have a great day :)

Paraphraser, 2022-06-19:
No further response. Should probably be considered stale.

closed 2022-09-11 by Slyke